beagleboard/bb.org-overlays #6: Makefile uses a dtc-1.4.0 option '-@' that doesn't exist in 1.4.1

ChrisDumont, 2015-10-07:
The documentation recommends installing dtc-1.4.1 which no longer has the '-@' option. However when I remove '-@' from the Makefile I still get an error (see below). When I use dtc-1.4.0 it runs smoothly.

# make
  DTC     src/arm/BB-I2C1-PCA9685-00A0.dtbo
Error: src/arm/BB-I2C1-PCA9685-00A0.dts:9.2-8 syntax error
FATAL ERROR: Unable to parse input tree
Makefile:133: recipe for target 'src/arm/BB-I2C1-PCA9685-00A0.dtbo' failed
make[1]: *** [src/arm/BB-I2C1-PCA9685-00A0.dtbo] Error 1
Makefile:81: recipe for target 'all_arm' failed
make: *** [all_arm] Error 2

RobertCNelson, 2015-10-07:
Use the provided "dtc-overlay.sh" script to install a patched version of dtc, those patches haven't hit mainline dtc yet...
https://github.com/beagleboard/bb.org-overlays/blob/master/dtc-overlay.sh
Regards,

RobertCNelson, 2015-10-07:
btw, the version of "v1.4.0 with @ support" is mostly like for "v3.8.x's overlays"... for v4.1.x's overlays there's an extra node helper inside the *.dtbo, thus you need to be "very" careful that you only use one dtc for v3.8.x overlay kernels and this newer patch for v4.1.x+ overlay's..

Regards,

ChrisDumont, 2015-10-07:
Now I get it. Thanks for the quick reply.

Chris

ChrisDumont, 2015-10-07:
This is a new jessie install with 4.1.10-bone16 kernel so I don’t have any 3.8 stuff. But thanks for the warning.

RobertCNelson, 2015-10-07:
My plan, once they get accepted in dtc mainline, i'll bump the device-tree-compiler debian package for (jessie/stretch+ and ubuntu wily+1 for v4.1.x kernels)  While Wheezy/Trusty will stay with the old 3.8-dtbo generation..

Regards,

ChrisDumont, 2015-10-07:
For sid/stretch yes, for jessie I doubt it. I would think that Debian policy would prevent a version bump especially since the standard/stock kernel is unlikely to advance to 4+ and dtc has to work with a stock kernel.

RobertCNelson, 2015-10-07:
yeah, i meant a bump in our external staging repo for beagleboard.org
https://github.com/rcn-ee/repos