beagleboard/bb.org-overlays #10: CTS/RTS support for UART capes?

agherzan, 2015-11-12:
Are there any plans to add capes for CTS/RTS pins setup for each UART?

RobertCNelson, 2015-11-12:
@agherzan if you get it working, feel free to send a pull request to this repo.

agherzan, 2015-11-12:
So the answer is no. :) If we make it work we will definitely contribute back. Thanks.

leearmstrong, 2015-11-24:
@agherzan Did you manage to get the CTS/RTS enabled in the UART overlays?

agherzan, 2015-11-24:
@leearmstrong Hello. Yes we managed to emable CTS/RTS. Nikitas will come back to you with details.

nchronas, 2015-11-25:
Hello, 

In order for CTS/RTS to work, I used <a href="https://github.com/beagleboard/linux/blob/4.1/arch/arm/boot/dts/am335x-boneblack-emmc-overlay.dts" target="_blank">am335x-boneblack-emmc-overlay.dts</a> and <a href="https://github.com/beagleboard/devicetree-source/blob/d46bcc20ef98d69eea34cd6ae3897a329559ebb2/arch/arm/boot/dts/BB-UART4-RTSCTS-00A0.dts" target="_blank">BB-UART4-RTSCTS-00A0.dts</a>.

The tests were made by using a python script to manual set high/low the UART4 RTS pin.

dmr, 2016-04-13:
Hi @nchronas, I'm trying to get UART-4 to work with a current 4.1.18 kernel.
How to I apply a .dts file?

When I do
`sudo sh -c "echo 'BB-UART4' > /sys/devices/platform/bone_capemgr/slots"`
I always get
`sh: echo: I/O error`

RobertCNelson, 2016-04-13:
@dmr after you get:

```
sh: echo: I/O error
```

run:
```
dmesg | tail
```

dmr, 2016-04-14:
`dmesg | tail` tells me:

```
[   55.604839] bone_capemgr bone_capemgr: Using override eeprom data at slot 5
[   55.604855] bone_capemgr bone_capemgr: slot #5: 'Override Board Name,00A0,Override Manuf,BB-UART4'
[   66.810502] bone_capemgr bone_capemgr: part_number 'BB-UART1', version 'N/A'
[   66.810535] bone_capemgr bone_capemgr: slot #6: override
```

I used the latest image from this wiki page:
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Flasher:_.28console.29_.28BeagleBone_Black.2FGreen_eMMC.29

`https://rcn-ee.com/rootfs/bb.org/testing/2016-04-10/console/BBB-eMMC-flasher-debian-8.4-console-armhf-2016-04-10-2gb.img.xz`

with a BeagleBoneBlack

RobertCNelson, 2016-04-14:
@dmr 
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Adding_bb.org-overlays_to_jessie_console_image

Regards,

closed 2016-04-14 by RobertCNelson

dmr, 2016-04-14:
Thank you @RobertCNelson for the quick help!