beagleboard/bb.org-overlays #17: CBB-Serial-r01 not working

e1d1s1, 2015-12-04:
Has anyone had any luck with the CBB-Serial cape? It does not seem to be working in the latest 4.1.13-ti-r35 kernel. I can add the can0 device, and bring it up with ifconfig, but no data coming over candump. I have tried using the CBB-Serial-r01.dts, and the virtual cape DCAN1. Neither is working.  I have a Waveshare RS 485 cape that is connecting fine using the DCAN1, so the network is OK. Something weird about this cape.

e1d1s1, 2015-12-05:
I dug a little bit deeper, and it appears that the manufacturers DTS works fine, but the one in this overlay repo is incorrect. Can you replace cape-CBB-Serial-r01.dts with https://github.com/lgxlogic/CBB-Serial/blob/master/overlay/cape-CBB-Serial-r01.dts?

nnarain, 2016-07-07:
Hey, did you manage to get your CBB-Serial cape working. I'm having the issue you describe also

e1d1s1, 2016-07-07:
I believe if you use the DTS in the link I provided, it should work.