beagleboard/bb.org-overlays #22: trying to get spi fails no matter what

birecki, 2016-04-23:
I am trying to get spi to work on Beaglebone with 4.1.21-bone-rt-r20  (but this does not matter). I had it working fine with 3.8. No matter what I do  I get the spidev1.0, but the pin assignment fails they do not change mode to 0 nor set input output.. They remain default 0x37. 

My default in uEnv.txt is to disable universal cape and HDMI:

dtb=am335x-boneblack-emmc-overlay.dtb
cmdline=coherent_pool=1M quiet cape_universal=disable

after boot I can just be lazy and load the overlay dtb BBB-SPIDEV0, or....

Anyone can help? thanks
Henryk Birecki

RobertCNelson, 2016-04-23:
Ignore the pin mux assignments.  Just try talking to "/dev/spidev1.0", as long as "spidev" loaded properly "dmesg | grep bone"...

Regards,

birecki, 2016-04-23:
I wish I could. Unfortunately real devices require wired connections with real signals. If I cannot get signals to Beaglebone P9 pins, rest ofthe  hardware does not work. If one does not have real hardware one can check using spiloop code ( https://github.com/scottellis/spiloop ). Just short p9.18 to p9.21.

What could prevent pinmux from working? I do not see any of the SPI0 pins listed when listing pingroups.

Regards,
Henryk

RobertCNelson, 2016-04-23:
@birecki SPIDEV's been documented a few dozen times as working just fine.

https://groups.google.com/d/msg/beagleboard/CGmH0_YpJbQ/GXlYp0i6AQAJ

Re-check your hardware connections and use the kernel "spidev_test" 

https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/tools/spi

Closing: not a bug, spidev works fine...

closed 2016-04-23 by RobertCNelson

birecki, 2016-04-23:
Well.. Reading thebeagleboard thread you point at I do not see that it works. 

Hardware connections are as simple as shorting two pins on the header for test. Full hardware is independent of 
OS version and it works just fine under 3.8 version. SPIloop and sptdev_test are identical except for cosmetics.

I am not insisting there is a bug, but I would  appreciate some pointers as to why pinmux is possibly failing

On 22 Apr 2016 at 23:06, Robert Nelson wrote:

@birecki SPIDEV's been documented a few dozen times as working just fine.
https://groups.google.com/d/msg/beagleboard/CGmH0_YpJbQ/GXlYp0i6AQAJ
Re-check your hardware connections and use the kernel "spidev_test"
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/tools/spi
Closing: not a bug, spidev works fine...
-

RobertCNelson, 2016-04-23:
We know spidev works as an overlay when we need to communicate with something over spi, and we know in our v4.1.x that the "pinmux" values are still showing "0x37", even thou spidev is working.

The issue: the kernel isn't updating the "reported" pinmux values...

spidev works, there are other issues i'm working on that are way more important..

Regards,

birecki, 2016-04-25:
I do not know how you "know" that it works, but I assure you it does not.

I am sending you code for spiloop in a c file (the spidev test you pointed me is in the same file, but it has too many 
bugs to deal with), a picture showing test connections on a Beaglebone (SPI clock pin P9_22 is connected to 
scope), and a log of spiloop output on a working 3.8 OS device and output on 4.1 OS device, where it fails. 

On 3.8 device you see clock switching on oscilloscope, on 4.1 it it stuck high (3V). You do not need oscilloscope, a 
DC voltmeter will suffice. It will change value while spiloop is running.

I understand the pressure of important things to work on, but .... dismissing bug reports eventually comes to bite.

We'd like to use OS 4.1 because it is supposed to handle battery power management better than 3.8 (or so I am 
told), but we can live without it

Regards,
Henryk Birecki

On 23 Apr 2016 at 9:05, Robert Nelson wrote:

We know spidev works as an overlay when we need to communicate with something over 
spi, and we know in our v4.1.x that the "pinmux" values are still showing "0x37", even thou 
spidev is working.
The issue: the kernel isn't updating the "reported" pinmux values...
spidev works, there are other issues i'm working on that are way more important..
Regards,
-
*****====*****====*****====*****====*****====*****====*****
Dr. Henryk Birecki                    phone: 1-650-857-4597
Senior project scientist             
Hewlett Packard Laboratories
1501 Page Mill Rd.  MS1161
Palo Alto, CA 94304

birecki, 2016-04-25:
OOPS... sorry. I sent you the wrong picture. correct one is attached

Henryk

On 23 Apr 2016 at 9:05, Robert Nelson wrote:

We know spidev works as an overlay when we need to communicate with something over 
spi, and we know in our v4.1.x that the "pinmux" values are still showing "0x37", even thou 
spidev is working.
The issue: the kernel isn't updating the "reported" pinmux values...
spidev works, there are other issues i'm working on that are way more important..
Regards,
-
*****====*****====*****====*****====*****====*****====*****
Dr. Henryk Birecki                    phone: 1-650-857-4597
Senior project scientist             
Hewlett Packard Laboratories
1501 Page Mill Rd.  MS1161
Palo Alto, CA 94304

RobertCNelson, 2016-04-25:
@birecki see:
https://eewiki.net/display/linuxonarm/FLIR+Lepton+on+BeagleBone+Black+and+Green

This demo was shown at Embedded World in Germany two months ago..

I had setup two versions, using both, BB-SPIDEV0/BB-SPIDEV1

All communication was over spidev to the FLIR sensors...

Regards,

birecki, 2016-04-25:
Thanks,

I took just a quick glance on the web page I will look more carefully later. In the menatime :
It looks like

 a) you start with Debian 8.2 from 2015 while I start with bone-debian-8.3-lxqt-4gb-armhf-2016-01-24-4gb.img
I tried to download image mentioned on web page, but could not find it on the web

 b) you are using BB-SPIDEV1 (on the web page) not BB-SPIDEV0 which means you are using different pins for 
communication

I just verified SPIDEV1 by loading BB-SPIDEV1 overlay and testing on pins p9.28-31  Same result as previously 
except that pins were low by default rather than high. Attaching a resistor to any of the pins and connecting the 
other end to 3 or 0V changes voltage on the pin indicating that it behaves as input.

We are using SPI0 pins for now. We will need to use SPI0 and SPI1 in the future.

Any idea what could be the root cause of the difference?

Regards,
Henryk

On 25 Apr 2016 at 13:15, Robert Nelson wrote:

@birecki see:
https://eewiki.net/display/linuxonarm/FLIR+Lepton+on+BeagleBone+Black+and+Green
This demo was shown at Embedded World in Germany two months ago..
I had setup two versions, using both, BB-SPIDEV0/BB-SPIDEV1
All communication was over spidev to the FLIR sensors...
Regards,
-
*****====*****====*****====*****====*****====*****====*****
Dr. Henryk Birecki                    phone: 1-650-236-8168
Senior project scientist             
Hewlett Packard Laboratories
1501 Page Mill Rd.  MS1161
Palo Alto, CA 94304

birecki, 2016-04-25:
OK I found the problem. Solution is on the last line of the web page you pointed to.

It turns out that things work if one loads overlay from uEnv.txt file, but NOT if one tries to echo to .../slots

What is annoying is that some things work and others do not.

Thanks for responding and "toggling my mind"

Regards,
Henryk Birecki

On 25 Apr 2016 at 13:15, Robert Nelson wrote:

@birecki see:
https://eewiki.net/display/linuxonarm/FLIR+Lepton+on+BeagleBone+Black+and+Green
This demo was shown at Embedded World in Germany two months ago..
I had setup two versions, using both, BB-SPIDEV0/BB-SPIDEV1
All communication was over spidev to the FLIR sensors...
Regards,
-
*****====*****====*****====*****====*****====*****====*****
Dr. Henryk Birecki                    phone: 1-650-236-8168
Senior project scientist             
Hewlett Packard Laboratories
1501 Page Mill Rd.  MS1161
Palo Alto, CA 94304