beagleboard/bb.org-overlays #24: I/O Error

TigerClaw33, 2016-06-05:
Hello,
I am using the Debian-Jessie-8.4-2016-05-13-Standalone Image with a new sd card and also this repo: https://github.com/beagleboard/bb.org-overlays
```
root@beaglebone:# dtc --version
Version: DTC 1.4.1-g1e75ebc9
```

After executing ./install.sh I reboot my system. 

If i try this commando I get the following error:
```
root@beaglebone:/# echo BB-I2C1 > /sys/devices/platform/bone_capemgr/slots
bash: echo: write error: File exists
Maybe it's because the Kernel-Version?
```

After I tried this on:
```
root@beaglebone:/sys/devices/platform/bone_capemgr# sudo sh -c "echo 'BB-I2C1' > slots"
sh: echo: I/O error

root@beaglebone:/sys/devices/platform/bone_capemgr# cat  slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
```

If i delete cape_universal=enable from uEnv.txt, it works fine. But i don't know if this is the best solutions. I would be very pleased about an answer.

fivdi, 2016-06-05:
One solution is to disable the cape_universal overlay and then user your own overlay as you have already noticed. Another would be to leave cape_universal enabled and use the `config-pin` tool to configure I2C on the appropriate pins. It's difficult to say what's best.

TigerClaw33, 2016-06-06:
Thanks for your answer. What do you mean to use my own overlay? I only want to use the Standard BB-I2C1 overlay and it works only if I disable the cape_universal overlay and execute "sudo sh -c "echo 'BB-I2C1' > slots"". But this is not my own overlay.

fivdi, 2016-06-06:
Sorry, I didn't express myself very well. I meant the standard BB-I2C1 overlay.

TigerClaw33, 2016-06-06:
Thanks. I don't delete the universal capemgr line. I enabled the Cape via bone_capemgr.enable_partno=BB-I2C1 because I don't know how to u config-pin

fivdi, 2016-06-06:
The command `config-pin --help` prints information about using `config-pin`.

TigerClaw33, 2016-07-03:
bone_capemgr.enable_partno=BB-I2C1 works. Now I want to use a second device on i2c2 P9_19 and P9_20. I know this pins are also using by capemgr. I tried to load DTO BB-I2C2 an get the echo: I/O error . bone_capemgr.enable_partno=BB-I2C2 also didn't work at boot/uEnv.txt. What can I do?