beagleboard/bb.org-overlays #25: Fail to load eqep driver?

yoneken, 2016-06-07:
When I execute the following command,

> $ sudo sh -c "echo 'bone_eqep0' > /sys/devices/platform/bone_capemgr/slots"

_/sys/devices/platform/ocp/48300000.epwmss/48300180.eqep/_ is appeared.
But there is no function directory (like "eqep") and also "driver" symbolic link.

> $ ls /sys/devices/platform/ocp/48300000.epwmss/48300180.eqep/
> driver_override  modalias  of_node  power  subsystem  uevent

I expect there is a function directory which contains "enable", "mode", "position" ...

My environment is Ubuntu 14.04.

> $ uname -a
> Linux beaglebone 4.4.9-ti-r25 #1 SMP Fri May 6 10:28:56 UTC 2016 armv7l armv7l armv7l GNU/Linux

**tieqep** module is seems to be loaded.
> $ lsmod
> Module                  Size  Used by
> pwm_tiehrpwm            4706  1
> tieqep                  8739  0
> pruss_intc              3592  0
> 8021q                  17828  0
> garp                    5769  1 8021q
> mrp                     7175  1 8021q
> stp                     2219  1 garp
> llc                     5123  2 stp,garp
> usb_f_ecm               9400  1
> g_ether                 4976  0
> usb_f_rndis            22191  2 g_ether
> u_ether                11917  3 usb_f_ecm,usb_f_rndis,g_ether
> libcomposite           43461  3 usb_f_ecm,usb_f_rndis,g_ether
> snd_soc_simple_card     8385  0
> pru_rproc              11295  0
> omap_aes_driver        18917  0
> omap_sham              21629  0
> pruss                  11615  1 pru_rproc
> omap_rng                4423  0
> rng_core                7471  1 omap_rng
> snd_soc_davinci_mcasp    16596  2
> snd_soc_edma            1290  1 snd_soc_davinci_mcasp
> snd_soc_omap            3058  1 snd_soc_davinci_mcasp
> spi_omap2_mcspi        10764  0
> snd_soc_hdmi_codec      5842  1
> snd_soc_core          155592  5 snd_soc_hdmi_codec,snd_soc_davinci_mcasp,snd_soc_edma,snd_soc_omap,snd_soc_simple_card
> snd_pcm_dmaengine       5209  2 snd_soc_core,snd_soc_omap
> snd_pcm                82689  5 snd_soc_hdmi_codec,snd_soc_davinci_mcasp,snd_soc_core,snd_soc_omap,snd_pcm_dmaengine
> snd_timer              19720  1 snd_pcm
> snd                    59559  3 snd_soc_core,snd_timer,snd_pcm
> soundcore               7601  1 snd
> evdev                  10695  2
> uio_pdrv_genirq         3539  0
> uio                     8822  1 uio_pdrv_genirq
> bluetooth             417817  2
> rfkill                 18226  3 bluetooth
> tilcdc                 26364  0
> tda998x                12754  0

There is a driver directory for eqep.
> $ ls /sys/bus/platform/drivers/eqep/
> bind  module  uevent  unbind

dmesg shows some errors. But I cannot understand it..

> [ 3804.958336] bone_capemgr bone_capemgr: part_number 'bone_eqep0', version 'N/A'
> [ 3804.958411] bone_capemgr bone_capemgr: slot #6: override
> [ 3804.958454] bone_capemgr bone_capemgr: Using override eeprom data at slot 6
> [ 3804.958500] bone_capemgr bone_capemgr: slot #6: 'Override Board Name,00A0,Override Manuf,bone_eqep0'
> [ 3804.972553] platform 48300180.eqep: Cannot lookup hwmod 'eqep0'
> [ 3804.992710] eqep 48300180.eqep: could not find pctldev for node /ocp/interrupt-controller@48200000, deferring probe
> [ 3804.993069] bone_capemgr bone_capemgr: slot #6: dtbo 'bone_eqep0-00A0.dtbo' loaded; overlay id #2

Bobox214, 2016-06-14:
I am encountering the same issue trying to load the eqep overlays.
Same error in dmesg : 
`could not find pctldev for node /ocp/interrupt-controller@48200000, deferring probe`

`Linux arm 4.4.12-bone-rt-r11 #1 PREEMPT RT Thu Jun 2 11:42:46 UTC 2016 armv7l armv7l armv7l GNU/Linux
`

RobertCNelson, 2016-06-14:
For eqep, drop back down to v4.1.x, the driver needs some work for v4.4.x kernel's..

crispc85, 2016-07-29:
I have run into the same problem with the eqep driver using kernel the 4.4.x kernel in the debian Jessie image. I've done a bit of googling trying to find out how to change the kernel to the 4.1.x version, but have come up short. Does anyone know of a good site with instructions on how to do this? Thanks!

crispc85, 2016-08-03:
Hi,

So I was able to get the kernel changed to version 4.1.28-ti-r65 and the eQEP drivers seem to load correctly according to the response from `dmesg | grep eqep`. I am using the bone_eqep2-00A0.dtbo device tree overlay that came preloaded on the Debian 8.4 Jessie image in the /lib/firmware directory. When I hook up my encoder A and B channels to pins 8.41 and 8.42, which are listed as the A and B input pins for eQEP2, only one of the pins is receiving pulses from the encoder. I know both channels on the encoder work since I hooked it up to an oscilloscope to ensure that wasn't the problem, and when I switched the channels on the pins the results were the same, only getting pulses on pin 8.41, nothing from 8.42. I have tried eQEP0, eQEP1, eQEP2, and eQEP2B all with the same results. It seems that only the pulses from one of the channels are being captured.

Any ideas what is going on or how to fix this? Any help would be greatly appreciated! Thanks!

crispc85, 2016-08-03:
Here is my `dmesg | grep eqep` response if it helps:

`[   11.863104] bone_capemgr bone_capemgr: enabled_partno PARTNO 'bone_eqep2' VER 'N/A' PR '0'
[   11.863137] bone_capemgr bone_capemgr: slot #7: 'Override Board Name,00A0,Override Manuf,bone_eqep2'
[   11.880996] bone_capemgr bone_capemgr: slot #7: dtbo 'bone_eqep2-00A0.dtbo' loaded; overlay id #3
[   19.933548] eqep 48304180.eqep: ver. 1.0`