beagleboard/bb.org-overlays #28: FATAL ERROR

Olfox59, 2016-09-01:
Hi,

There is now more than one week i search everywhere the answer to this simple error when i try to compil with this line :

```
root@beaglebone:~/bb.org-overlays/src/arm# dtc -O dtb -o BB-BONE-LCD7-01-00A3.dtb BB-BONE-LCD7-01-00A3.dts

Error: BB-BONE-LCD7-01-00A3.dts:13.1-9 syntax error
FATAL ERROR: Unable to parse input tree
```

It seems that is line is the problem.
_#include <dt-bindings/board/am335x-bbw-bbb-base.h>_

I'm sur of it cause i move it on line 15 and problem point on line 15.

I don't understand why i can't just compile a simple scratch BB-BONE-LCD7-01-00A3.dts

I already reinstalled from scratch all my BBB with the latest version 4.4.9.
I followed all the steps you mentioned even if i'm not sure of what it does exactly after install.sh concerning kernel?
i try to use this version : 

_v4.4.x mainline:
sudo ./update_kernel.sh --lts-4_4 --bone-channel_

There was some error and it complain about the fact that this version is for 4.4.19 and i'm only 4.4.9. So i don't know if this step is required but it seems to finished completly.
Also the dt-bindings/board/am335x-bbw-bbb-base.h exist in the good folder.

My lcd works fine by unloading hdmi and loading this LCD device tree. I need to modify it only for the touchscreen cause it uses in the dts channel 4 5 6 7 but channel 7 on BBB is wired to a 1.65V reference voltage.

I have seen in other men that had the same issue, but the answer on how to solve it was not clear ( just said to check manual...).

Thank you very much for your precious help cause in my previous 3.8 version and 4.0 dtc, i had no issue.

Florian :)

RobertCNelson, 2016-09-01:
You need to call the gcc pre-processor first to take care of the #include's..  Or just call "make" from the bb.org-overlays base directory, which takes care of it.

Please attach your terminal log error from running:

```
sudo ./update_kernel.sh --lts-4_4 --bone-channel
```

Regards,

RobertCNelson, 2016-09-01:
Or even
```
voodoo@hades:/opt/github/bb.org/bb.org-overlays$ make src/arm/BB-BONE-LCD7-01-00A3.dtbo
  DTC     src/arm/BB-BONE-LCD7-01-00A3.dtbo
```

Regards,

closed 2016-09-01 by RobertCNelson

Olfox59, 2016-09-01:
Thanks for the fast answer Robert, i join you the log.

[error with kernel.sh.TXT](https://github.com/beagleboard/bb.org-overlays/files/450705/error.with.kernel.sh.TXT)

Concerning the build, i make like this : 

`root@beaglebone:~/bb.org-overlays# make -v
GNU Make 4.0
Built for arm-unknown-linux-gnueabihf
Copyright (C) 1988-2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
`

After i try t build the dts , but same result as before:

root@beaglebone:~/bb.org-overlays/src/arm# dtc -O dtb -o BB-BONE-LCD7-01-00A3.dtb BB-BONE-LCD7-01-00A3.dts
Error: BB-BONE-LCD7-01-00A3.dts:11.1-9 syntax error
FATAL ERROR: Unable to parse input tree

If i do it with the make, it works !

`root@beaglebone:~/bb.org-overlays# make

  DTC     src/arm/BB-BONE-LCD7-01-00A4.dtbo
`

versio A4 is only a copy of a3 without changing except the version field into the file : 

```
        /* identification */
        part-number = "BB-BONE-LCD7-01";

        version = "00A4";
```
I copy to lib/firmware by using install.sh (even if i think its not fully necessary, a cp command should be enough).

Now i change the /boot/uEnv.txt to launch it and disable hdmi also:

```
##BeagleBone Black: HDMI (Audio/Video) disabled:
dtb=am335x-boneblack-emmc-overlay.dtb

##Fle add
#cape_disable=bone_capemgr.disable_partno=
cape_enable=bone_capemgr.enable_partno=BB-BONE-LCD7-01:00A3
```

i have seen by uname -a that now i'm in kernel 4.4.19 . In the boot uEnv.txt its also mentionned 

`uname_r=4.4.19-bone13`

So now save and reboot

It works ! perfect.

Now i will try to change the ADC channel: 

```
                        adc {
                                /*ti,adc-channels = <4 5 6 7>;*/
                                ti,adc-channels = <0 1 2 3>;
                        };
```

save, make, install, reboot

After boot the calibration software run, but it seems that touchscreen is not correctly set cause the curser moves in some strange way and come back at initial place after ... I don't know really how to make the link between channel in dts and the wiring .

Thank you so much Robert for your help, i was really deep in the sadness this evening ... It would be the paradise if i succeed to make my touchscreen works but let see next step . If you have any clue , i take it :) 

Bye from France :) :+1:

Olfox59, 2016-09-02:
Note : in uEnv it is well : 

```
##BeagleBone Black: HDMI (Audio/Video) disabled:
dtb=am335x-boneblack-emmc-overlay.dtb

##Fle add
#cape_disable=bone_capemgr.disable_partno=
cape_enable=bone_capemgr.enable_partno=BB-BONE-LCD7-01:00A4
```

and not A3 like i mentionned i my previous post

I also have seen some discussion about a patch for this TSC driver implemented by Micka that you integrated from 3.8 kernel version.

So i guess i should have it in 4.4.19 kernel version.

So may be i skip something but how is it possible to use in dts channel 7 ? Cuase if we see the schematics ch7 is wired to a resistor divider that create a voltage reference of 1.65V:

[image]