beagleboard/bb.org-overlays #32: Could not find symbol 'gpio4'

Biffco, 2016-09-30:
kernel version - uname -r: 4.1.33-bone24
dtc --version:  Version: DTC 1.4.1-g1e75ebc9

After running ./install.sh in bb.org-overlays:
# echo cape-universala > /sys/devices/platform/bone_capemgr/slots
[  259.444391] bone_capemgr bone_capemgr: part_number 'cape-universala', version 'N/A'
[  259.452290] bone_capemgr bone_capemgr: slot #6: override
[  259.457733] bone_capemgr bone_capemgr: Using override eeprom data at slot 6
[  259.464803] bone_capemgr bone_capemgr: slot #6: 'Override Board Name,00A0,Override Manuf,cape-universala'
[  259.546047] of_resolve_phandles: Could not find symbol 'gpio4'
[  259.551975] bone_capemgr bone_capemgr: slot #6: Failed to resolve tree
bash: echo: write error: Invalid argument

# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-univ-hdmi

cape-universaln uses gpio0-3 while cape-universala uses gpio1-4.   Config-pin needs cape-universala to run properly on my system.

Suggestions?  Thanks.

RobertCNelson, 2016-10-01:
@Biffco thanks for the report, it looks like i forgot to x=x-1 when i added cape-universala, it should be fixed on master now..

Regards,

closed 2016-10-01 by RobertCNelson

Biffco, 2016-10-01:
Yes it is. Thanks!