beagleboard/bb.org-overlays #38: Can't add device overlay

Rebel660, 2017-02-17:
Hi,
I've follwed your instructions in readme, everything works till I cat slots.

I don't get empty slots as shown I have universaln in slot 4.

If I try to add anything it reports an I/O error, I can't remove the slot (issue 13) so I comment out universal in uEnv.txt, the BBB just hangs and won't boot.

I have an LCD cape (BB-CAPE-DISP-CT4-00A0) that works by adding the hardware and booting, but I want to mess about with the LCD settings to drive another LCD.

I changed the source for the cape, finally managed to compile it to dtbo using make instead of dtc, I compared the dtbo with the original, it was different but nothing changes in reality hence I'm trying to enable the overlay manually.

I'm new to this, I bought Derek Molloy's book and have worked through it to chapter 6 so I'm not a complete duffer but I seem to have got to a step where my kernel version doesn't correspond with any of the advice given and everything I try gets me nearly but not quite there. I'm using 4.4.45 but followed your upgrade to 4.4.49. Sorry if I'm asking stupid questions, I'm at my wits end trying to do something that I reckon should be simple.

Would appreciate your help.

RobertCNelson, 2017-02-17:
@Rebel660 remove cape-universal as shown:

http://elinux.org/Beagleboard:BeagleBone_Debian_Image_Migration#cape-universal

Regards,

Rebel660, 2017-02-17:
I tried this:
  cmdline=coherent_pool=1M quiet cape_universal=enable
to
  cmdline=coherent_pool=1M quiet
and every permutation I could think of, the board won't reboot, the 4 LED's come on solid and that's it, sometimes they go off and back on momentarily but nothing happens, I can understand that without an overlay the I/O might not behave but it never boots and the only way I can get it back to life it reflash it again.

I'll try the other methods now as I'm desperate to get past this and get some work done.

Thanks for helping.

RobertCNelson, 2017-02-17:
That one change won't brick the boot..

What editor are you using?

Regards,

Rebel660, 2017-02-17:
FireSSH, Putty, Dreamweaver, vi.
Image Debian 8.6 2016-11-06 4GB SD LXQT flashed onto MMC
I then follow your readme file and use 4.4 mainline.

RobertCNelson, 2017-02-17:
As a test, use nano..

Regards,

Rebel660, 2017-02-17:
Ok, with nano I removed the universaln from uEnv.txt
This time 3 led's lit and it's stuck there. It's different which is slightly concerning!

RobertCNelson, 2017-02-17:
Do you have a usb serial adapter to log/debug the boot process, so we can see where u-boot get's stuck?

Regards,

Rebel660, 2017-02-17:
No, sorry I don't and I can only get access via ethernet, the USB drivers would not install on any machine that I can find. 
Maybe I should try an older image?

RobertCNelson, 2017-02-17:
New windows driver were posted a few weeks back:

https://github.com/beagleboard/beaglebone-getting-started/tree/master/Drivers

BUT, that serial port only get's you access to a getty login..

For the debug/log i'm asking for.. You need a usb-serial adapter to plug in the debug connector, so we can monitor the u-boot booting process.

You can try an older or newer image:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Image_Testing_Snapshots

Regards,

Rebel660, 2017-02-17:
End to end, (In case I'm being stupid) I do:
Get image from here http://beagleboard.org/latest-images
This one  Debian 8.6 2016-11-06 4GB SD LXQT
Using image writer put this on an SD card
I then follow this https://github.com/beagleboard/bb.org-overlays/blob/master/readme.md and nothing else doing a reboot after each update
Then with vi/nano/dreamweaver I change this:
cmdline=coherent_pool=1M quiet cape_universal=enable
to
cmdline=coherent_pool=1M quiet
leaving line breaks the same etc.
Then reboot (via the terminal and it's dead).

Rebel660, 2017-02-17:
Will try those drivers, I will order a usb-serial adapter as you suggest. In the meantime I guess I'll try 4.1 that would seem a sensible choice.

RobertCNelson, 2017-02-17:
btw, steps 1->3 are already done in the base image.. If you do apt-get update ; apt-get upgrade, it'll pull in the latest version of this repo already pre-built..

RobertCNelson, 2017-02-17:
I rewrote the readme:

https://github.com/beagleboard/bb.org-overlays/blob/master/readme.md

and moved the dtc/overlay building under developers, most end users won't need to deal with that anymore as everything is pre-installed/packaged now..

Regards,

Rebel660, 2017-02-17:
I'm going to try that approach while I wait for 4.1 to download, I think I've already done that but I'll try anything to get past this.
Many thanks

Rebel660, 2017-02-17:
I reflashed the board, did apt-get update ; apt-get upgrade removed the universaln overlay, reboot, identical result 4 LED's on solid and the board is non-responsive. I've ordered the serial to usb cable. In the meantime I'll try an older version of the image.
Thanks again

Rebel660, 2017-02-18:
I have reflashed the board again with v4.1.15 exactly the same result. I'll have to wait for the serial cable now and see what is blocking the boot. I can't think of anything else to try, any ideas?
Thanks

Rebel660, 2017-03-01:
Success!!
I have finally succeeded in removing the universal overlay and attaching another.

I reflashed the latest version 8.6 (4.30.0) and updated it, this time it worked. The only difference? I unplugged my usb mouse. Doh!!

The only thing I would say is it took two tries, first time it failed again because the mouse was still plugged in when I flashed the MMC, when I tried to reboot with the DTO removed from uEnv.txt it still failed, I had to flash the MMC with the mouse removed but the ethernet connected.

Thanks for your help.