beagleboard/bb.org-overlays #39: Beaglebone Green Wireless I2S Audio

tderensis, 2017-03-02:
I am trying to get audio to output on the I2S port with a Beaglebone Green Wireless. I have installed the Debian 8.6 2016-11-06 4GB SD SeeedStudio IoT image and run apt-get update/upgrade. I have also updated to the latest 4.9 kernel using `./update_kernel.sh --ti-channel --lts-4_9`. The output of `uname -a` is 4.9.10-ti-r23. I have noticed that some overlays will load fine, such as univ-bbgw. With the univ-bbgw, aplay -L shows default, null, and pulse. I was expecting to see default:CARD=black. Running `speaker-test` does not show any output on p28, which I believe to be the audio output pin for I2S. Note that I don't actually have an I2S device hooked up at the moment. This is just a test before I purchase a device.

After that I tried to load BB-BONE-AUDI-02 in which case I get an error
`[54270.254981] OF: resolver: of_resolve_phandles: Could not find symbol 'clk_mcasp0'`
`[54270.263050] bone_capemgr bone_capemgr: slot #6: Failed to resolve tree`

Is there a different cape I should be loading to get audio output?

RobertCNelson, 2017-03-02:
@tderensis you are not going to get any output over i2s, until you load an audio overlay to enable those pins/function/peripheral.  Note the BBGW has the i2s already connected to the bluetooth audio port of the wl1835.  Note bluetooth audio on the bbgw is currently disable due to a design defect:

https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/am335x-bonegreen-wireless.dts#n111

Regards,

tderensis, 2017-03-02:
Hi Robert, thanks for the quick response. Yea, I noticed that the i2s is connected to the bluetooth audio port of the wl1835. I was hoping to ignore that and use an external i2s audio amplifier instead of the bluetooth audio. Not sure if I would need any hardware changes to disconnect the bluetooth audio. 

Is there a specific cape I could look at which enables the i2s audio? I could use it or modify it for my own use. Or is there absolutely no way i2s audio will work due to this design defect?

tderensis, 2017-03-17:
Looking at the schematic and the dts file you linked, I see that the pin with the issue is the I2S data input. For my application I don't need audio input, so I was thinking I could still salvage the I2S port. I tried to add to the existing am335x-bonegreen-wireless file in order to get audio, but my compiled dtb is not working as expected. It loads correctly with no errors in dmesg, but I still don't see the audio device with aplay -L. Note that I am using a MAX98357a, which appears to have a driver. Below is my additions to the dts. Let me know if I should be posting this in the google group instead and if you need more information.

```
&am33xx_pinmux {
    mcasp0_pins: mcasp0_pins {
		pinctrl-single,pins = <
			AM33XX_IOPAD(0x9ac, PIN_INPUT_PULLUP | MUX_MODE0)   /* mcasp0_ahcklx.mcasp0_ahclkx */
			AM33XX_IOPAD(0x99c, PIN_OUTPUT_PULLDOWN | MUX_MODE2) /* mcasp0_ahclkr.mcasp0_axr2*/
			AM33XX_IOPAD(0x994, PIN_OUTPUT_PULLUP | MUX_MODE0) /* mcasp0_fsx.mcasp0_fsx */
			AM33XX_IOPAD(0x990, PIN_OUTPUT_PULLDOWN | MUX_MODE0) /* mcasp0_aclkx.mcasp0_aclkx */
			AM33XX_IOPAD(0x86c, PIN_OUTPUT_PULLDOWN | MUX_MODE7) /* gpmc_a11.GPIO1_27 */
		>;
	};
};

....

&mcasp0	{
	#sound-dai-cells = <0>;
	pinctrl-names = "default";
	pinctrl-0 = <&mcasp0_pins>;
	status = "okay";
	op-mode = <0>;	/* MCASP_IIS_MODE */
	tdm-slots = <2>;
	serial-dir = <	/* 0: INACTIVE, 1: TX, 2: RX */
			0 0 1 0
		>;
	tx-num-evt = <32>;
	rx-num-evt = <32>;
};

/ {
	clk_mcasp0_fixed: clk_mcasp0_fixed {
		#clock-cells = <0>;
		compatible = "fixed-clock";
		clock-frequency = <24576000>;
	};

	clk_mcasp0: clk_mcasp0 {
		#clock-cells = <0>;
		compatible = "gpio-gate-clock";
		clocks = <&clk_mcasp0_fixed>;
		enable-gpios = <&gpio1 27 0>; /* BeagleBone Black Clk enable on GPIO1_27 */
	};

	max98357a: max98357a {
		#sound-dai-cells = <0>;
		status = "okay";

		pinctrl-names = "default";
		pinctrl-0 = <&mcasp0_pins>;

		compatible = "maxim,max98357a";
	};

	sound {
		compatible = "simple-audio-card";
		simple-audio-card,name = "TI BeagleBone Black";
		simple-audio-card,format = "i2s";
		simple-audio-card,bitclock-master = <&dailink0_master>;
		simple-audio-card,frame-master = <&dailink0_master>;

		dailink0_master: simple-audio-card,cpu {
			sound-dai = <&mcasp0>;
			clocks = <&clk_mcasp0>;
		};

		simple-audio-card,codec {
			sound-dai = <&max98357a>;
		};
	};
};
```

tderensis, 2017-03-23:
I'm sorry to keep bothering you, but I would really like to get I2S audio working in some fashion. I notice that the MAX98357a kernel module is actually not built on the 4.4.x kernel. It can't be selected when running "make menuconfig" either. Is it possible to use the existing tda998x module or another module even without the I2C hooked up (the MAX98357a doesn't have I2C control). 

I tried this approach on a Beaglebone Black and I can only get output when the HDMI cable is actually plugged in. I have not tried the BBGW with the tda998x module yet.

RobertCNelson, 2017-03-23:
@tderensis it's probally the clock.  It's one of the cases where kernel overlay's run into fun issues in v4.1.x+ kernel's.. Switch to u-boot overlays:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

Regards

jadonk, 2017-04-18:
Why do you need I2S audio anyway? Talking to things like Bluetooth speakers doesn't use it.

tderensis, 2017-04-18:
I'm not talking to Bluetooth speakers. I need to talk to an I2S amplifier off board. I was hoping to use that module, but it looks like the MAX98357a module is not built for the beaglebone. I wonder if there is a really simple I2S driver that is built in (no I2C interface) that I can use.

tderensis, 2017-04-19:
I got audio working. There were a few pieces I needed to add to my dts file.

Add the mcasp0 pins:
```
&am33xx_pinmux {
	mcasp0_pins: mcasp0_pins {
		pinctrl-single,pins = <
			AM33XX_IOPAD(0x9ac, PIN_INPUT_PULLUP | MUX_MODE0)   /* mcasp0_ahcklx.mcasp0_ahclkx */
			AM33XX_IOPAD(0x99c, PIN_OUTPUT_PULLDOWN | MUX_MODE2) /* mcasp0_ahclkr.mcasp0_axr2*/
			AM33XX_IOPAD(0x994, PIN_OUTPUT_PULLUP | MUX_MODE0) /* mcasp0_fsx.mcasp0_fsx */
			AM33XX_IOPAD(0x990, PIN_OUTPUT_PULLDOWN | MUX_MODE0) /* mcasp0_aclkx.mcasp0_aclkx */
			AM33XX_IOPAD(0x86c, PIN_OUTPUT_PULLDOWN | MUX_MODE7) /* gpmc_a11.GPIO1_27 */
		>;
	};
};
```
Add mcasp0 (same as when using HDMI audio)
```
&mcasp0	{
	#sound-dai-cells = <0>;
	pinctrl-names = "default";
	pinctrl-0 = <&mcasp0_pins>;
	status = "okay";
	op-mode = <0>;	/* MCASP_IIS_MODE */
	tdm-slots = <2>;
	serial-dir = <	/* 0: INACTIVE, 1: TX, 2: RX */
			0 0 1 0
		>;
	tx-num-evt = <32>;
	rx-num-evt = <32>;
};
```
Add simple sound card and max98357a codec. Note that your SD Mode pin may be connected to a different pin.
```
/ {
	clk_mcasp0_fixed: clk_mcasp0_fixed {
		#clock-cells = <0>;
		compatible = "fixed-clock";
		clock-frequency = <24576000>;
	};

	clk_mcasp0: clk_mcasp0 {
		#clock-cells = <0>;
		compatible = "gpio-gate-clock";
		clocks = <&clk_mcasp0_fixed>;
		enable-gpios = <&gpio1 27 GPIO_ACTIVE_HIGH>; /* BeagleBone Black Clk enable on GPIO1_27 */
	};

	max98357a: max98357a {
		#sound-dai-cells = <0>;
		status = "okay";

		compatible = "maxim,max98357a";
		sdmode-gpios = <&gpio1 14 0>;
	};

	sound {
		compatible = "simple-audio-card";
		status = "okay";
		simple-audio-card,name = "TI BeagleBone Green Wireless";
		simple-audio-card,format = "i2s";
		simple-audio-card,bitclock-master = <&dailink0_master>;
		simple-audio-card,frame-master = <&dailink0_master>;

		dailink0_master: simple-audio-card,cpu {
			sound-dai = <&mcasp0>;
			clocks = <&clk_mcasp0>;
		};

		simple-audio-card,codec {
			sound-dai = <&max98357a 0>;
		};
	};
};
```
Next I had to compile the max98357a module, which can be obtained from the linux kernel
https://raw.githubusercontent.com/torvalds/linux/5691f0e9a3e7855832d5fd094801bf600347c2d0/sound/soc/codecs/max98357a.c

Install kernel headers:
`sudo apt-get install linux-headers-$(uname -r)`

The makefile I used was:
```
obj-m := max98357a.o
KDIR := /lib/modules/$(shell uname -r)/build
PWD := $(shell pwd)

all:
	make -C $(KDIR) M=$(PWD) modules

clean:
	make -C $(KDIR) M=$(PWD) clean

install:
	sudo cp max98357a.ko /lib/modules/$(shell uname -r)/extra/
	sudo depmod -a
```

Now `aplay -L` shows:
```
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=Wireless
    TI BeagleBone Green Wireless,
    Default Audio Device
sysdefault:CARD=Wireless
    TI BeagleBone Green Wireless,
    Default Audio Device
dmix:CARD=Wireless,DEV=0
    TI BeagleBone Green Wireless,
    Direct sample mixing device
dsnoop:CARD=Wireless,DEV=0
    TI BeagleBone Green Wireless,
    Direct sample snooping device
hw:CARD=Wireless,DEV=0
    TI BeagleBone Green Wireless,
    Direct hardware device without any conversions
plughw:CARD=Wireless,DEV=0
    TI BeagleBone Green Wireless,
    Hardware device with all software conversions
```

The only thing missing is volume control, which doesn't seem to be configurable through `alsamixer`.

ncobbett, 2017-04-25:
tderensis,

I am trying to get an old design running on the BBGW that use to run on the BBB.  The major feature I use is I2S to stream 96KHz recorded from a VLF radio receiver.  I'm having issues getting this working.  

I use the TLV320AIC3106 codec and the BB-BONE-AUDI-01-00A0.dts (and disabled the hdmi audio in uEnv.txt).  It worked fine on the BBB with decent 96KHz recorded audio.  I tried yesterday to get this running on a BBGW but I was having issues loading the BB-BONE-AUDI device tree overlay.

Did you make any further progress with I2S?  Will it ever with Bluetooth module installed - will ir interfere with general I2S?

I'm looking to repass the PCB I designed and I am looking at what would work best - sticking with BBB, BBGW or even RPI.

I have had a few issues with device tree loading in the past.  It always seems over complicated.  Also it appears to have evolved since I last used it.  Is it better to load the device trees in Uboot and not bother with doing it at run time?

tderensis, 2017-04-26:
Is the chip that uses the TLV320AIC3106 codec off board on a cape? The BBGW does not have an hdmi audio codec on it. I was able to get I2S audio working as I described above, but I instead put everything in my own dtb file.

tderensis, 2017-04-26:
For all those curious I got volume control through alsa by using the the following asound.conf file to create a softvol control. The codec I am using does not have hardware volume control.
```
pcm.!default {
	type		plug
	slave.pcm	"softvol"
}

pcm.softvol {
	type		softvol
	slave {
		pcm	"dmix"
	}
	control {
		name	"Softmaster"
		card	0
	}

	max_dB		20.0
	min_dB		-51.0
}
```

ncobbett, 2017-05-04:
Yes, the chip is on a cape board I designed myself.  I now have it in the device tree.  the dts file was complaining with a conflict on P9.25 with the BBGW device tree using it for something else (~ possibly PRU).

I can now see the device in alsamixer but I'm getting the following error using arecord...

oot@beaglebone:~# alsamixer
root@beaglebone:~# arecord -D hw:EVM -c 2 -r 96000 -f S32_LE -d 10 bbgw1.wav
Recording WAVE 'bbgw1.wav' : Signed 32 bit Little Endian, Rate 96000 Hz, Stereo
arecord: pcm_read:2031: read error: Input/output error

tderensis, 2017-05-04:
Do you see the device with `aplay -L` ?

Try playing from the default device using speaker-test. `speaker-test -c 2 -t sine -f 440`

RiteshSingh323, 2017-12-14:
Hi tderensis,
I am also badly stuck for registering my tlv320dac3100 codec as audio device on beaglebone green wireless. I want to enable McAsp0(on P8 headers) as I2S and I2C1 as control interface.

I am able to see I2C1 by enabling bone-capemgr  in uEnv.txt but failed to load my dtbo of BB-BONE-AUDI-02 while saw the dmseg.

are you using same dts file to modified according to requirement or anyother. please help me to resolve this as i am working on this since last two weeks but unable to get success.
any help would be appreciated.

thanks

ncobbett, 2017-12-16:
Hi RiteshSingh323
I got it to eventually work.  There are a few things you need to get right:
Kernel 3.x you need to load the dtbo file after boot in /etc/defaults/capemgrKernel 4.x you can do it in /boot/uEnv.txt but you need to set a few things correctly in this file or it will not work.  Principally disable hdmi sound but also you need to enable a few other things for it to pick up the sound card.  I will try to send this on to you on Monday when I am back in work.
Also, I found Audi2 dts file gave issues for me but Audi1 dts file worked fine.
AND using Alsamixer if you set/unset certain mixers, it causes the soundcard to crash.
It's not easy to make headway.  I've spent many months working with my own cape design incorporating this 96KHz sound codec.

The best way to debug is to use the ttyO0 header pins with a USB to serial connector and minicom.  A lot of the device tree mappings happens prior to kernel boot so you wont see what was loaded and what went wrong without the header attached.
Neil

RiteshSingh323, 2017-12-18:
Thanks ncobbett looking into it.
Please send that would be really helpful for me. 
I am using Linux arm 4.4.30-ti-r66 kernel and my hardware connections are:-
P8.37 - mcasp0-aclkx
P8.38 - mcasp0-fsx
P8.34 - mcasp0-axr2
P8.32 - mcasp0-ahclkx
P8.36 - mcasp0-acr0
for I2S audio and I2C1 for Control interface.
I am getting error -   failed to load .dtbo file in slots and  
[  103.180866] of_resolve_phandles: Could not find symbol 'clk_mcasp0'
[  103.187269] bone_capemgr bone_capemgr: slot #4: Failed to resolve tree 

unable to solve this issue.
please provide your .dts file if possible and steps how can i able to register my codec and show using
aplay -L on terminal.

Thanks

ncobbett, 2017-12-21:
Hi Ritesh,
Sorry for the delay in replying...
The first thing to check is that the device tree blob *.dtbo is loaded properly.  You can see this with the serial output on the header using minicom or similar and a serial to usb converter lead.  Check for errors on boot.  If this isn't being loaded properly then everything else wont work.  You definitely need to disable the HDMI sound.  The "disable_uboot_overlay_video=1"  disables in my uEnv.txt disables video AND sound.  However, if you still want to use the video and not run headless like I do, then use "disable_uboot_overlay_audio=1" instead.  (I haven't checked this but it should work!)
The next step is to ensure the audio DTS file is correct.  I have developed my own cape board utilising the TI 96KHz codec and used the version 1 Audio cape as a starting point as part of my design.  There is a version 2 of this board now that uses a slightly different DTS file.  This doesn't appear to work with my hardware but I have not looked into the detail to see why,   I have attached the AUDI_01 dts to this email for you to try.  Simply compile and insert it in /lib/firmware so it can be found by uEnv.txt.
Once it is loaded then install Alsa tools.  You need at at least aplay, arecord and alsamixer.I have found the alsamixer to be a little fiddly.  If you set or disable some of the switches it causes the sound system to complain.  There's dozens of settings so this can take some time to work through.  If this has just one switches wrong it will cause also to error.
Thse commands should work if the sound card is installed correctly and the dtbo is working:
alsamixerCard: DA830 EVM
arecord -L
This will identify the hardware device name

arecord -D hw:EVM -f S16_LE -c2 -r96000 -t wav -d 10 test.wav
This will record 10 seconds of 2 channel audio data at 96KHz.

I do not play audio on my device just record.  aplay should however work if arecord works!  I transfer the sound files to my desktop and analyse them using audacity and spectrogram mode.
Let me know how far you get with this.... I have been fiddling with audio on the Beaglebone for a while with this chip.  It took me some time to get this far by trial and error.
All the bestNeil

using 4.4 kernel - this is my uEnv.txt file:
****************
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0 
uname_r=4.4.91-ti-r133
#uuid=
#dtb=

###U-Boot Overlays###
###Master Enable
enable_uboot_overlays=1

cmdline=coherent_pool=1M net.ifnames=0 quiet
#cmdline=quiet  

###Disable auto loading of virtual capes (emmc/video/wireless/adc)  
#disable_uboot_overlay_emmc=1  
disable_uboot_overlay_video=1  
#disable_uboot_overlay_audio=1  
#disable_uboot_overlay_wireless=1  
#disable_uboot_overlay_adc=1  

###Cape Universal Enable  
enable_uboot_cape_universal=1  

uboot_overlay_addr4=/lib/firmware/BB-UART1-00A0.dtbo  
uboot_overlay_addr5=/lib/firmware/BB-UART2-00A0.dtbo  
uboot_overlay_addr6=/lib/firmware/BB-PWM2-00A0.dtbo  
uboot_overlay_addr7=/lib/firmware/BB-BONE-AUDI-01-00A0.dtbo  
# ADC loaded as standard but PWMs and UARTs need loading as above 

************************************
My AUDI-01.dts file looks like this:
/*
 * Copyright (C) 2012 Texas Instruments Incorporated - http://www.ti.com/
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License version 2 as
 * published by the Free Software Foundation.
 */
/dts-v1/;
/plugin/;

/ {
        compatible = "ti,beaglebone", "ti,beaglebone-black";

        /* identification */
        part-number = "BB-BONE-AUDI-01";
        version = "00A0", "A0";

        /* state the resources this cape uses */
        exclusive-use =
                /* the pin header uses */
                "P9.14",        /* leds:   gpio1_18 */
[... 114 lines cut ...]

        };
};

*******************
This is my asound.conf file:
debian@beaglebone:~$ cat /etc/asound.conf  
defaults.pcm.dmix.rate 96000
defaults.pcm.dmix.format S16_LE

pcm.dsnooped {
    type dsnoop
    slave {
        pcm "hw:0,0"
        format S16_LE
        rate 96000
        channels 2

        period_size 1024
        buffer_size 4096

        periods 0
        period_time 0
    }
}

jjghjjgh, 2018-03-23:
@tderensis @ncobbett 
Hello,
I saw that you have tried playing audio on beaglebone black as well using  tlv320dac3100 codec.  If yes can you please help me with some issues? One thing is is the dts file shared above will be same for BBB as well as i saw its explained for BBG wireless. Actually i want to implement this on 9.3 debian with 4.15 kernel.On older debian versions so many people have explained the same but those things cant be used directly on newer versions as device tree overlay has changed in newer versions. I am a newbie to Linux so any solution will be really helpful.And i also feel that the dts file explained for BBBG wireless is running on 4.9 kernel and when i compared 4.9 kernel is similar to 4.15 as well but its just that i wanted to confirm what chnages will i have to do in order to implement it on BBB with latest 4.15 kernel version. waiting for your response.
thanks in advance,
Jithu.

rogerlucas, 2019-01-17:
Hello @tderensis , how are you?

I'm working with MAX98357 (Adafruit I2S 3W Class D Amplifier Breakout - MAX98357A) and encountering some difficulties.
Could you forward me the DTS you used?

Thank You.

rogerlucas, 2019-01-18:
Hello @tderensis, can you send me your custom DTS for MAX98357A?

I was able to configure the audio output for a MAX98357A breakout board, but it does not work in conjunction with a LCD cape, only working alone, without the LCD. I do not know what else to do..

rogerlucas, 2019-01-18:
Hi @tderensis 
Many thanks for the answer.
I'm having trouble understanding the DTS syntax and where I need to insert the changes you've made.
If you can help me and send the DTS tuned I will be eternally grateful. I'm developing a custom version of Black Beaglebone Uam, I'm going to insert an LCD terminal and the max98357a straight into the board design.

tderensis, 2019-01-18:
I can't provide the entire DTS, but I can guide you a little on how the DTS files work. Usually there will be a number of DTSI files (include files) which are used to compose the final DTS file. The DTSI file will have SoC-level specifications, or sometimes definitions common to several boards. When you include a DTSI and then redefine a value in the DTS, then the new value is used instead. This is used a lot to define SoC devices such as UART which are disabled in the DTSI and then expected to be fully configured and enabled in the DTS file.

It may be easier if you send me the DTS file you are working with. Then I can help add the required parts.

rogerlucas, 2019-01-18:
Hello @tderensis, thank you very much for the help.

My scenario today is as follows:

A beaglebone black wireless that works perfectly with the panel "NHD-7.0CTP-CAPE-L"

I can also play audio using a
"Adafruit I2S 3W Class D Amplifier Breakout - MAX98357A"

and with this overlay loaded:
uboot_overlay_addr2=/lib/firmware/BB-BONE-AUDI-02-00A0.dtbo

The LCD works perfectly.
When I use the max98357a i2s output alone without the LCD works perfectly too, but in conjunction with the LCD does not.

I would like to create a unified DTS for the LCD and i2s, if possible or a separate DTS for the i2s that worked without conflict with the LCD.

rogerlucas, 2019-01-18:
The i2s are still recognized with the LCD, but do not play audio:
root@beaglebone:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: B [AudioCape Rev B], device 0: davinci-mcasp.0-tlv320aic3x-hifi tlv320aic3x-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@beaglebone:~#
root@beaglebone:~# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=B
    AudioCape Rev B,
    Default Audio Device
sysdefault:CARD=B
    AudioCape Rev B,
    Default Audio Device
dmix:CARD=B,DEV=0
    AudioCape Rev B,
    Direct sample mixing device
dsnoop:CARD=B,DEV=0
    AudioCape Rev B,
    Direct sample snooping device
hw:CARD=B,DEV=0
    AudioCape Rev B,
    Direct hardware device without any conversions
plughw:CARD=B,DEV=0
    AudioCape Rev B,
    Hardware device with all software conversions

root@beaglebone:~# aplay -vv ac_dc.wav
Playing WAVE 'ac_dc.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Plug PCM: Hardware PCM card 0 'AudioCape Rev B' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1572864000
  appl_ptr     : 0
  hw_ptr       : 0
#            +                                     | 24%

rogerlucas, 2019-01-20:
@tderensis 
I was able to understand the DTC syntax and created a new custom using the snippets that you made available here to work with max98357a on my hardware.

The material you posted here has helped me a lot and I believe I have helped many people too.

Thank you very much for the instructions here.

rogerlucas, 2019-01-22:
@tderensis
I was able to understand the DTC syntax and created a new custom using the snippets that you made available here to work with max98357a on my hardware.

It worked!

Thank you.