beagleboard/bb.org-overlays #41: 100% CPU usage after loading BB-ADC DT overlay

fivdi, 2017-03-19:
If the [BB-ADC](https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-ADC-00A0.dts) DT overlay is loaded, the process `battery_monitor` which initially doesn't appear to have much to do immediately starts using as much CPU time as it can, in this case ~100%.

This was observed on a BeagleBone Black with "`Debian 8.6 2016-12-09 4GB SD IoT`" downloaded at [BeagleBoard.org Latest Firmware Images](https://beagleboard.org/latest-images). The BeagleBone Black was running from a wall power supply, not a battery.

Here's the output of `top` before loading `BB-ADC`:

```
top - 12:30:40 up 5 min,  1 user,  load average: 0.04, 0.50, 0.33
Tasks:   1 total,   0 running,   1 sleeping,   0 stopped,   0 zombie
%Cpu(s):  6.2 us,  7.1 sy,  0.0 ni, 86.3 id,  0.2 wa,  0.0 hi,  0.1 si,  0.0 st
KiB Mem:    499736 total,   122160 used,   377576 free,     9144 buffers
KiB Swap:        0 total,        0 used,        0 free.    49056 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND      
  882 root      20   0   57256   1524   1428 S  0.0  0.3   0:00.34 battery_mon+ 
```

And here's the output of `top` slightly less than 40 minutes after loading `BB-ADC`:

```
top - 13:09:40 up 44 min,  1 user,  load average: 1.00, 1.00, 1.00
Tasks:   1 total,   1 running,   0 sleeping,   0 stopped,   0 zombie
%Cpu(s): 86.3 us,  1.2 sy,  0.0 ni, 12.3 id,  0.0 wa,  0.0 hi,  0.1 si,  0.0 st
KiB Mem:    499736 total,   123408 used,   376328 free,     9376 buffers
KiB Swap:        0 total,        0 used,        0 free.    49452 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND      
  882 root      20   0   57256   1524   1428 R 99.9  0.3  38:05.04 battery_mon+ 
```

Apologies if this issue was created in the wrong place.

RobertCNelson, 2017-03-19:
@fivdi "battery_mon" update your roboticscape deb package (sudo apt update ; sudo apt upgrade)

Regards,

fivdi, 2017-03-20:
@RobertCNelson Thanks, that fixes the issue. (I stopped doing `sudo apt update ; sudo apt upgrade` a long time ago because it can take so long. This time it took 40 minutes :))

closed 2017-03-20 by fivdi