beagleboard/bb.org-overlays #51: BB-UART4-RS485-00A0.dts Overlay

ercint, 2017-08-13:
Hi,

I applied overlays to beaglebone black but cannot use RS485 overlay. 

When I try to make MODBUS communication with UART4, the RTS pin cannot be set.

I can use usbtoRS485 converter and able to communicate but cannot make it with UART4.

I also make OMAP driver as 
CONFIG_SERIAL_8250_OMAP=n
CONFIG_SERIAL_OMAP=y

Can anyone help me about that?

Thanks in advance

RobertCNelson, 2017-08-13:
@ercint for rs485, dont use the seria_8250_omap, we might finally have this working in v4.14.x/v4.15.x timeframe..

Regards,

ercint, 2017-08-13:
Hi,

I tried also Silern RS485 Overlay before but at the boot BBB cannot load that.

Yours can be load but I cannot use GPIO as RTS.

Knaldgas, 2021-03-06:
@RobertCNelson has this been solved for 4.19.94-ti-r42 ?
The issue is still open so probably not. At least I've tried to install https://debian.beagleboard.org/images/bone-eMMC-flasher-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz and changed uEnv.txt:
```
dtb_overlay=/lib/firmware/BB-UART4-RS485-00A0.dtbo
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1

```
I've tried with the code suggested by https://www.kernel.org/doc/html/latest/driver-api/serial/serial-rs485.html
```
#include <stdio.h>
#include <unistd.h>
#include <fcntl.h>
#include <linux/serial.h>
#include <sys/ioctl.h>

void main (void) {
        int fd = open ("/dev/ttyO4", O_RDWR);
        if (fd < 0) {
                fprintf (stderr, "Error opening device\n");
        }

        struct serial_rs485 rs485conf;

        rs485conf.flags |= SER_RS485_ENABLED;
        rs485conf.flags |= SER_RS485_RTS_ON_SEND;
        rs485conf.flags &= ~(SER_RS485_RTS_AFTER_SEND);
	rs485conf.delay_rts_before_send = 1;
	rs485conf.delay_rts_after_send = 1;

        if (ioctl (fd, TIOCSRS485, &rs485conf) < 0) {
                fprintf (stderr, "Error configuring device\n");
        }

        char write_buffer[] = "ABCDEF";
        int  bytes_written  = 0;

        bytes_written = write (fd, write_buffer, sizeof (write_buffer));
        if (bytes_written != sizeof (write_buffer)) {
                fprintf (stderr, "Error sending data\n");
        }

        if (close (fd) < 0) {
                fprintf (stderr, "Error closing device\n");
        }
}

```
I have my oscilloscope on UART4 TX with good signal, and on P9.27 with constant low (not working).
(P9.27 as specified here: https://beagleboard.org/Support/bone101)

My goal is to have RS485 working with a Waveshare CAN/RS485 cape. (CAN controller removed from board.)

~Per

SarfarazShaikh-eaton, 2024-08-23:
any luck in using the RTS with driver ?