beagleboard/bb.org-overlays #52: request to add ETH WIZ click

easyw, 2017-10-05:
Hi,
would be possible to add this 10/100 BASE ethernet adapter?
https://shop.mikroe.com/eth-wiz-click
This is much better than the ETH Click which is a 10base adapter available as cape atm.

Thanks
Maurice

RobertCNelson, 2017-10-05:
I see you forgot to attach your patch to update the wiznet driver to support device tree's..

Regards,

easyw, 2017-10-05:
Hi Robert, 
unfortunately I don't have any patch... mine was just a request to eventually support this nice ethernet module for beagle bone ...
Thanks for considering that
Maurice

RobertCNelson, 2017-10-05:
Well, this project is run by volunteers.  To speed up your request, you may have to do a little development.

Regards,

RobertCNelson, 2017-10-05:
https://github.com/beagleboard/pocketbeagle/wiki/mikroBus%E2%84%A2-Click-Boards

merged

closed 2017-10-05 by RobertCNelson