beagleboard/bb.org-overlays #53: Need help loading custom cape with Stretch

emckenzie, 2017-10-11:
Hi,
I'm using stretch and am trying to load a custom cape.  I can't seem to affect the log file at all with respect to loading capes.  Everything I try yields the same log showing nothing about my cape.

The dtbo is: BB-BONE-PRU-00A0.dtbo
I have placed this file in /lib/firmware.

I have modified my uEnv.txt with the line:  enable_uboot_overlays=1

I have tried the following in uEnv.txt:
dtb_overlay=/lib/firmware/BB-BONE-PRU-00A0.dtbo

and also at a separate time:
uboot_overlay_addr0=/lib/firmware/BB-BONE-PRU-00A0.dtbo

I have also tried adding the following line to /etc/default/capemgr:
CAPE=BB-BONE-PRU  (also tried CAPE=BB-BONE-PRU-00A0.dtbo)

I never see any change in the boot log.  There is nothing about my cape and I always see the line " bone_capemgr.uboot_capemgr_enabled=1" in the log even if I set it to 0.

I'm sure I'm missing something about the new cape system.  I am excited to get it working.  It might help with some issues on my board with loading the cape after bootup.   Hot swapping cape files was a neat idea, but I think in practical use, rebooting each time you want to change a cape is just fine.

Thanks for any help.

(I'm using this image:  https://debian.beagleboard.org/images/bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz)

- Eric McKenzie

RobertCNelson, 2017-10-12:
@emckenzie so with u-boot overlays, you will not seeing any special "loading" messages in the kernel.  As the overlay is added before the kernel boots.  Instead all you should see is your new node added seemlessly in /proc/device-tree/ so if it worked, your device should just work.

To debug, you'll need a usb-uart adapter, since U-Boot is doing the loading..

One quick debug tool, use the "version.sh" script, this will give us a hint if u-boot is able to do it's job.

```
sudo /opt/scripts/tools/version.sh
```
Regards,

emckenzie, 2017-10-12:
Hi Robert,

Thank you for those debugging tools.  I'll see if I can figure out if it's loading.

- Eric

closed 2017-10-12 by emckenzie