beagleboard/bb.org-overlays #55: Cannot find /sys/firmware/devicetree/base/bone_capemgr/slots or /sys/devices/platform/bone_capemgr/slots

shivanandvp, 2017-12-06:
I am trying to get encoder input using eQEP on the BBB. I am running the Ubuntu 16.04 image downloaded from rcn-ee website.

```
ubuntu@arm:/sys/firmware/devicetree/base/bone_capemgr$ uname -r
4.9.54-ti-r68
ubuntu@arm:~$ ls -ltr /sys/devices/platform/bone_capemgr
total 0
-rw-r--r-- 1 root root 4096 Dec  6 17:29 uevent
lrwxrwxrwx 1 root root    0 Dec  6 17:29 subsystem -> ../../../bus/platform
drwxr-xr-x 2 root root    0 Dec  6 17:30 power
lrwxrwxrwx 1 root root    0 Dec  6 17:30 of_node -> ../../../firmware/devicetree/base/bone_capemgr
-r--r--r-- 1 root root 4096 Dec  6 17:30 modalias
-rw-r--r-- 1 root root 4096 Dec  6 17:30 driver_override
lrwxrwxrwx 1 root root    0 Dec  6 17:30 driver -> ../../../bus/platform/drivers/bone_capemgr
ubuntu@arm:~$ ls -ltr /sys/firmware/devicetree/base/bone_capemgr/
total 0
-r--r--r-- 1 root root 16 Dec  6 17:53 compatible
-r--r--r-- 1 root root 20 Dec  6 17:53 nvmem-cells
-r--r--r-- 1 root root  4 Dec  6 17:53 #slots
-r--r--r-- 1 root root  5 Dec  6 17:53 status
drwxr-xr-x 4 root root  0 Dec  6 17:53 baseboardmaps
-r--r--r-- 1 root root 13 Dec  6 17:53 name
-r--r--r-- 1 root root 34 Dec  6 17:53 nvmem-cell-names
ubuntu@arm:~$ cat /sys/firmware/devicetree/base/bone_capemgr/#slots
ubuntu@arm:~$ 
```

I tried following the readme, but this crucial file 'slots' is missing!! Could you please help?

RobertCNelson, 2017-12-06:
@shivanandvp slots file is gone, see:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

As U-Boot is in control.

Regards,

shivanandvp, 2017-12-06:
Although I enabled everything as in https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays and used config-pin on 11 and 12, I still cannot read any encoder position:

ubuntu@arm:/sys/devices/platform/ocp/48304000.epwmss/48304180.eqep$ tail -f position
0

Could you help me?

RobertCNelson, 2017-12-06:
Encoders are documented very well in the adafruit lib:

https://github.com/adafruit/adafruit-beaglebone-io-python/tree/master/Adafruit_BBIO#usage

Regards,

shivanandvp, 2017-12-07:
I followed the above documentation exactly and still cannot read the encoder position.
root@arm:/home/ubuntu# uname -r
4.9.54-ti-r68
# I have the encoder connected at Pins 11 and 12.
```
root@arm:/home/ubuntu# config-pin P8.11 qep
root@arm:/home/ubuntu# config-pin P8.12 qep
root@arm:/home/ubuntu# cat /sys/devices/platform/ocp/48304000.epwmss/48304180.eqep/position
289
```
# I turn the motor shaft at this point
```
root@arm:/home/ubuntu# cat /sys/devices/platform/ocp/48304000.epwmss/48304180.eqep/position
289
```
# There is no change in encoder position

Here is my /boot/uEnv.txt file:
```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0
uname_r=4.9.54-ti-r68
cmdline=coherent_pool=1M quiet cape_universal=enable
#cape_enable=bone_capemgr.enable_partno=cape-universala
#uuid=
#dtb=
###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/bone_eqep2b-00A0.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###
cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable
#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable video=HDMI-A-1:1024x768@60e
##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=
##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=
##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

cape_enable=capemgr.enable_partno=BB-UART2
```

Why do you think I cannot read the encoder? @RobertCNelson

RobertCNelson, 2017-12-07:
What encoder (part #) and how do you have power running to it..

RobertCNelson, 2017-12-07:
See this page:

https://books.google.com/books?id=xDXKBwAAQBAJ&pg=PA66&lpg=PA66&dq=beaglebone+black++wireing+encoder&source=bl&ots=pVHyAdjGUv&sig=gh987OIO2ApWQQ8a-2BP7kVRKfo&hl=en&sa=X&ved=0ahUKEwj26LzZwfjXAhVRy2MKHU9HCM0Q6AEITTAF#v=onepage&q=beaglebone%20black%20%20wireing%20encoder&f=false

shivanandvp, 2017-12-07:
Encoder part #: https://www.dfrobot.com/wiki/index.php/12V_DC_Motor_251rpm_w/Encoder_(SKU:_FIT0186)

Power: Green to ground (P8.2), Blue to VDD5V (P9.5), Yellow to P8.12, and White to P8.11. The diagram for these colors are at the above link

I have the dtbo files in the right location:
```
bone_eqep0-00A0.dtbo  bone_eqep1-00A0.dtbo  bone_eqep2-00A0.dtbo  bone_eqep2b-00A0.dtbo
root@arm:/lib/firmware# ls -ltr bone*
-rw-r--r-- 1 root root 1082 Dec  7 01:09 bone_eqep0-00A0.dtbo
-rw-r--r-- 1 root root 1098 Dec  7 01:09 bone_eqep1-00A0.dtbo
-rw-r--r-- 1 root root 1098 Dec  7 01:09 bone_eqep2-00A0.dtbo
-rw-r--r-- 1 root root 1096 Dec  7 18:27 bone_eqep2b-00A0.dtbo
```
How do I load the right overlays with Uboot enabled? (I need Uboot for running a motor through a cape) @RobertCNelson

RobertCNelson, 2017-12-07:
@shivanandvp well for starters, get a new BBB..
You probably fried P8.12/P8.11.. Those pin's are 3.3V.. and shoving 5V fried them...

That encoder is 5V, not going to work unless you add a level shifter.

Regards,

shivanandvp, 2017-12-07:
@RobertCNelson Can I use other pins? Can I read the output some other way?

RobertCNelson, 2017-12-07:
till you get a 5v -> 3.3v level converter setup, you shouldn't be using that encoder with the BBB.

shivanandvp, 2017-12-07:
The pins are not fried. I disable Uboot in /boot/uEnv.txt file and do the below, and the encoder works:
```
root@arm:/home/ubuntu# cd /lib/firmware
root@arm:/lib/firmware# echo bone_eqep2b > /sys/devices/platform/bone_capemgr/slots
root@arm:/lib/firmware# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,bone_eqep2b
root@arm:/lib/firmware# cat /sys/devices/platform/ocp/48304000.epwmss/48304180.eqep/position
0
root@arm:/lib/firmware# cat /sys/devices/platform/ocp/48304000.epwmss/48304180.eqep/position
-242
```
1) If the pins are not fried, why do you think uboot does not enable my encoder? I am stuck with uboot because my cape uses it to run motors.
2) Can I use a voltage divider instead?
@RobertCNelson

RobertCNelson, 2017-12-07:
This would do the same thing as above, just with u-boot doing the load:

```
uboot_overlay_addr0=/lib/firmware/bone_eqep2b-00A0.dtbo
```
Yes a voltage divider would also work..

If that doesn't work run:

```
sudo /opt/scripts/tools/version.sh | grep bootloader
```

Regards,

shivanandvp, 2017-12-07:
@RobertCNelson  Thank you so much!!! This enables both the motor and encoder!!

Below is my /boot/uEnv.txt file:

```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0
uname_r=4.9.54-ti-r68
cmdline=coherent_pool=1M quiet cape_universal=enable
#cape_enable=bone_capemgr.enable_partno=cape-universala
#uuid=
#dtb=
###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
uboot_overlay_addr0=/lib/firmware/bone_eqep2b-00A0.dtbo
#dtb_overlay=/lib/firmware/bone_eqep2b-00A0.dtbo
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/bone_eqep2b-00A0.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###
cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable
#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable video=HDMI-A-1:1024x768@60e
##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=
##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=
##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

cape_enable=capemgr.enable_partno=BB-UART2
```

shivanandvp, 2017-12-07:
What is special about addr0 in uboot_overlay_addr0? How does that work, while dtb_overlay=/lib/firmware/bone_eqep2b-00A0.dtbo doesn't? @RobertCNelson

RobertCNelson, 2017-12-07:
it shouldn't matter, 

uboot_overlay_addr0 is one of the first ones:
```
+				"if test -n ${uboot_overlay_addr0}; then " \
+					"if test -n ${disable_uboot_overlay_addr0}; then " \
+						"echo uboot_overlays: uboot loading of [${uboot_overlay_addr0}] disabled by /boot/uEnv.txt [disable_uboot_overlay_addr0=1]...;" \
+					"else " \
+						"setenv uboot_overlay ${uboot_overlay_addr0}; " \
+						"run capeloadoverlay;" \
+					"fi;" \
+				"fi;" \
```

vs: dtb_overlay is almost last:

```
+				"if test -n ${dtb_overlay}; then " \
+					"setenv uboot_overlay ${dtb_overlay}; " \
+					"echo uboot_overlays: [dtb_overlay=${uboot_overlay}] ... ;" \
+					"run capeloadoverlay;" \
+				"fi;" \
```

other then the disable, the syntax is the same;

https://github.com/RobertCNelson/Bootloader-Builder/blob/master/patches/v2018.01-rc1/0002-U-Boot-BeagleBone-Cape-Manager.patch

shivanandvp, 2017-12-07:
@RobertCNelson  Do I use a different address if I need to add in more encoders? Like uboot_overlay_addr1=/lib/firmware/bone_eqep2-00A0.dtbo ?

RobertCNelson, 2017-12-07:
Yeap, there's 8 total:

```
uboot_overlay_addr0=
uboot_overlay_addr1=
uboot_overlay_addr2=
uboot_overlay_addr3=
uboot_overlay_addr4=
uboot_overlay_addr5=
uboot_overlay_addr6=
uboot_overlay_addr7=
```

They are loaded in order...