beagleboard/bb.org-overlays #56: Adding spi1_cs0 on PocketBeagle

patrickpoirier51, 2017-12-09:
We need to have 2 devices on spi1,  following the overlay logic, we mapped using config:
config-pin P2_30 spi_cs
cat /sys/devices/platform/ocp/ocp:P2_30_pinmux/state  == spi_cs

but this is just a mapping, so the driver does not reload and the new port is not present
ls /dev/spi*
/dev/spidev0.0 /dev/spidev1.1

Do we need to make a new DTS for the Pocket or there is a method to map within uEnv.txt ?

Best regards

RobertCNelson, 2017-12-09:
Doh! I originally had both cs options enabled by default, but then removed them.. 

@patrickpoirier51 in teh main *.dtb spi1 looks like:

```
&spi1 {
	#address-cells = <1>;
	#size-cells = <0>;
	status = "okay";

	channel@0 {
		#address-cells = <1>;
		#size-cells = <0>;
		compatible = "spidev";
		reg = <0>;
		spi-max-frequency = <24000000>;
		status = "disabled";
	};

	channel@1 {
		#address-cells = <1>;
		#size-cells = <0>;
		compatible = "spidev";
		reg = <1>;
		spi-max-frequency = <24000000>;
	};
};
```

Create an overlay like:

```
/dts-v1/;
/plugin/;

/ {
	fragment@0 {
		target = <&spi1>;
		__overlay__ {

			channel@0 {
				status = "okay";
			};
		};
	};
};

```
Next week i'll patch the main dtb to enable spidev1.0 by default..

Regards,

patrickpoirier51, 2017-12-09:
Thanks Robert
FYI == https://files.gitter.im/mirkix/BBBMINI/sLkV/image.png

RobertCNelson, 2017-12-09:
Awesome!

patrickpoirier51, 2017-12-14:
Robert,

Following your last comment : 

> Next week i'll patch the main dtb to enable spidev1.0 by 

Do you have a timeframe for this  ?

RobertCNelson, 2017-12-14:
This should be fixed across the board as of last week..

Regards,

patrickpoirier51, 2017-12-14:
Great, Thanks again

patrickpoirier51, 2017-12-15:
There is a little of you in this ;-)
https://www.youtube.com/watch?v=BBnUvO6x0oY&feature=youtu.be

RobertCNelson, 2017-12-15:
@patrickpoirier51 that's AWESOME!!!

@jadonk

closed 2017-12-18 by patrickpoirier51

pdp7, 2019-01-09:
@jadonk @RobertCNelson I thought that P2.32 is supposed to be SPI1 CS0, and that P2.30 is supposed to be SPI1 CLK?  [https://lh3.googleusercontent.com/-j2bV1px5PLs/WiQ4m3bRolI/AAAAAAABGd4/RofRJySbdrgeumCJ1crRRkUn5FhczdxNQCHMYBhgL/w9999/bboard_pocketbeagle_back.jpg](That is my impression from the PB silkscreen)