beagleboard/bb.org-overlays #57: Cannot  config-pin P2_18 pru_ecap

patrickpoirier51, 2017-12-14:
Cannot set pru_ecap , all other modes can be set:

debian@beaglebone:~$ config-pin -l P2_18
default gpio gpio_pu gpio_pd gpio_input qep pru_ecap pruin

debian@beaglebone:~$ sudo config-pin P2_18 pru_ecap
[sudo] password for debian:
dash: echo: I/O error
Cannot write pinmux file: /sys/devices/platform/ocp/ocp:P2_18_pinmux/state

Look ok on DTS: https://github.com/beagleboard/bb.org-overlays/blob/master/tools/pinmux-generator/PocketBeagle.dts#L570

Latest Debian 9.2 IoT + rt 4.14

RobertCNelson, 2017-12-14:
@patrickpoirier51  thanks for testing, so it's listed:

https://github.com/beagleboard/bb.org-overlays/blob/master/tools/beaglebone-universal-io/config-pin#L355-L356

Please run:
```
sudo /opt/scripts/tools/version.sh
```
It's either the kernel or the config-pin app needs an update

Regards,

patrickpoirier51, 2017-12-14:
Boy , You are Fast !!! 
Sorry I did not mention the update.
I swap with my other SD card that has vanilla 9.2 + 4.14 RT and it works.
The only difference I can think of is that I loaded Adafruit_BBIO... otherwise its just pure corruption on the drivers side... should use shutdown more often ;-)

Thanks again for your fast response Robert

RobertCNelson, 2017-12-14:
This was something i just added last week (i think?)... I just want to make sure we have all the possibilities now correctly exported. ;)

patrickpoirier51, 2017-12-14:
Sorry for the delay, I just reflashed I have the same problem

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[ef2c5982e3788e07d9ec4a3d23089dc5e5a3f9cd]
eeprom:[A335PBGL00A21741GPB43449]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.14.5-bone-rt-r8]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171009.0-0rcnee1~stretch+20171009]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

RobertCNelson, 2017-12-14:
```
pkg:[bb-cape-overlays]:[4.4.20171009.0-0rcnee1stretch+20171009]
```
please upgrade that package

```
sudo apt upgrade bb-cape-overlays
```

and re-test:

https://github.com/beagleboard/bb.org-overlays/commits/master

Regards,

patrickpoirier51, 2017-12-14:
Here is the info about my SD that can == config-pin P2_18 pru_ecap

git:/opt/scripts/:[d36fe9a7be9ebfc872b10a470e904ab4c61c4516]
eeprom:[A335PBGL00A21741GPB43449]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.14.4-bone-rt-r7]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171207.0-0rcnee1~stretch+20171207]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

patrickpoirier51, 2017-12-14:
Yes that was it
pkg:[bb-cape-overlays]:[4.4.20171207.0-0rcnee1~stretch+20171207]

Thanks

closed 2017-12-14 by patrickpoirier51