beagleboard/bb.org-overlays #59: Using custom cape doesn't seems to work

m2mvpadha, 2017-12-18:
For our HW solution with BBBW we need to use pins in different setup as they are set in universal capes.

We are running with 

debian@beaglebone:~$ uname -a
Linux beaglebone 4.4.68-ti-r115 #1 SMP Sat Jul 29 21:04:28 UTC 2017 armv7l GNU/Linux
debian@beaglebone:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 8.9 (jessie)
Release:        8.9
Codename:       jessie
debian@beaglebone:~$ dtc --version 
Version: DTC 1.4.4

The problem is that custom cape we created doesn't seems to work. Bellow will you find our cape code
I extracted this code out of universal cape, and modified it. I compiled it along with /opt/source/bb.org-overlays/ inside BBB, so I don't expect that the resulting dtbo is wrong

[M2M_MUMA_PINS-00A0.txt](https://github.com/beagleboard/bb.org-overlays/files/1567862/M2M_MUMA_PINS-00A0.txt)

I tried to use our custom cape in /boot/uEnv.txt in 
- cape_enable=bone_capemgr.enable_partno=
- dtb_overlay=

I tried to load it using config-pin utility, but none of these possibilities seems to work.

The goal is to have our custom "muma" mode preset. In pinctrl-names "muma" is mentioned

It seems that the cape is loaded, but just partially.
When I cat for pins status, I see the pins I would like to use (as they are present in our custom cape)
debian@beaglebone:~$ sudo cat /sys/devices/platform/ocp/ocp\:cape-universal/status
 0 P9_31                    110 IN  0
 1 P9_30                    112 IN  0
 2 P9_29                    111 IN  0
 3 P9_27                    115 IN  0
 4 P9_26                     14 IN  0
 5 P9_24                     15 IN  0

When I query with config-pin -i/-l on particular pin, I don't see in the list our custom mode (muma).
In fact I see the "original" setup for all the pins

debian@beaglebone:~$ config-pin -l P9.24
**default gpio gpio_pu gpio_pd uart can i2c pru_uart pruin**
debian@beaglebone:~$ config-pin -i P9.24
Pin name: P9_24
Function if no cape loaded: gpio
Function if cape loaded: **default gpio gpio_pu gpio_pd uart can i2c pru_uart pruin**
Function information: gpio0_15 default gpio0_15 gpio0_15 gpio0_15 uart1_txd dcan1_rx i2c1_scl pr1_uart0_txd pr1_pru0_pru_r31_16
Cape: cape-universal cape-universala cape-universaln cape-universalh
Kernel GPIO id: 15
PRU GPIO id: **47**

I would expect to see **"default", "muma";**

What am I doing wrong?

BR
Vladimir

RobertCNelson, 2017-12-18:
@m2mvpadha please share the output of:

```
sudo /opt/scripts/tools/version.sh
```
Regards,

RobertCNelson, 2017-12-18:
ps, config-pin isn't dynamic, so any custom "names" won't be found.. but you still echo the custom name into the sys directory config-pin uses..

RobertCNelson, 2017-12-18:
For example see what jason did here:

https://gist.github.com/jadonk/0e4a190fc01dc5723d1f183737af1d83#file-blue_adafruit24_setup-sh

```
cho gpio > /sys/devices/platform/ocp/ocp\:P9_23_pinmux/state # RESET - V14 - GP0_4
echo gpio > /sys/devices/platform/ocp/ocp\:U16_pinmux/state # D/C - U16 - GP0_3
echo spi > /sys/devices/platform/ocp/ocp\:P9_31_pinmux/state # SCLK - A13 - S1.1_5
echo spi > /sys/devices/platform/ocp/ocp\:P9_29_pinmux/state # MISO - B13 - S1.1_4
echo spi > /sys/devices/platform/ocp/ocp\:P9_30_pinmux/state # MOSI - D12 - S1.1_3
```

m2mvpadha, 2017-12-19:
Here is output from version.sh

git:/opt/scripts/:[8e0b1e6f135ff3c30468f49a7a952c98be97219a]
eeprom:[A335BNLTBWA51646BBWG1744]
dogtag:[BeagleBoard.org Debian Image 2017-08-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.07-00002-g61c3ee0fb5]
kernel:[4.4.68-ti-r115]
nodejs:[v4.8.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~jessie+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170727-0rcnee0~jessie+20170727]
pkg:[firmware-ti-connectivity]:[20161130-3]

m2mvpadha, 2017-12-19:
Hi Robert

Well I tried to echo our custom name, but it didn't worked, so it really seems that out custom cape was not loaded correctly? Even I see it in dmesg

[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.enable_partno=M2M_MUMA_PINS bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_poo
l=1M net.ifnames=0 quiet
[    2.494333] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,BWA5,1646BBWG1744'
[    2.494364] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
[    2.494402] bone_capemgr bone_capemgr: slot #0: auto loading handled by U-Boot
[    2.494424] bone_capemgr bone_capemgr: slot #1: auto loading handled by U-Boot
[    2.494446] bone_capemgr bone_capemgr: slot #2: auto loading handled by U-Boot
[    2.494491] bone_capemgr bone_capemgr: slot #3: auto loading handled by U-Boot
[    2.494512] bone_capemgr bone_capemgr: enabled_partno PARTNO '**M2M_MUMA_PINS**' VER 'N/A' PR '0'
[    2.494523] bone_capemgr bone_capemgr: slot #4: override
[    2.494537] bone_capemgr bone_capemgr: slot #4: auto loading handled by U-Boot
[    2.495142] bone_capemgr bone_capemgr: initialized OK.

echo muma to P9.24

debian@beaglebone:~$ echo muma > /sys/devices/platform/ocp/ocp\:P9_24_pinmux/state 
-bash: echo: write error: No such device

In dmesg I see that our mode is not present
bone-pinmux-helper ocp:P9_24_pinmux: Failed to find state muma

m2mvpadha, 2017-12-19:
The only way which worked for me was to modify /boou/uEnv.txt, where I added the cape as custom dtb_overlay. I think this is not the proper way to deal with capes.

###Custom Cape
dtb_overlay=/lib/firmware/M2M_MUMA_PINS-00A0.dtbo

The proper way should be to "enable" our custom cape with "cape_enable" in uEnv.txt, right?

cape_enable=bone_capemgr.enable_partno=M2M_MUMA_PINS

RobertCNelson, 2017-12-19:
@m2mvpadha the proper way to handle this today is;

```
dtb_overlay=/lib/firmware/M2M_MUMA_PINS-00A0.dtbo
```

The old "cape_enable" kernel options is no ripped out as of v4.15.x

m2mvpadha, 2017-12-19:
Ok GR8, in such case it's solved

Thanks for your time

closed 2017-12-19 by m2mvpadha