beagleboard/bb.org-overlays #60: Cannot load the slots 7 and 8

sanjay265, 2017-12-20:
Hi

while I am trying to do dtos I cannot see all the slots loaded after exporting and doing " cat $SLOTS "
it shows till slots 4 and 5. And when i look into bone_capemgr.9 directory, I cannot see slots 7 and 8. By the way, I am using debian Image 7.8 on my beaglebone balck. Please help me, Im just a beginner.

Thanks

RobertCNelson, 2017-12-20:
@sanjay265 

3.8.x based kernels are not supported by this repo

kernel overlays are now considered EOL

For Support with this repo, please upgrade to v4.4.x and U-Boot Overlays

Otherwise, feel free to contact or search the archive on this mailing list: https://groups.google.com/forum/#!forum/beagleboard

Regards,

sanjay265, 2017-12-20:
ok. 

thanks robert

closed 2017-12-20 by sanjay265