beagleboard/bb.org-overlays #62: BBBlue Cannot set P8_15 to pruecapin_pu using config-pin

patrickpoirier51, 2018-01-29:
Hello Robert

We are updating  the BBLUE Image - Kernel and Configuration
We want to implement config-pins and replace:
echo pruecapin_pu > /sys/devices/platform/ocp/ocp:P8_15_pinmux/state
with
sudo config-pin P8_15  pruecapin_pu
But we get this error:
Invalid mode: pruecapin_pu
 
Using config-pin list gives:
sudo config-pin -l P8_15
default gpio gpio_pu gpio_pd gpio_input qep pru_ecap pruin
Seems that pruecapin_pu is not in the profile

And if we try 
sudo config-pin P8_15 pru_ecap
We have this error:
[  494.769840] bone-pinmux-helper ocp:P8_15_pinmux: Failed to find state pru_ecap
dash: echo: I/O error
Cannot write pinmux file: /sys/devices/platform/ocp/ocp:P8_15_pinmux/state

RobertCNelson, 2018-01-29:
@patrickpoirier51 for the Blue, keep using:
```
/sys/devices/platform/ocp/ocp:P8_15_pinmux/state
```
P8_15: Black
```
P8_15_pru_ecap_pin: pinmux_P8_15_pru_ecap_pin { pinctrl-single,pins = <
	AM33XX_IOPAD(0x083c, PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE5) >; };	/* gpmc_ad15.pr1_ecap0_ecap_capin_apwm_o */
```

P8_15: Blue:
```	P8_15_pruecapin_pu_pin: pinmux_P8_15_pruecapin_pu_pin {
		pinctrl-single,pins = <0x03c  0x35>; };     /* Mode 5, Pull-Up, RxActive */
```
I'm working on a big rewrite/unification, so every board will have every pin option, and cape's will only "disable" the "config-pin" options they use. (You can see this today with the PocketBeagle). Part of that rewrite will help cleanup "config-pin", as of right now, config-pin has it's own pin database..  The rewrite will pull it from a kernel pin list..

Regards,

patrickpoirier51, 2018-01-29:
Great !!! Keep on the good job Robert
No problem for us to echo >  for the moment

Best Regards

skfelix, 2018-03-14:
Use the cape-universalh overlay in your uEnv.txt. I will work.
The cape-universalh may conflict your other device tree overlays, so I've edited the cape-universalh-00A0.dts and put only the pins I wanted to use, and it worked: