beagleboard/bb.org-overlays #63: GPIO pins not mapped with NH7C LCD

mvhall, 2018-01-30:
Hi,
Running a fresh 9.1 LXQT from microSD with the Newhaven 7" display. Display works as expected, but the GPIO pins are not available (i.e. P9_12). 
The /sys/class/gpio shows only  export gpiochip0 ...gpiochip96
In dmesh I can't see the allocation of the pins,  when the LCD is disconnected I see lots of these:
[    2.194122] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    2.195133] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0 etc. etc.

Is this a known issue? Is there a configuration that works? Tried all sorts over the last five days.
uEnv.txt only has these lines enabled, the rest is commented out:
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0
uname_r=4.9.45-ti-r57
enable_uboot_overlays=1
enable_uboot_cape_universal=1
cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable

version.sh shows this: 

git:/opt/scripts/:[2ce750d881941c5189db9e189af90517e11c079f]
eeprom:[A335BNLT000C1750BBBG0502]
dogtag:[BeagleBoard.org Debian Image 2017-08-31]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-rc2-00002-g7c9353]
kernel:[4.9.45-ti-r57]
nodejs:[v6.11.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

Any help would be greatly appreciated.
Thanks Maurice

RobertCNelson, 2018-01-30:
This is a planned feature down the road.

mvhall, 2018-01-30:
In other words not possible yet? No workaround? I wanted to switch from RPi3 to BBB for a customer. Gr M

mvhall, 2018-01-30:
I only need three input pins to be allocated. Is it possible to create a dtbo for it?

RobertCNelson, 2018-01-30:
You just need to add the gpio pins' to the Newhaven 7" display overlay.

RobertCNelson, 2018-01-30:
or a 2nd custom overlay.