beagleboard/bb.org-overlays #66: Linux beaglebone 4.9.78-ti-r94

comarius, 2018-02-16:
config-pin -l P9.14 pwm
default gpio gpio_pu gpio_pd gpio_input pwm

root@beaglebone:/sys/class/pwm/pwmchip0# config-pin P9.14 pwm
P9_14 pinmux file not found!
Pin has no cape: P9_14

cat $SLOTS
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,am33xx_pwm
 5: P-O-L-   1 Override Board Name,00A0,Override Manuf,bsana_AIN0_39
 6: P-O-L-   2 Override Board Name,00A0,Override Manuf,bspwm_P9_14_6
 7: P-O-L-   3 Override Board Name,00A0,Override Manuf,bspwm_P9_16

Sys system:

sys/class/pwm# tree
.
├── pwmchip0 -> ../../devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0
├── pwmchip1 -> ../../devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1
├── pwmchip3 -> ../../devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3
├── pwmchip5 -> ../../devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip5
└── pwmchip6 -> ../../devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6

RobertCNelson, 2018-02-16:
@comarius please add the output of: 
```
sudo /opt/scripts/tools/version.sh
```
and your /boot/uEnv.txt

Regards,

comarius, 2018-02-16:
Thank you for such a prompt answer

As today (Feb 16 2018) debian image
uname -a
Linux beaglebone 4.9.78-ti-r94 #1 SMP PREEMPT Fri Jan 26 21:26:24 UTC 2018 armv7l GNU/Linux

 cat /etc/dogtag 
BeagleBoard.org Debian Image 2018-01-28

sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLTGW1ABBGW16050629]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]
kernel:[4.9.78-ti-r94]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single

root@beaglebone:/opt/scripts/tools# cat /boot/uEnv.txt 
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.9.78-ti-r94
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
### mco-mco added audio
disable_uboot_overlay_audio=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

RobertCNelson, 2018-02-16:
Your fighting an old bootloader:

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]
```
Kill it via:

This should be enough to fix it:
```
sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k
```

Otherwise this will defiantly fix it:
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

Regards,

comarius, 2018-02-16:
FIXED. 
Oh!  Many thanks. eMMC u-boot...?!?.  That fix it.

closed 2018-02-16 by comarius

mahdishojaei, 2018-07-26:
hi every one i use beagle bone black and  i have same problem with boot overlay 
my linux (Linux beaglebone 4.9.82-ti-r102 #1 SMP PREEMPT Thu Feb 22 01:16:12 UTC 2018 armv7l GNU/Linux) last image 
i build my own overlay and copy in lib/firmware/ 
next step
<echo SRL > /sys/devices/platform/bone_capemgr/slots>
until this the overlay work good but for use after reboot 
in my directory 
/etc/default 
didn't exit capemgr file so i did it :
first : nano capemgr
and put i t<CAPE=SRL>
second : goto boot/uEnv.txt 
and add 
<cape_enable=bone_capemgr.enable_partno=SRL>
but after reset didn't work my overlay and also in slots by 
cat /sys/devices/platform/bone_capemgr/slots
didn't exit 
please help me !! thanks

RobertCNelson, 2018-07-26:
@mahdishojaei 

What's the output of:

```bash
sudo /opt/scripts/tools/version.sh
```

Regards,

mahdishojaei, 2018-07-29:
@RobertCNelson 
hi thank you for reply 
output is : 

git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLT00C00418BBBK0D80]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
kernel:[4.9.82-ti-r102]
nodejs:[v6.13.0]
pkg:[bb-cape-overlays]:[4.4.20180305.0-0rcnee0~stretch+20180305]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.enable_partno=SRL root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M quiet cape_universal=enable]
dmesg | grep pinctrl-single
[    1.360401] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.361417] gpio-of-helper ocp:cape-universal: ready
END