beagleboard/bb.org-overlays #67: config-pin error on 4.16.0-rc1-bone2 kernel

pdp7, 2018-02-18:
@RobertCNelson I'm running 4.16.0-rc1-bone2 kernel just to give it a try and I noticed that config-pin seems to fail.
```
root@beaglebone:~# uname -a
Linux beaglebone 4.16.0-rc1-bone2 #1 Fri Feb 16 01:27:36 UTC 2018 armv7l GNU/Linux
root@beaglebone:~# config-pin -q p9.12
P9_12 pinmux file not found!
Cannot read pinmux file: /sys/devices/ocp.*/P9_12_pinmux.*/state
```
It looks like the path has changed from `/sys/devices/ocp` to `/sys/devices/platform/ocp`?
```
root@beaglebone:~# find /sys/devices/ -type f |grep P9_12_pinmux 
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/runtime_active_kids
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/runtime_suspended_time
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/autosuspend_delay_ms
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/runtime_enabled
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/runtime_active_time
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/control
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/async
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/runtime_usage
/sys/devices/platform/ocp/ocp:P9_12_pinmux/power/runtime_status
/sys/devices/platform/ocp/ocp:P9_12_pinmux/driver_override
/sys/devices/platform/ocp/ocp:P9_12_pinmux/modalias
/sys/devices/platform/ocp/ocp:P9_12_pinmux/state
/sys/devices/platform/ocp/ocp:P9_12_pinmux/uevent
root@beaglebone:~# cat /sys/devices/platform/ocp/ocp:P9_12_pinmux/state
default
```

RobertCNelson, 2018-02-18:
Yeah I noticed that too on Friday, I think I pushed a fix to u-boot for that, make sure you use the --beta option so it grabs v2018.03-rc2

https://github.com/RobertCNelson/Bootloader-Builder/commit/a00470f20c4fb4196225c38e7c45feadb554cc81

Of course I didn't retest it, as I had to head home.

Regards,

RobertCNelson, 2018-02-18:
(config-pin was heading down the old 3.8.x path)

pdp7, 2018-02-18:
I have upgrade the bootloader with the `--beta` switch:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
[sudo] password for debian: 
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
[sudo] password for debian: 
git:/opt/scripts/:[89f2e9309c2322c15e2bb6b55afd214313d94842]
eeprom:[A335BNLT000C3014BBBK1316]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-rc2-00002-ged1e50]:[location: dd MBR]
kernel:[4.16.0-rc1-bone2]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.871551] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.882586] gpio-of-helper ocp:cape-universal: ready
END
```

However, it seems config-pin still runs into the same problem:
```
debian@beaglebone:~$ config-pin -q p9.12
P9_12 pinmux file not found!
Cannot read pinmux file: /sys/devices/ocp.*/P9_12_pinmux.*/state
```

Something else I need to upgrade?

RobertCNelson, 2018-02-19:
Fixed via:
https://github.com/beagleboard/bb.org-overlays/commit/32722a33bee8addbbcabdd48179686a8ecd52790

closed 2018-02-19 by RobertCNelson