beagleboard/bb.org-overlays #74: Advice on possible CAN issue

Wez50, 2018-03-17:
Hello, i have a feeling i am unable to sent any CAN bus packets to the P9_24 , P9_26 pins as i am not able to work with BB-CAN1-00A0.dtbo active.

virtual Can0 worked and i was able to send packets..... progressing to a real physical Can0 i am unable to see any transmission on the GPOI pins (P9 - 24&26) i monitored both with my oscilloscope while i only expect one to transmit.

for some reason in the capemgr/slots keeps showing "cape-universaln"
and cant get BB-CAN1 active.

I checked my pinmux and that says the GPIO is UNCLAIMED of these pins..... i would think this should show some reference to CAN1.

i just want to ask opinion of people who will know far more than me!! "if i find a way to disable cape-universaln..... would this be a good move? i have seen a reference to it in the uEnv.txt file

any advice will be appreciated, if this is the wrong place to post this question sorry.

Wez50, 2018-03-17:
Sorry missed this information : 
a dmesg provides this insight..

slot#5 : BB-CAN1 conflict P9.24 (#4:cape-unoversaln)
slot#5: Failed varification

RobertCNelson, 2018-03-18:
Did you enable the pinmux when cape universal was enabled?

```
config-pin P9_24 can
config-pin P9_26 can
```

Wez50, 2018-03-18:
thank you for the reply, i am thinking i should have waited until i got a reply !
i managed to force the overlay for BB-CAN1 to show in slots..... i am now able to send data from the pin and decode. and an interesting point....... things started to work when i disabled eth0 (ifconfig eth0 down) ???  

I have not been able to get any information from a candump yet (receive), but the same packet is being transmitted by the servo drive (heartbeat). so i am thinking this might be the issue for candump to initiate. but i will look into this tomorrow.

in your opinion should i revert back to cape universal and try enabling the pinmux? i am quite new to this BBB world so i do like to get the opinion of people who know the details.

again thank you for the reply much appreciated 
Wess

AlexEshoo, 2018-05-07:
I am having issues with CAN now as well. I had previously used CAN on the beaglebone black by following this guide: http://www.thomas-wedemeyer.de/beaglebone-canbus-python.html
However I recently updated my bootloader and now I am having issue with CAN communication.

I have a very similar situation to @Wez50 , I have tried both BB-CAN1 overlay and the cape-universala overlay, but I cannot see any transmissions on P9.24 and P9.26. I have configured the can device as follows:
```bash
config-pin P9.24 can
config-pin P9.26 can
sudo ip link can0 up type can bitrate 125000 loopback on
sudo ifconfig can0 up
```
After doing that, I can open a second terminal and run `./candump can0` (from can-utils) on one of them and use `./cansend can0 5A1#00.01.02.03` and see that the loopback is working appropriately, but I cannot see any activity on the physical pins using a logic analyzer/scope. I have also tried it with turning the loopback off, connecting a second CAN node (through transceiver) but there is still no activity on the physical bus.

vladrotea, 2018-10-11:
Any updates on this? I am experiencing exactly the same issue on a BBB wireless with the latest image (2018-10-7)

nezza, 2018-11-06:
I also can not get CAN to work on the latest image. (2018-10-07)

nezza, 2018-11-07:
Revisiting this, my CAN works fine on 2018-10-07 now after a re-install.

This is all that is needed:
```
config-pin P9.24 can
config-pin P9.26 can
ip link set can1 up type can bitrate 125000
```
Note that can1 is used now.

vladrotea, 2018-12-06:
@nezza Thanks for the update. I tried with the latest version and it works.