beagleboard/bb.org-overlays #76: Unable to configure pins

jackw2050, 2018-03-28:
Unable to configure pins

I am attempting to setup Beagle Bone Wireless using the new U-Boot Overlays.
When I try to configure any of the pins I get the following errors:

GPIO:
	debian@beaglebone:/var/lib/cloud9$ config-pin P8.03 gpio                                                                                                                                                        
	P8_03 pinmux file not found!
	Pin has no cape: P8_03

PWM:
	debian@beaglebone:/var/lib/cloud9$ config-pin P9.14 pwm
	P9_14 pinmux file not found!
	Pin has no cape: P9_14

Following recommendations found for related issues I have updated the kernal and the boot loader.
I have tried configuring many other pins with the same result.
Based on documentation here: https://github.com/cdsteinkuehler/beaglebone-universal-io all pins are enabled and in a default condition.
It looks like Cape Universal is not loading.

What am I doing wrong?

Best regards,

Jack Walker

RobertCNelson, 2018-03-29:
@jackw2050 please add output off

```
sudo /opt/scripts/tools/version.sh
```

jackw2050, 2018-03-30:
git:/opt/scripts/:[70edebd65fe6ea7de16ea8efe3c48b2a4062034a]
eeprom:[A335BNLTBWA51650BBWG1628]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-03-25]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-g254339602c]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.11-rc3-00002-g73df7f]:[location: dd MBR]
kernel:[4.14.29-ti-r40]
nodejs:[v6.13.1]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180322.0-0rcnee0~stretch+20180322]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.945639] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.946716] gpio-of-helper ocp:cape-universal: ready
END

RobertCNelson, 2018-03-30:
@jackw2050 
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.11-rc3-00002-g73df7f]:[location: dd MBR]
```
Your bootloader in eMMC is block u-boot overlays, fix via:

```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

and since your running:
```
kernel:[4.14.29-ti-r40]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo]
```

Change in /boot/uEnv.txt
```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo
to
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
```
to get the correct pru overlay.
```
sudo apt update
sudo apt install --only-upgrade bb-cape-overlays
```

Regards,

jackw2050, 2018-03-30:
It is working correctly now.  Thank you for the help and quick response.

Regards,

Jack

gemlrm, 2018-05-23:
I am having a similar issue with the exception being that I have an NHD-7.0CTP-CAPE-N display attached. Without the display all works as it should. When the display is attached I have the above problem using to use GPIO pins and UART1, UART4, and SPI0. I believe I have done the updates listed above. Below is the listing from sudo /opt/scripts/tools/version.sh. Any help is appreciated.

git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT00C03016BBBK1C29]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.9.78-ti-r94]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180424.0-0rcnee0~stretch+20180424]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    1.411564] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END

gemlrm, 2018-05-23:
OK - I found that by adding the following line to the uEnv.txt the config-pin function works but, the touch screen function does not?

uboot_overlay_addr4=/lib/firmware/cape-universal-00A0.dtbo

RobertCNelson, 2018-05-23:
@gemlrm 

```
kernel:[4.9.78-ti-r94]
```

What you ask is implemented in v4.16.x+ kernel's..

Regards,

gemlrm, 2018-06-22:
OK - I believe I have everything up to date and still no touchscreen activity on the display.  Any more thoughts?

sudo ./version.sh | grep bootloader
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]

uname -a
Linux beaglebone 4.18.0-rc1-bone1 #1 PREEMPT Mon Jun 18 22:32:40 UTC 2018 armv7l GNU/Linux

cat /proc/version
Linux version 4.18.0-rc1-bone1 (root@b9-am57xx-beagle-x15-2gb) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1)) #1 PREEMPT Mon Jun 18 22:32:40 UTC 2018

RobertCNelson, 2018-06-22:
@gemlrm which touchscreen? what's the full output of "version.sh"..

Regards,

gemlrm, 2018-06-22:
Touchscreen is: NHD-7.0CTP-CAPE-N

git:/opt/scripts/:[c190498f25b51d83e452a296de31e2e65883d001]
eeprom:[A335BNLT00C03016BBBK1C29]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7                                                                                                             c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[l                                                                                                             ocation: dd MBR]
kernel:[4.18.0-rc1-bone1]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/cape-universal-00A0.dtb                                                                                                             o]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-0                                                                                                             0A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180619.0-0rcnee0~stretch+20180619]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BB-BONE-NH7C-01, coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    2.680038] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    3.020568] pinctrl-single 44e10800.pinmux: pin PIN13 already requested by panel; cannot claim for ocp:P8_11_pinmux
[    3.031202] pinctrl-single 44e10800.pinmux: pin-13 (ocp:P8_11_pinmux) status -22
[    3.038714] pinctrl-single 44e10800.pinmux: could not request pin 13 (PIN13) from group pinmux_P8_11_default_pin  on device pinctrl-single
[    3.060764] pinctrl-single 44e10800.pinmux: pin PIN12 already requested by panel; cannot claim for ocp:P8_12_pinmux
[    3.071352] pinctrl-single 44e10800.pinmux: pin-12 (ocp:P8_12_pinmux) status -22
[    3.078872] pinctrl-single 44e10800.pinmux: could not request pin 12 (PIN12) from group pinmux_P8_12_default_pin  on device pinctrl-single
[    3.100882] pinctrl-single 44e10800.pinmux: pin PIN9 already requested by panel; cannot claim for ocp:P8_13_pinmux
[    3.111378] pinctrl-single 44e10800.pinmux: pin-9 (ocp:P8_13_pinmux) status -22
[    3.118808] pinctrl-single 44e10800.pinmux: could not request pin 9 (PIN9) from group pinmux_P8_13_default_pin  on device pinctrl-single
[    3.140652] pinctrl-single 44e10800.pinmux: pin PIN10 already requested by panel; cannot claim for ocp:P8_14_pinmux
[    3.151234] pinctrl-single 44e10800.pinmux: pin-10 (ocp:P8_14_pinmux) status -22
[    3.158743] pinctrl-single 44e10800.pinmux: could not request pin 10 (PIN10) from group pinmux_P8_14_default_pin  on device pinctrl-single
[    3.181064] pinctrl-single 44e10800.pinmux: pin PIN15 already requested by panel; cannot claim for ocp:P8_15_pinmux
[    3.191644] pinctrl-single 44e10800.pinmux: pin-15 (ocp:P8_15_pinmux) status -22
[    3.199167] pinctrl-single 44e10800.pinmux: could not request pin 15 (PIN15) from group pinmux_P8_15_default_pin  on device pinctrl-single
[    3.221620] pinctrl-single 44e10800.pinmux: pin PIN14 already requested by panel; cannot claim for ocp:P8_16_pinmux
[    3.232221] pinctrl-single 44e10800.pinmux: pin-14 (ocp:P8_16_pinmux) status -22
[    3.239744] pinctrl-single 44e10800.pinmux: could not request pin 14 (PIN14) from group pinmux_P8_16_default_pin  on device pinctrl-single
[    3.261942] pinctrl-single 44e10800.pinmux: pin PIN11 already requested by panel; cannot claim for ocp:P8_17_pinmux
[    3.272529] pinctrl-single 44e10800.pinmux: pin-11 (ocp:P8_17_pinmux) status -22
[    3.280051] pinctrl-single 44e10800.pinmux: could not request pin 11 (PIN11) from group pinmux_P8_17_default_pin  on device pinctrl-single
[    3.302149] pinctrl-single 44e10800.pinmux: pin PIN35 already requested by panel; cannot claim for ocp:P8_18_pinmux
[    3.312720] pinctrl-single 44e10800.pinmux: pin-35 (ocp:P8_18_pinmux) status -22
[    3.320227] pinctrl-single 44e10800.pinmux: could not request pin 35 (PIN35) from group pinmux_P8_18_default_pin  on device pinctrl-single
[    3.342522] pinctrl-single 44e10800.pinmux: pin PIN8 already requested by panel; cannot claim for ocp:P8_19_pinmux
[    3.353010] pinctrl-single 44e10800.pinmux: pin-8 (ocp:P8_19_pinmux) status -22
[    3.360442] pinctrl-single 44e10800.pinmux: could not request pin 8 (PIN8) from group pinmux_P8_19_default_pin  on device pinctrl-single
[    3.582728] pinctrl-single 44e10800.pinmux: pin PIN105 already requested by ocp:P9_27_pinmux; cannot claim for 2-0038
[    3.593537] pinctrl-single 44e10800.pinmux: pin-105 (2-0038) status -22
[    3.600270] pinctrl-single 44e10800.pinmux: could not request pin 105 (PIN105) from group pinmux_edt_ft5x06_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    2.708791] gpio-of-helper ocp:cape-universal: ready
END

RobertCNelson, 2018-06-22:
Your overlays are fighting over pins...
```
[ 3.031202] pinctrl-single 44e10800.pinmux: pin-13 (ocp:P8_11_pinmux) status -22
[ 3.038714] pinctrl-single 44e10800.pinmux: could not request pin 13 (PIN13) from group pinmux_P8_11_default_pin on device pinctrl-single
[ 3.060764] pinctrl-single 44e10800.pinmux: pin PIN12 already requested by panel; cannot claim for ocp:P8_12_pinmux
```

```
kernel:[4.18.0-rc1-bone1]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/cape-universal-00A0.dtb o]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-0 0A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
```
In /boot/uEnv.txt

Remove/Disable:
```
uboot_overlay_addr4=/lib/firmware/cape-universal-00A0.dtb 
```
and disable
```
uboot_overlay_addr4=/lib/firmware/cape-universal-00A0.dtb 
```

You can use:
```
uboot_overlay_addr4=/lib/firmware/cape-universal-00A0.dtb 
```
Regards,

gemlrm, 2018-06-22:
Success. Thank you very much for you assistance and have a great day.

gemlrm, 2018-06-22:
All seems good until you loose the SPI bus. Display and touch are working, but SPI is now gone. Config-pin is showing correct for uart and spi.  

ls -al /dev/spidev* returns no such file or directory.

Any thoughts on this?

RobertCNelson, 2018-06-22:
You can add them back via:

```
uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo
```

in /boot/uEnv.txt

Regards,

comarius, 2018-10-20:
Hi,
BBG
After doing 

`sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
`
I cannot see /dev/mmcblk1 after reboot 

but I have PWM's just fine:

```
:/sys/class/pwm$ tree
.
├── pwmchip0 -> ../../devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0
├── pwmchip1 -> ../../devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1
├── pwmchip3 -> ../../devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3
├── pwmchip5 -> ../../devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip5
└── pwmchip6 -> ../../devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6

```

My system:

```
sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLTGW1ABBGW16054396]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.9.78-ti-r94]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    1.490980] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END

```
My env:
```

cat /boot/uEnv.txt 
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.9.78-ti-r94
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
### mco-mco added audio
### disable_uboot_overlay_audio=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
### mco-mco

# uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo

###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet omap_wdt.nowayout=0

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

# mco-mco
### optargs=quiet capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN
### optargs=quiet capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN

#dtb=am335x-boneblack-emmc-overlay.dtb

```

RobertCNelson, 2018-10-21:
@comarius answered here: https://groups.google.com/forum/#!category-topic/beaglebone/wMrXlY7R0Uw

Closing thread, new questions = new issue..

closed 2018-10-21 by RobertCNelson

sychen917, 2019-02-19:
i have bought a COMMS cape for my BeagleBone Black wireless. i want to use it to get CAN message. But i have not fund the overlay for this cape.
I want to know, do i need to manual configure Pin P9.24 and P9.26 to CAN. (config-pin P9.24 can).
 i configure the two pin to can, but after reboot, they  automatic change to default pin. please help me.
@RobertCNelson @gemlrm @comarius

RobertCNelson, 2019-02-19:
@sychen917 , please don't directly spam everyone on a closed issue.

In /boot/uEnv.txt use:

```
uboot_overlay_addr4=/lib/firmware/BB-CAN1-00A0.dtbo
```
Should achieve what you want, but open a new issue, i'll be locking this "closed" issue.

Regards,