beagleboard/bb.org-overlays #77: Unable to compile BB-W1-P9.12-00A0.dts

myusername015, 2018-04-12:
I'm trying to edit this file so that I can read the temperature from two different DS18B20 sensors.  Before editing, I tried to just compile this file to see that it works.  I ran the command: 

dtc -O dtb -o BB-W1-P9.12-00A0.dtbo -b 0 -@ BB-W1-P9.12-00A0.dts

I got this error: BB-W1-P9.12.dts:14.1-9 syntax error

Eventually, I'd like to edit the file to connect a sensor to P8.11, but I have to make this work first.  Any advice?

RobertCNelson, 2018-04-12:
Use make, you didn't process the #includes by just calling dtc..

```
voodoo@hades:/opt/github/bb.org/bb.org-overlays$ make src/arm/BB-W1-P9.12-00A0.dtbo 
  DTC     src/arm/BB-W1-P9.12-00A0.dtbo
```
Regards

closed 2018-04-12 by RobertCNelson

myusername015, 2018-04-12:
Thanks!  Next question: is there a way I can read temperature data from P9.12 and P8.11 simultaneously?

RobertCNelson, 2018-04-12:
I've never tried loading two the onewire overlays. Looking at the linux tree, there is one example:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/aspeed-bmc-opp-zaius.dts#n30

so in your 2nd overlay change onewire to onewireX/etc...

"w1-gpio" is a fake peripheral, it's bitbanged, so it'll work with any gpio number...

Regards,