beagleboard/bb.org-overlays #78: What is the correct approach to using pins P9.24 and P9.26 as UART at BBB with Debian Strech 9?

eversonfr77, 2018-04-29:
Hi @RobertCNelson 

I'm trying to use pins P9.24 and P9.26 as a UART on a BBB with Debian Stretch, but I do not know the correct way to set the pins. Previously in Debian Jessie I used the configuration cape_enable = bone_capemgr.enable_partno = BB-UART1 but in the new version apparently this configuration is not valid.
At another time I tried to use config-pin but I would like it to be done during boot.

RobertCNelson, 2018-04-30:
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

```
uboot_overlay_addr4=/lib/firmware/BB-UART1-00A0.dtbo
```

Regards,

RobertCNelson, 2018-05-17:
@hjmagadum2002 valid options are listed here:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

Regards,

agusjumadi, 2018-10-02:
> Hi @RobertCNelson ,
> 
> I made the configuration according to its orientation, when we initialize the system from the flash memory everything works as expected, however, when we initialize from the microSD card the link **/dev/ttyO1** does not exist and the config-pin does not work.
> 
> I renamed the bbb-uEnv.txt file in the root to uEnv.txt so that the boot always occurs from the microSD card.
> 
> Regards,

after rename uEnv.txt my beaglebone cannot boot from sdcard and emmc. cannot flash emmc. please help me to restore uEnv.txt

RobertCNelson, 2018-10-03:
@agusjumadi just reflash, using "bbb-uEnv.txt" as "uEnv.txt" bypass's a lot of boot customization and options, thus your really on your own if you do that..

Regards

agusjumadi, 2018-10-09:
> @agusjumadi just reflash, using "bbb-uEnv.txt" as "uEnv.txt" bypass's a lot of boot customization and options, thus your really on your own if you do that..
> 
> Regards

I tried using serial console, but I type wrong command with "write" so that I erased entire emmc. after that I cannot boot using uboot from serial console anymore. and I tried use boot button on startup using sdcard but still cannot boot. maybe there is other solution. thanks