beagleboard/bb.org-overlays #81: Unable to interface I2S mic with beagle bone

AjinkyaPrabhu, 2018-05-09:
Hi I am unable to interface the i2s microphone (https://www.adafruit.com/product/3421) with the beagle bone green because there are no devices listed with 
<pre> arecord -l
arecord: device_list:270: no soundcards found...
</pre>

RobertCNelson, 2018-05-09:
Did you write an overlay to connect it?

AjinkyaPrabhu, 2018-05-10:
It give a compile error when I try to make it
```
&am33xx_pinmux {
        mcasp1_pins: mcasp1_pins {
             pinctrl-single,pins = <
                /* sink must enable receivers */
                0x1a0 0x23
                /* P9_42 mcasp1_aclkx - bit clock */
                0x1a4 0x23
                /* P9_27 mcasp1_fsx - frame sync */
                0x1a8 0x23
                /* P9_41 mcasp1_axr0 - i2s input */
             >;
        };
};

&mcasp1 {
        #sound-dai-cells = <0>;
        pinctrl-names = "default";
        pinctrl-0 = <&mcasp1_pins>;
        status = "okay";
        op-mode = <0>;
        tdm-slots = <2>;
        num-serializer = <4>;
        serial-dir = < /* 1 TX 2 RX 0 unused */
                2 0 0 0
                >;
        rx-num-evt = <1>;
        tx-num-evt = <1>;
};

/ {
        pcm5102a: pcm5102a {
                #sound-dai-cells = <0>;
                compatible = "ti,pcm5102a";
                status = "okay";
        };

        sound1: sound@1 {
                compatible = "simple-audio-card";
                simple-audio-card,name = "PCM5102a";
                simple-audio-card,format = "i2s";
                simple-audio-card,bitclock-master = <&sound1_master>;
                simple-audio-card,frame-master = <&sound1_master>;
                simple-audio-card,bitclock-inversion;

                simple-audio-card,cpu {
                        sound-dai = <&mcasp1>;
                };

                sound1_master: simple-audio-card,codec {
                #sound-dai-cells = <0>;
                sound-dai = <&pcm5102a>;
                clocks = <&mcasp1_fck>;
                clock-names = "mclk";
                };
        };
};

```