beagleboard/bb.org-overlays #85: config-pin P9_28 pwm doesn't start eCAP2_in_PWM2_out

terachip, 2018-06-10:
Hi There,

In the config-pin utility, there is only one "pwm" option for pin P9_28. However P9_28 is either EHRPWM0_syncouin, or ECAPPWM2 output, which is far more useful. P9_28 can be set to PWM output via "echo pwm2 > state" in /sys/devices/platform/ocp/ocp:P9_28_pinmux, but it would be spiffy if the config-pin utility had both options.

Thanks,
terachip

RobertCNelson, 2018-06-11:
@terachip yeah P9_28 is a pain.. i need to fix the generation script:

On Oct 13, 2017 i fixed this:

https://github.com/beagleboard/bb.org-overlays/commit/0ac13339bab18027fbcdb968e6d1eda5db4c5040#diff-8c06097db3e81f863e4f63cb4f4bac43

and then on Oct 25th i  broke it again..

https://github.com/beagleboard/bb.org-overlays/commit/01b90cf1daad69d8e0c80cd9558de5eb4955fb20#diff-8c06097db3e81f863e4f63cb4f4bac43

DOH!

Regards,

terachip, 2018-06-11:
Lol.   Oops!  Yeah seems like a simple fix though.

RobertCNelson, 2018-06-11:
@terachip okay should be fixed permanently now.. ;)

closed 2018-06-11 by RobertCNelson

terachip, 2018-06-11:
Hey Cool!  Thank you!

On a side note, is there a simple way I can stop my C++ program from having numerous file write permission errors when trying to interface to the GPIOs and PWMs?  The program is written in VS15, cross-compiled with Visual GDB, then deployed to the BBB to run.  However when I first run the program I get a bunch of file write permission errors (as it tries to set up pins, enable as input/output, etc), but if I run the program a second time then it works.  Run it a third time though and it again gives permission errors.  Everything is being run as root.

I'm tempted to chmod 777 the entire /sys/ directory, but that doesn't seem prudent.  I've even added root to the gpio and pwm groups, but that hasn't helped either.  Is there a chmod setting to the executable itself I need to set (even though it belongs to root:root?).  Any help would be most appreciated.

Thank you!

RobertCNelson, 2018-06-11:
Upgrade your version of bb-customizations, it'll add udev rules for gpio and pwm. (although on pwm, full control needs v4.11.x+ based kernel, less then v4.11.x, can still export, but can't do the others..)

```
sudo apt update
sudo apt install --only-upgrade bb-customizations
```

Then make sure you are part of the "gpio pwm" groups...

Regards,

terachip, 2018-06-11:
Hi Robert,

1)  Thank you in advance!

2)  Sounds good regarding the update and bb-customizations.  I try to run in semi-regularly, but since I'm trying to get my project finished, I've also hesitant to with regards to updates breaking pieces of code.  But if it fixes the larger permission issue then c'est la vie.

3)  I'm currently on kernel 4.9.88-r111, which was upgraded via the kernel upgrade script in /opt/tools/scripts/ (something like that).  I did that approximately a week ago.  Maybe two weeks ago now.  Is 4.11.x the new latest, or is that a different version I need to download differently than with the script already on the BBB?

What's perplexing though is that I was able to write to PWM settings via C++ previously (also a couple weeks ago).

4)  Confirmed that root is included in the gpio and pwm groups.  I did that just yesterday in fact, including changing it, immediately rebooting the BBB, and checking that it stuck.  I'll re-check it's still set though.

5)  One (terrible) workaround I've considered is using a bunch of system() commands to interface to and change the GPIOs / PWMs, instead of file-open-file-close.  It does get around the permissions though.  I realize it's not the best, but is there a significant disadvantage to doing this, especially since I'm solely running via root?  I realize there may be a small amount of latency increase, but that will be acceptable (nor do I know if that would actually be the case though).

Thanks,
-terachip

RobertCNelson, 2018-06-11:
bb-customizations doesn't get updated too often:

https://github.com/rcn-ee/repos/commits/master/bb-customizations

we spend more time adding more distro, than updating..

No, 4.11.x is where we first started using a patch to make pwm work under udev.

https://github.com/RobertCNelson/bb-kernel/blob/am33x-v4.11/patches/drivers/pwm/0001-pwm-Create-device-class-for-pwm-channels.patch

Note above patch changes the abi..

see:
https://github.com/StrawsonDesign/Robotics_Cape_Installer/blob/master/library/src/io/gpio.c

https://github.com/StrawsonDesign/Robotics_Cape_Installer/blob/master/library/src/io/pwm.c

These are good for v4.9.x+ kernels.

Regards,

terachip, 2018-06-11:
Sounds good, however you said above (1.5 hours ago) that "although on pwm, full control needs v4.11.x+ based kernel", but then the Robotics_Cape_Installer will work for 4.9, yes?  Just trying to be clear because I have two BBBs I'm working on, both same kernel and rev, but one is a breadboard tester while the other is built into the machine.  So less to keep concurrent the easier :)

Thanks!

terachip, 2018-06-11:
Ok thanks.  I'm running everything on root

terachip, 2018-06-11:
Is it a bad idea just running a bunch of system() commands though?