beagleboard/bb.org-overlays #86: When BB-W1-P9.12-00A0 is loaded in uboot, cape-universal seems not to load

caoimhinuibrian, 2018-06-13:
I’m running 4.17.1 and I’ve made sure that I have the latest boot loader both in eMMC and in the SD card. If I don’t add the uboot_overlay_addr0 line in uEnv.txt, then I can access the GPIOs, but can’t read the temperature. If I add that line then I can see the thermometers but the GPIOs are not available. Unfortunately, I need both at the same time.

RobertCNelson, 2018-06-13:
@caoimhinuibrian please include the output of:

```
sudo /opt/scripts/tools/version.sh
```
Regards,

caoimhinuibrian, 2018-06-13:
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT00C02016BBBK2613]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.17.1-bone2]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    0.946978] pinctrl-single 44e10800.pinmux: 142 pins, size 568
END

caoimhinuibrian, 2018-06-13:
Here is what I see:
root@beaglebone:/home/debian# ls /sys/bus/w1/devices
28-021564bf13ff  28-021564d531ff  28-031564cbceff  28-051670e48dff
28-021564d34dff  28-021564dcccff  28-051670e127ff  w1_bus_master1
root@beaglebone:/home/debian# python /opt/heatpump/read_temps.py
Traceback (most recent call last):
  File "/opt/heatpump/read_temps.py", line 248, in <module>
    setup()
  File "/opt/heatpump/read_temps.py", line 61, in setup
    GPIO.setup("P8_15",GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions.

If I don't load BB-W1-P9.12-00A0.dtbo then the python code runs (but it can't read the thermometers).

RobertCNelson, 2018-06-13:
```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
```
Old older version of u-boot in the eMMC doesn't support this newer feature. I first implemented what you are trying to accomplish with 2018.03

Just run:
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```
and reboot, and it should work..

Regards,

RobertCNelson, 2018-06-13:
and you'll need to update:
```
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0stretch+20180126]
```
via:

```
sudo apt update
sudo apt install --only-upgrade bb-cape-overlays
```

as you'll need the "universal rewrite" change shown here:

https://github.com/beagleboard/bb.org-overlays/commit/675e258e08c24417e815984a6c63aad17bba6c66

Regards,

caoimhinuibrian, 2018-06-13:
Cool! Thanks, works now ....

I had seen your earlier remarks about the bootloader and I had thought I'd fixed that by flashing the eMMC with a recent debian. I guess I hadn't realized HOW recent the bootloader needed to be!

closed 2018-06-13 by caoimhinuibrian