beagleboard/bb.org-overlays #87: U-Boot Overlay

FabioGenesy, 2018-06-14:
Hi,
until now I have used Debian 8.9 release with kernel overlay.
Now I moved on Debian 9.3 that use uboot overlay and I've several dubts.
1) With kernel I can activate overlay at runtime and I can look which overlays are active with the follow command:
`cat /sys/devices/platform/bone_capemgr/slots`
How can I do the same thing now?
The only way I found it is to see the uboot logs via the debug serial port but it is also less clear.
2) To always make active overlays can I use `cape_enable=bone_capemgr.enable_partno=` on /boot/uEnv.txt or in no active this mechanism?
3) I had modified an LCD management overlay that worked with kernel overlay, but the same overlay
    gives me the following error with the uboot overlays:
    ```uboot_overlays: loading /lib/firmware/AM335X-20-00A0.dtbo ... 
      378 bytes read in 173 ms (2 KiB/s)
       uboot_overlays: loading /lib/firmware/BB-BONE-4D7C-01-00A1.dtbo ... 
      3708 bytes read in 314 m(10.7 KiB/s)
     fdt_overlay_apply(): FDT_ERR_BADPHANDLE
     uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
     1105 bytes read in 74 ms (13.7 KiB/s)
     libfdt fdt_check_header(): FDT_ERR_BADMAGIC
     fdt_overlay_apply(): FDT_ERR_BADMAGIC 
    uboot_overlays: loading /lib/firmware/BB-ADC-00A0.dtbo ...
    695 bytes read in 274 ms (2 KiB/s)
    libfdt fdt_check_header(): FDT_ERR_BADMAGIC
    fdt_overlay_apply(): FDT_ERR_BADMAGIC
    uboot_overlays: loading /lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo ...
   2402 bytes read in 309 ms (6.8 KiB/s)
   libfdt fdt_check_header(): FDT_ERR_BADMAGIC
   fdt_overlay_apply(): FDT_ERR_BADMAGIC
   uboot_overlays: cape universal disabled, external cape enabled or detected...
   loading /boot/initrd.img-4.4.91-ti-r133 ...
   9723091 bytes read in 628 ms (14.8 MiB/s)
   debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 
   root=/dev/mmcblk1p1       rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet] ...
   debug: [bootz 0x82000000 0x88080000:945cd3 88000000] ...
   ERROR: Did not find a cmdline Flattened Device Tree
   Could not find a valid device tree
   ** Invalid partition 3 **
   ** Invalid partition 4 **
   ** Invalid partition 5 **
   ** Invalid partition 6 **
   ** Invalid partition 7 **
   starting USB...
   USB0:   Port not available.
   link up on port 0, speed 100, full duplex
   BOOTP broadcast 1```
The system does not start right and I have to restore the debian image every time.
I also tried to decompile, modify and recompile the overlay on the same machine with the command `dtc - @ -O dtb -o foo.dtb -b 0 foo.dts` but it returns me the same error.

Can we help please for this problems and show me the correct way to manage the overlays in this new version?
Thank you very much!

RobertCNelson, 2018-06-14:
@FabioGenesy which image is this?

What change did you make to the overlay? and which compiler did you use.. It's best to just issue "make" in the base repo dir.

```
  378 bytes read in 173 ms (2 KiB/s)
   uboot_overlays: loading /lib/firmware/BB-BONE-4D7C-01-00A1.dtbo ... 
  3708 bytes read in 314 m(10.7 KiB/s)
 fdt_overlay_apply(): FDT_ERR_BADPHANDLE
 uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
 1105 bytes read in 74 ms (13.7 KiB/s)
 libfdt fdt_check_header(): FDT_ERR_BADMAGIC
 fdt_overlay_apply(): FDT_ERR_BADMAGIC 
```

u-boot overlays do not have any error protection, so as soon as the first fails the rest will always fail..

Regards,

FabioGenesy, 2018-06-14:
Hi Robert,
is the log printed on debug usart by UBoot.
To modify my overlay I used this procedure:
I take /lib/firmware/BB-BONE-4D7C-01-00A1.dtbo
Decompile it with the command `dtc -I dtb -O dts BB-BONE-4D7C-01-00A1.dtbo -o BB-BONE-4D7C-01-00A1.dtso`
I modify it and I recompile with the command `dtc -I dts -O dtb BB-BONE-4D7C-01-00A1.dtso -o BB-BONE-4D7C-01-00A1.dtbo`
I replace this file with the same on /lib/firmware

With the kernel overlay I do also the command `sudo update-initramfs -uk 'uname -r' ` and it works correctly.

But now for apply an overlay, for exampe the BB-UART1, what is the correct procedure? And to see which overlay are active?

Thank you very much Robert!

RobertCNelson, 2018-06-14:
@FabioGenesy 

where's your "-@"..

https://github.com/beagleboard/bb.org-overlays/blob/master/Makefile#L135

Regards,

FabioGenesy, 2018-06-14:
these were the commands used with debian 8.9
With the 9.3 I also tried the command `dtc -O dtb -o dtb BB-BONE-4D7C-01-00A1.dtbo -b 0 - @ dtb BB-BONE-4D7C-01-00A1.dtso`
but I have the same error.

RobertCNelson, 2018-06-14:
@FabioGenesy yeah, just use the repo, its' there for a reason, i'm not going to debug dtc command line options..

```
voodoo@hades:/opt/github/bb.org/bb.org-overlays$ make src/arm/BB-BONE-4D7C-01-00A1.dtbo
  DTC     src/arm/BB-BONE-4D7C-01-00A1.dtbo
```
Regards,

FabioGenesy, 2018-06-14:
This is the source overlay modified.
I removed pin 9.14 and pwm to manage backlight manually. 

[BB-BONE-4D7C-01-00A1.txt](https://github.com/beagleboard/bb.org-overlays/files/2102909/BB-BONE-4D7C-01-00A1.txt)

FabioGenesy, 2018-06-14:
Ok, I try, thanks

RobertCNelson, 2018-06-14:
That decompiled version is broken:

```
target = <0xffffffff>;
interrupt-parent = <0xffffffff>;
```

FabioGenesy, 2018-06-14:
I need the magic, like 0xde...., is right?

FabioGenesy, 2018-06-14:
I have this error:
  DTC     src/arm/BB-BONE-4D7C-01-00A1.dtbo
FATAL ERROR: Unrecognized check name "unit_address_vs_reg"
Makefile:141: recipe for target 'src/arm/BB-BONE-4D7C-01-00A1.dtbo' failed
make[1]: *** [src/arm/BB-BONE-4D7C-01-00A1.dtbo] Error 1
Makefile:105: recipe for target 'src/arm/BB-BONE-4D7C-01-00A1.dtbo' failed
make: *** [src/arm/BB-BONE-4D7C-01-00A1.dtbo] Error 2

FabioGenesy, 2018-06-14:
It's the original file, not modified

RobertCNelson, 2018-06-14:
```
FATAL ERROR: Unrecognized check name "unit_address_vs_reg"
```
your version of dtc is too old.. Probally built for 3.8.x era.

Another reason why u-boot overlays failed..

You can build dtc-1.4.4 via:

```
bb.org-overlays$ ./dtc-overlay.sh
```

Or just use mainline dtc..

Regards

FabioGenesy, 2018-06-14:
the version is dtc-1.4.1.
To update I have to do your command or these?
`sudo apt update ; sudo apt install bb-cape-overlays`

FabioGenesy, 2018-06-14:
OK, it works!
Now, I tried to modify it and to put on /lib/firmware

FabioGenesy, 2018-06-14:
In the meantime, can you answer the other questions, Robert?
It is no longer clear to me how to manage (activate and see which ones are active) the overlays with UBoot

Thank you very much!

RobertCNelson, 2018-06-14:
All should be available, here's the options to pass thru /boot/uEnv.txt:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

All the LCD's and anything special is white listed in u-boot:

https://github.com/RobertCNelson/Bootloader-Builder/blob/master/patches/v2018.03/0002-U-Boot-BeagleBone-Cape-Manager.patch#L299-L368

Specially if they need pins from eMMC/hdmi/etc...

Regards,

FabioGenesy, 2018-06-14:
I had read but honestly I understand little.
For example, the rows
```
U-Boot Capes added:
NHD-7.0CTP-CAPE Family
BB-BONE-NH7C-01-A0.dtbo (first added in: 2017.05-rc2-00002-g86566339a9 (Apr 28 2017 - 10:18:57))
```
where are printed? On debug UART? Is the only way to see active capes?

To activate an overlay, like BB-UART2, I have to do this 
`uboot_overlay_addr0=/lib/firmware/BB-UART2-00A0.dtbo` is right?
How do I choose the address?

Another question.
Can I use cape_universal with 4D7C LCD that has several pin in exclusive use or I've a collision?

Thank you in advance!

RobertCNelson, 2018-06-14:
Correct, the only way to debug/etc is thru the debug serial port in u-boot.

At this time, there is no way to pass that info to the kernel.

Correct, that's how you'd active BB-UART2..

But: addr0 -> addr3 are special, they relate to the original 4 eeprom's in the beaglebone cape spec. So if you have a cape plugged, it can default to something..

Your also free to disable/override that..

For example, on a project here:

```
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
debug: process_cape_part_number:[DLPDLCR2000]
debug: process_cape_part_number:[444C50444C435232303030]
BeagleBone: cape eeprom: i2c_probe: 0x57: /lib/firmware/DLPDLCR2000-00A0.dtbo [0xd15b80f]
```
addr3 is going to load: /lib/firmware/DLPDLCR2000-00A0.dtbo

In this case, I'm still "free" to use, addr0, addr1, addr2, addr4, addr5, addr6, addr7..

If the pin's collide, you need to take care of that.

For example, i needed P9.30 for spidev:

https://eewiki.net/display/linuxonarm/FLIR+Lepton+on+BeagleBone+Black+and+Green#FLIRLeptononBeagleBoneBlackandGreen-LCD7Setup

Regards,

sergpenshin, 2019-03-16:
Hi, everyone! I have a similar problem. I can't make U-Boot_Overlays in Debian 9.5 work propely. 
for example, i try to add the cape ADAFRUIT-SPI0 to /boot/uEnv.txt how
_uboot_overlay_addr4=/lib/firmware/ADAFRUIT-SPI0-00A0.dtbo_ but it doesn't work.
>_ls -al /dev/spidev*_
> ls: cannot access '/dev/spidev*': No such file or directory

And it's the same for other capes.
Also I tried to do this how 
###Custom Cape
dtb_overlay=/lib/firmware/ADAFRUIT-SPI0-00A0.dtbo
but result is the same.

> _debian@beaglebone:~$ cat /sys/devices/platform/bone_capemgr/slots_
>  0: PF----  -1
>  1: PF----  -1
>  2: PF----  -1
>  3: PF----  -1

U-Boot Overlays are enable:
> ###Master Enable
> enable_uboot_overlays=1

If i do 
_root@beaglebone:~# echo ADAFRUIT-SPI0 > /sys/devices/platform/bone_capemgr/slots_
it's work 

> 0: PF----  -1
>  1: PF----  -1
>  2: PF----  -1
>  3: PF----  -1
>  4: P-O---  -1 Override Board Name,00A0,Override Manuf,ADAFRUIT-SPI0

but obviously until reboot
Please help!!! What i do wrong?

RobertCNelson, 2019-03-16:
@sergpenshin ,

please post output of:

sudo /opt/scripts/tools/version.sh

Regards,

sergpenshin, 2019-03-16:
> git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
> eeprom:[A335BNLT00C04016BBBK188C]
> model:[TI_AM335x_BeagleBone_Black]
> dogtag:[BeagleBoard.org Debian Image 2018-10-07]
> bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
> bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.10-00001-g143c9ee]:[location: dd MBR]
> kernel:[4.14.71-ti-r80]
> nodejs:[v6.17.0]
> uboot_overlay_options:[enable_uboot_overlays=1]
> uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/ADAFRUIT-SPI0-00A0.dtbo]
> uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
> uboot_overlay_options:[enable_uboot_cape_universal=1]
> pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
> pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
> pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
> pkg:[kmod]:[23-2rcnee1~stretch+20171005]
> pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
> pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
> groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
> cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
> dmesg | grep pinctrl-single
> [    1.106017] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
> dmesg | grep gpio-of-helper
> [    1.107201] gpio-of-helper ocp:cape-universal: ready
> END
>

RobertCNelson, 2019-03-17:
@sergpenshin , u-boot installed on the eMMC is too old to support u-boot overlays;

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.10-00001-g143c9ee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
```
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

Regards

sergpenshin, 2019-03-18:
Hi Robert!
Thanks a lot for your advise. i updated the bootloader on the eMMC and it seems to work.
I had not idea that bootloader on the eMMC influencing sistem on microSD. 
Now if i do ls -al /dev/spidev* i have

> debian@beaglebone:~$ ls -al /dev/spidev*
> crw-rw---- 1 root spi 153, 0 Mar 18 07:53 /dev/spidev1.0
> crw-rw---- 1 root spi 153, 1 Mar 18 07:53 /dev/spidev1.1
> crw-rw---- 1 root spi 153, 2 Mar 18 07:53 /dev/spidev2.0
> crw-rw---- 1 root spi 153, 3 Mar 18 07:53 /dev/spidev2.1

and one more question, what is the SPI0 from them,  if in /boot/uEnv.txt is uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo

Before it was clear what of cape in slots from cat /sys/devices/platform/bone_capemgr/slots.  Now (how it mentioned in https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays) slots file is disable. How can i know is cape enable or not now?

Sorry for my bothering

RobertCNelson, 2019-03-18:
@sergpenshin mainline updated the node name, /dev/spidev1.x = SPIDEV0, /dev/spidev2.x = SPIDEV1 overlay..

Don't enable both BB-SPI0-MCP3008-00A0.dtb and BB-SPIDEV0-00A0.dtbo, pick one.

Regards,

sergpenshin, 2019-03-20:
Thank you Robert. SPI works. I tried to start my Lepton3 and everything OK

closed 2019-03-20 by RobertCNelson