beagleboard/bb.org-overlays #91: Issues creating a PPS overlay

tehkillerbee, 2018-08-07:
I have been trying to create a PPS GPIO overlay in conjunction with the existing UART4 overlay (BB-UART4-00A0).

My BBB is running the latest kernel, latest uboot at this point. I have also made sure that the bb-cape-overlays package is up to date (https://github.com/beagleboard/bb.org-overlays/issues/76)..

I am able to enable UART4, by adding it as follows and it works perfectly.
###Additional custom capes
uboot_overlay_addr4=/lib/firmware/BB-UART4-00A0.dtbo

Then I tried adding my own PPS overlay (based on existing examples), containing the following:
```
/*
 *
 */
/dts-v1/;
/plugin/;

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black";

	/* identification */
	part-number = "BB-PPS-P9_12";
	version = "00A0";
	
	/* state the resources this cape uses */
	exclusive-use =
	/* the pin header uses */
	"P9.12";	/* pps gpio */

	fragment@0 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			pps_pins: pinmux_pps_pins {
				pinctrl-single,pins = <
					0x078	0x27	/* gpmc_a0.gpio1_28, INPUT | PULLDIS | MODE7 */
				>;
			};
		};
	};

	fragment@2 {
		target = <&ocp>;
		__overlay__ {
			pps {
				compatible = "pps-gpio";
				status = "okay";
				pinctrl-names = "default";
				pinctrl-0 = <&pps_pins>;

				gpios = <&gpio1 28 0>; /*really gpio1!*/
				assert-falling-edge;
			};
		};
	};
};

```

The above is based largely on http://mythopoeic.org/BBG-NTP/NF3H-PPS-00A0.dts. I took note that [mythopoeic.org] mentioned:

> Apparently at some point in the past, the kernel PPS driver used one-based numbering for the GPIO bank, while everything else used zero-based numbering. For example, you'd use "gpio1_28" in the reservations, but "gpio2 28 0" in the gpios clause for the PPS driver. In the 4.x kernels, that is no longer the case; everything uses zero-based numbering. Much less confusing -- unless you're looking at examples that use the old method.)

So I therefore used the same approach; i.e. zero-based numbering for all variables. I compiled it and copied it to /lib/firmware:

dtc -O dtb -o BB-PPS-P9_12-00A0.dtbo -b 0 -@ BB-PPS-P9_12-00A0.dts
sudo cp BB-PPS-P9_12-00A0.dtbo /lib/firmware/BB-PPS-P9_12-00A0.dtbo

I then added it to the list of custom capes, as follows:
uboot_overlay_addr5=/lib/firmware/BB-PPS-P9_12-00A0.dtbo

When I reboot the BBB, I get the following errors:

 dmesg | grep pps
[    0.613866] pps_core: LinuxPPS API ver. 1 registered
[    0.613874] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[   29.974000] pinctrl-single 44e10800.pinmux: pin PIN30 already requested by ocp:P9_12_pinmux; cannot claim for ocp:pps
[   30.025884] pinctrl-single 44e10800.pinmux: pin-30 (ocp:pps) status -22
[   30.065959] pinctrl-single 44e10800.pinmux: could not request pin 30 (PIN30) from group pinmux_pps_pins  on device pinctrl-single
[   30.142863] pps-gpio ocp:pps: Error applying setting, reverse things back
[   30.180262] pps-gpio: probe of ocp:pps failed with error -22

I have not added any other cape overlays at this point but as I understand from the above, my overlay is interfering with an existing one. I understand that an "universal cape overlay" is automatically added, making it possible to control all GPIO (including the PIN30). At first, I thought that could be interfering with my overlay so I set on to disable it. But what is the correct way to handle pins overlapping with the universal cape overlay? Do you disable it completely? If so, how is that done correctly?

I tried setting the "enable_uboot_cape_universal=0" in "/boot/uEnv.txt", but it does not make any difference. I still get the exact same error message. 

On http://mythopoeic.org/beaglebone-green-time-server/, it is mentioned that you can remove " cape_universal=enable" from the cmdline in "/boot/uEnv.txt". As my cmdline does not contain cape_universal, I tried adding cape_universal=disable instead, although that did not make any difference either..

I hope this post is not too irrelevant. Most of the information I have been able to find regarding PPS on BBB is quite old by now.

RobertCNelson, 2018-08-07:
Your just missing the "new" cape universal rewrite, pinmux disable..

Regards,

```bash
/*
 *
 */
/dts-v1/;
/plugin/;

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black";

	/* identification */
	part-number = "BB-PPS-P9_12";
	version = "00A0";
	
	/* state the resources this cape uses */
	exclusive-use =
	/* the pin header uses */
	"P9.12";	/* pps gpio */

	/*
	 * Free up the pins used by the cape from the pinmux helpers.
	 */
	fragment@0 {
		target = <&ocp>;
		__overlay__ {
			P9_12_pinmux { status = "disabled"; }; 
		};
	};

	fragment@1 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			pps_pins: pinmux_pps_pins {
				pinctrl-single,pins = <
					0x078	0x27	/* gpmc_a0.gpio1_28, INPUT | PULLDIS | MODE7 */
				>;
			};
		};
	};

	fragment@2 {
		target = <&ocp>;
		__overlay__ {
			pps {
				compatible = "pps-gpio";
				status = "okay";
				pinctrl-names = "default";
				pinctrl-0 = <&pps_pins>;

				gpios = <&gpio1 28 0>; /*really gpio1!*/
				assert-falling-edge;
			};
		};
	};
};
```

tehkillerbee, 2018-08-08:
Brilliant! I was previously using some older .dts files as reference when setting up the PPS, so I never realized that it is necessary to perform pinmux disable. After looking into the latest .dts sources in this repository (https://github.com/beagleboard/bb.org-overlays/tree/master/src/arm), it makes good sense.

Thank you very much, naturally the PPS works now (Both on P9_12 and P9_15):
debian@beaglebone:~$ dmesg | grep pps
[    0.613860] pps_core: LinuxPPS API ver. 1 registered
[    0.613868] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[   77.655840] pps pps0: new PPS source ocp:pps.-1
[   77.655954] pps pps0: Registered IRQ 108 as PPS source

I have uploaded both my .dts files as a .zip, in case anyone is interested in them.

[BB-PPS.zip](https://github.com/beagleboard/bb.org-overlays/files/2269268/BB-PPS.zip)

leearmstrong, 2018-11-28:
This is a great help and I am trying to create a Uboot overlay myself having come across from kernel ones!

Our cape uses P9_27 for the PPS pin.  Do you know the pinctrl for those?

RobertCNelson, 2018-11-28:
@leearmstrong 

```
uboot_overlays: loading /lib/firmware/BB-PPS-P9_27-00A0.dtbo ...
1101 bytes read in 81 ms (12.7 KiB/s)
failed on fdt_overlay_apply(): FDT_ERR_BADOVERLAY
```

when you build your overlay, stick in this repo's ./src/arm/ directory and use the built-in makefile..

Regards,

leearmstrong, 2018-11-28:
Thanks @RobertCNelson I will give that a go.

I ran `dtc -O dtb -o output.dtbo -b 0 -@ input.dts`.  Would that have been the issue then?

RobertCNelson, 2018-11-28:
@leearmstrong what version of dtc?

Current tested and support dtc tags are: v1.4.4, v1.4.6, v1.4.7 and mainline

https://github.com/beagleboard/bb.org-overlays/blob/master/jenkins_build.sh#L36-L70

Regards,

leearmstrong, 2018-11-28:
I had just installed via...
https://raw.githubusercontent.com/RobertCNelson/tools/master/pkgs/dtc.sh

RobertCNelson, 2018-11-28:
oh god, that's for 3.8.x... for v4.0.x+ kernel/u-boot overlays we use "mainline" dtc, not the old 3.8.x dtc hacks..

i'll update the Readme...

leearmstrong, 2018-11-28:
Thanks, it's been a while and that would explain the problem.

I suppose that is the trouble with progress.  So much stuff around!

Will report back but will try...

1. Cloning this repo.
2. Adding my dts file
3. Running the makefile
4. Editing the uEnv to use the file

leearmstrong, 2018-11-28:
@RobertCNelson That worked great thank you.  Seems I should not trust my older notes!

What is the rule now with updating the kernel?  Will it also migrate across my custom dtbo when I upgrade? (Sadly can't test this as using the latest image as of today)

Also is there a way to use a fresh BBB without all the dev/build tools on and create a .deb to install all of these?  I used to remember that option before.

RobertCNelson, 2018-11-28:
@leearmstrong Well, the kernel and overlays are separate... So long as you use an un-used overlay name under /lib/firmware/, all the bb-cape-overlays upgrades will ignore it.

Otherwise, you could also store your overlay elsewhere, just update the path to it used in /boot/uEnv.txt (this is why all the examples start with /lib/firmware/*.dtbo, it's a hint that it can be really anywhere in the file system).

For kernel updates, really any of the latest v4.4.x, v4.9.x, v4.14.x, or v4.19.x based kernels will work "without" changing the overlay:

https://github.com/beagleboard/bb.org-overlays/blob/master/readme.md

Regards,

leearmstrong, 2018-11-28:
Ah ok thanks, so the entire file system is mounted and accessible at that point?  From memory it didn’t used to be and they had to be in a specific location? /boot ??

I am using the latest 4.14 I think in the Debian 9.5 image.

RobertCNelson, 2018-11-28:
Well, we've only had one partition for a couple years now, most of the time everything defaulted under /boot/* the overlays where the first thing to be referenced outside of /boot/* as they are under /lib/firmware/*

closed 2019-03-16 by RobertCNelson