beagleboard/bb.org-overlays #92: Audio Cape Rev. B

larsh666, 2018-09-10:
Hello,
i am having problems getting the Audio Cape Rev. B working. I have made an own PCB where the TLV320AIC3104 is exactly connected as it is on the Audio Cape Rev.b 
My version.sh shows the following:

```
git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLT000C1821BBBG1094]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-BONE-AUDI-02-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180803.0-0rcnee0~stretch+20180804]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.050322] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.051916] gpio-of-helper ocp:cape-universal: ready
END

```

larsh666, 2018-09-10:
The only signals which shows a valid response is MCLK. All other Pins are on a logic '0' Level.

Any ideas where to start Debugging ?

best regards Lars

closed 2018-09-19 by larsh666

deeplyembeddedWP, 2019-10-27:
@larsh666, any insights on what was the issue and how did you manage to fix it?