beagleboard/bb.org-overlays #99: device tree overlay for skywire bbb cape not loading properly?!

cuca94, 2018-12-11:
after installing the latest Debian version my Skywires bbb cape device tree overlay is not loading correctly anymore. i want to use the can transceiver of the board, but can0 is obviously not reachable.
on the old kernel the capes device tree overlay is loaded automatically.

the cape seems to be detected, but not loaded. running kernel 4.4.x works without problems. the difference which i figured out with the new kernel is the missing "L" for "loaded" in the status message of the capemgr.

```
debian@beaglebone:~$ cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: P-----  -1 Skywire Beaglebone Cape,00D0,Nimbelink,NL-AB-BBBC
```

is there a way to fix this problem? 
may the .dts file is compiled with an outdated compliler?

here is the .dts file which should be read from capes EEPROM:

```
* Copyright (C) 2014 Nimbelink, Corp <product.support@nimbelink.com>
*
* This program is free software; you can redistribute it and/or modify
* it under the terms of the GNU General Public License version 2 as
* published by the Free Software Foundation.
*
* This is a device tree overlay file for Nimbelinks Skywire Beaglebone Black Cape. 
* This code initializes CAN1, UART2, UART4, and GPIOs P8.8 and P8.9 for use with the capes default operating mode.
* This overlay file was developed from referencing the projectM device tree files (https://github.com/gntoni/projectM/tree/master/devTree)
* and from the embedded things.com CANBUS tutorial (http://www.embedded-things.com/bbb/enable-canbus-on-the-beaglebone-black/) 
* 
*/

/dts-v1/;
/plugin/;

/{
	compatible = "ti,beaglebone", "ti,beaglebone-black";
	part-number = "NL-AB-BBBC";
	version = "00D0";

	/* State resources this cape uses */
	exclusive-use = 
		/* the pin header and pins used */
		"P9.13", /* UART4 TXD */
		"P9.11", /* UART4 RXD */
		"P9.21", /* UART2 TXD */
		"P9.22", /* UART2 RXD */
		"P9.24", /* CAN1 RXD */
		"P9.26", /* CAN1 TXD */
		"P8.8", /*SKYWIRE ON-OFF*/
		"P8.9", /*SKYWIRE DTR*/
		
		 /* the hardware IP use */
		"uart4_txd_mux2",
		"uart4_rxd_mux2",
		"uart2_txd",
		"uart2_rxd",
		"CAN1_rxd",
		"CAN1_txd",
		"gpio2_3",
		"gpio2_5";

	/*Mux the Pins */

	fragment@0 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			/*CONFIGURE CAN1*/
			can1_pins: pinmux_can1_pins {
				pinctrl-single,pins = <
					0x180 0x12  /* d_can1_tx, SLEWCTRL_FAST | INPUT_PULLUP | MODE2 */
					0x184 0x32  /* d_can1_rx, SLEWCTRL_FAST | RECV_ENABLE | INPUT_PULLUP | MODE2 */
					>;
			};
			/*CONFIGURE UART2*/
			uart2_pins: pinmux_uart2_pins {
				pinctrl-single,pins = <
					0x154 0x01 /* P9_24 = GPIO0_15 = spi0_d0.uart2_txd, OUTPUT */
					0x150 0x21 /* P9_26 = GPIO0_14 = spi0_sclk.uart2_rxd, INPUT */
					>;
			};
			/*CONFIGRE UART4*/
			uart4_pins: pinmux_uart4_pins {
				pinctrl-single,pins = <
					0x074 0x06 /* P9_13 = GPIO0_31 = gpmc_wpn.uart4_txd_mux2, OUTPUT */
					0x070 0x26 /* P9_11 = GPIO0_30 = gpmc_wait0.uart4_rxd_mux2, INPUT */
					>;
			/*CONFIGURE GPIO*/
			};
			gpio_pins: pinmux_gpio_pins{
				pinctrl-single,pins = <
					0x094 0xf		/* P8.8, gpio2_3, SLEWCTRL_FAST | PULLUP_DISABLED | MODE7*/
					0x09c 0xf		/* P9.9, gpio2_5, SLEWCTRL_FAST | PULLUP_DISABLED | MODE7*/
					>;
			};
		};
	};

/*=============Enable the device ==================*/

/*============Connect the device to user interface driver ==============*/
/*To connect the device with a user interface, 
/*specify the “compatible” field in the “ocp” section. 
/*ocp stand for On Chip Peripherals, 
/*BTW. The compatible field should match the driver you want to use.*/

	/*Turn on CAN1*/
	fragment@1 {
		target = <&dcan1>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;

			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&can1_pins>;
		};
	};

	/*Turn on UART2*/
	fragment@2 {
		target = <&uart3>; /* really uart2 */
		__overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <&uart2_pins>;
			status = "okay";
		};
	};

	/*Turn on UART4*/
	fragment@3 {
		target = <&uart5>; /* really uart4 */
		__overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <&uart4_pins>;
			status = "okay";
		};
	};

	/*Turn on GPIO*/
	fragment@4 {
		target = <&ocp>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <1>;
			gpio {
				compatible = "skywire-gpio";
				status 	= "okay";
				pinctrl-names   = "default";
				pinctrl-0       = <&gpio_pins>;

				/*declare the gpios*/
				SKYWIREDTR {
					gpio-name = "SKYDTR";
					gpio = <&gpio3 5 0x00>; /*GPIO3 is GPIO2*/
					output;
					init-low;
				};

				SKYWIREONOFF {
					gpio-name = "SKYONOFF";
					GPIO = <&gpio3 3 0x00>;
					output;
					init-high;
				};
			};
		}; 
	};
};
```

cuca94, 2018-12-11:
I have read this article several times but i dont get how to use it. Do i have to use uboot_overlay_3 and insert the skywire .dtbo?

RobertCNelson, 2018-12-11:
The NL-AB-BBBC has an eeprom, it's going to load https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/NL-AB-BBBC-00D0.dts via u-boot 'automatically' as long as u-boot overlays is enabled....

Regards,

cuca94, 2018-12-11:
But u-boot overlays is enabled by default on a fresh debian 9.5, isnt it? So what should i „enable“?

RobertCNelson, 2018-12-11:
@cuca94 
please run and report the output of:

```
sudo /opt/scripts/tools/version.sh
```

Regards,

cuca94, 2018-12-12:
`git:/opt/scripts/:[e9bcff232834702c1c810710706ee815d77b080b]
eeprom:[A335BNLT00C00916BBBK0CEC]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-11-25]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gd5b4c4b656]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.10-00001-g143c9ee]:[location: dd MBR]
kernel:[4.14.79-ti-r84]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20181120.0-0rcnee0~stretch+20181120]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20181123.0-0rcnee0~stretch+20181124]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.045325] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.046499] gpio-of-helper ocp:cape-universal: ready
END`

RobertCNelson, 2018-12-12:
Yeap, just clear out the old version of u-boot via:

```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

and the cape should be auto-detected..

closed 2019-06-13 by cuca94