beagleboard/bb.org-overlays #100: Pin multiplexing (pinmux) file not found

dihelium, 2018-12-22:
When I try to do:
config-pin P9_28 pwm
or, for that matter, any state with P9_28, I get the following error:
P9_28 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P9_28_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_28_pinmux/state

For other pins, such as P9_14, P9_16, etc. these commands work perfectly fine.

I am aware of your answer to issue #76 but your answer doesn't seem to help my case.
The output of 
sudo /opt/scripts/tools/version.sh   is written below:
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT00C02018BBBK2D0A]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
device-tree-override:[dtb=am335x-boneblack-overlay.dtb]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20181120.0-0rcnee0~stretch+20181120]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.107019] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.118877] gpio-of-helper ocp:cape-universal: ready
END

RobertCNelson, 2018-12-22:
```uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo]```

P9.28 is spi1_cs0...

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-SPIDEV1-00A0.dts#L30

The "pinmux" doesn't work as you are using it for spi1_cs0...

Regards,

closed 2018-12-22 by RobertCNelson

dihelium, 2018-12-23:
What should I change it to? Is it supposed to be: BB-PWM2-00A0 ?