beagleboard/bb.org-overlays #102: Pinmux file not found for P9_12

milesfrain, 2018-12-29:
I looked through the related issues: #100 #67 #76 #62 and tried the overlay and uboot update suggestions along with rebooting, but I'm still stumped on what is claiming P9_12. Is there a good technique for figuring out what is claiming a pin?
Some suspicions are overlays from `librobotcontrol`, although I'd expect this package to have no impact if I am not using any attached capes. I may also try upgrading from kernel `4.14` to `4.16`.

```
debian@beaglebone ~> config-pin -q P9_12
P9_12 pinmux file not found!
Cannot read pinmux file: /sys/devices/platform/ocp/ocp*P9_12_pinmux/state
```

```
debian@beaglebone ~> sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[0ca42da8c7e3d81e9cbab71b1e2b0fad00bf79ca]
eeprom:[A335BNLTGW1ABBGW18041596]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-gd5b4c4b656]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20181228.0-0rcnee0~stretch+20181228]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.115450] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.126566] gpio-of-helper ocp:cape-universal: ready
END
```

```
debian@beaglebone ~> grep -v '#' /boot/uEnv.txt 
uname_r=4.14.71-ti-r80
enable_uboot_overlays=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_adc=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
enable_uboot_cape_universal=1
cmdline=coherent_pool=1M net.ifnames=0 quiet
```

RobertCNelson, 2018-12-29:
@milesfrain P9_12 on the Green Wireless is Bluetooth Enable... Here is the full list of "special" pins on the Green Wireless..

https://elinux.org/EBC_Exercise_10b_Green_Wireless_Pins_Used

If you don't need Bluetooth, and just need it as a gpio out, just use the led name..

Otherwise, here's the bits you'll need to nuke:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BBGW-WL1835-00A0.dts#L23

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BBGW-WL1835-00A0.dts#L41-L45

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BBGW-WL1835-00A0.dts#L123-L133

Regards,

milesfrain, 2018-12-30:
Thank you very much for the prompt response.

I was looking in the wrong places for notes about the pin conflicts on the BBGW.
Bummer that they omitted this detail in [System Reference Manual](http://wiki.seeedstudio.com/BeagleBone_Green_Wireless/#pin-map).
I was previously assuming the BBGW expansion header conflicts ([P8](http://exploringbeaglebone.com/wp-content/uploads/resources/BBBP8Header.png), [P9](http://exploringbeaglebone.com/wp-content/uploads/resources/BBBP9Header.png)) were identical to the BBB (excluding HDMI).

The [BBGW schematic](https://github.com/SeeedDocument/BeagleBone_Green_Wireless/tree/master/resources/BeagleBone_Green%20Wireless_V1.0_SCH_20160314.pdf) has some clues about `P8_12 --> GPIO1_28 --> BT_EN`. So I'll keep this in mind for investigating future pin conflicts. Are there any other recommended strategies for troubleshooting pin conflicts? Is there some sort of log to show what component calls dibs on each pin?

Now I'm confused about the purpose of this BT_EN LED. I assume it should be lit when BT is enabled. I have BT powered-down, but the LED stays on. I can successfully toggle the WIFI LED with the `connmanctl enable/disable wifi` command, and so I'm not sure why the BT led is stuck on.
```
debian@beaglebone ~> connmanctl technologies
/net/connman/technology/wifi
  Name = WiFi
  Type = wifi
  Powered = False
  Connected = False
  Tethering = False
/net/connman/technology/bluetooth
  Name = Bluetooth
  Type = bluetooth
  Powered = False
  Connected = False
  Tethering = False
```

milesfrain, 2018-12-30:
@RobertCNelson
Is there a mechanism to enable/disable bluetooth that relies on `BT_EN_SOC` from `GPIO1_28`?
I'm surprised that for WLAN/WIFI, `connmanctl` drives `GPIO0_26` which is attached to `WL_EN`/`WLAN_EN_SOC` and the indicator LED, but bluetooth state is controlled through a path that bypasses the indicator LED.

RobertCNelson, 2018-12-30:
Here's the updated binding..

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/Documentation/devicetree/bindings/net/ti-bluetooth.txt?id=07c7edd0db441f83eeee087470add5be280cc018

This binding is newer then "v4.4.x" which is the minimal this repo currently supports.  When buster get's tagged this spring, we can eol v4.4.x (and maybe v4.9.x) along with jessie. Then bump the minimal to v4.9.x/v4.14.x..

Regards,

closed 2018-12-31 by milesfrain