beagleboard/bb.org-overlays #105: Cannot make the PB-SPI1-ETH-WIZ-CLICK work

mkudlacek, 2019-01-05:
Hello all,
 I'm trying to connect two Wiznet W5500 to the BeagleBone Black - first to the SPI0 and second to the SPI1. Unfortunately I cannot get past the loading of the DT overlay. I found out the problem lies in the pin being already registered by OCP, but I don't know how to disable the OCP. I checked the [PB-SPI0-ETH-WIZ-CLICK.dts](https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/PB-SPI0-ETH-WIZ-CLICK.dts) and it looks like it should be done automatically in the fragment@0...

I don't plan to use anything else from the P8 and P9 pins other than the SPI0 and SPI1, so I disabled all the other overlays: video, audio, wireless and adc. I also disabled the PRU since I don't plan to use it. This is my uEnv.txt:

```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.79-ti-r84
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
uboot_overlay_addr0=/lib/firmware/PB-SPI0-ETH-WIZ-CLICK.dtbo
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable overlayroot=tmpfs

##enable Generic eMMC Flasher:
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

I also tried to disable the universal cape by setting the `enable_uboot_cape_universal=0` and removing the `cape_universal=enable` from kernel parameters. Didn't help either. Here's the output of the version.sh script:

```
eeprom:[A335BNLT00C02118BBBK2325]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Debian Image 2018-12-10]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-gd5b4c4b656]:[location: dd MBR]
kernel:[4.14.79-ti-r84]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/PB-SPI0-ETH-WIZ-CLICK.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190104.0-0rcnee0~stretch+20190104]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.106806] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.227912] pinctrl-single 44e10800.pinmux: pin PIN84 already requested by ocp:P9_22_pinmux; cannot claim for 48030000.spi
[    1.239147] pinctrl-single 44e10800.pinmux: pin-84 (48030000.spi) status -22
[    1.246264] pinctrl-single 44e10800.pinmux: could not request pin 84 (PIN84) from group pinmux_pb_spi0_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    1.118697] gpio-of-helper ocp:cape-universal: ready
END
```

Thanks for helping me out in advance,
Martin

RobertCNelson, 2019-01-05:
@mkudlacek PB = PocketBeagle

For the BeagleBone look at:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-mBC3-ETH-WIZ-CLICK.dts

but even that 'might' need some tweaks for your hardware layout.. so update accordingly.

Regards,

RobertCNelson, 2019-01-11:
@mkudlacek sorry, i missed your reply.  No on this board the SPI and USB are not shared..  On the PocketBeagle i pushed this interface to 24Mhz

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-mBC3-ETH-WIZ-CLICK.dts#L95

I believe the am335x's max is around 48Mhz, so try bumping it up in 4Mhz delta's..  For most of our BBB overlays, we don't push it over 16Mhz...

Regards,

mkudlacek, 2019-02-10:
Hi @RobertCNelson,
 I finally got through a lot of testing. Cranking up the BBB overlays up to 48MHz works and it did have quite linear increase in the performance:

 8MHz: 3.73 Mbps
12MHz: 4.59 Mbps
16MHz: 5.33 Mbps
20MHz: 5.33 Mbps
24MHz: 6.42 Mbps
28MHz: 6.43 Mbps
32MHz: 6.40 Mbps
36MHz: 6.39 Mbps
40MHz: 6.42 Mbps
44MHz: 6.45 Mbps
48MHz: 7.44 Mbps

The test was done with BBB equipped with mikroBUS cape and original MikroElektronika ETH WIZ Click. BBB act as a `iperf` TCP server and was connected to another BBB's Gbit port which act as a `iperf` client. Each test was run for 1800s. What baffels me how I got to the 8Mbps throughput in the first test with the default 24MHz SPI speed.

Anyway, another problem arose. Not all of the possible 4  ETH Wiz Clicks work at one time. I conducted this test:

On position 1 and 2 -> work both
On position 1 and 3 -> pos #1 works, pos #3 doesn't show up
On position 1 and 4 -> works only pos #4

On position 2 and 3 -> pos #2 works, pos #3 doesn't show up
On position 2 and 4 -> work both

On position 3 and 4 -> pos #3 doesn't show up, pos #4 works

On position 3 only -> doesn't show up in the system

I am especially anxious to get it working on the position 1 and 3 to have the ethernet interfaces on different SPI lanes thus maximizing throughput / minimizing the latency. I am trying to use this setup for  Codesys EtherCAT master with redundant connections.

Thank you for helping me out on this one.
Martin

RobertCNelson, 2019-02-10:
@mkudlacek pos 3 needs a soldered jumper moved:

```Make sure to change the J7/J8 Jumpers to SPI Mode...```

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-mBC3-ETH-WIZ-CLICK.dts#L11

Regards,

closed 2019-03-16 by RobertCNelson