beagleboard/bb.org-overlays #106: Can't use SPI interface

himanshujha199640, 2019-01-08:
I wish to use the SPI interface to use Bosch BME680 sensor.

I enabled uboot-overlays but still can't find anything under:
```
debian@beaglebone:~$ ls -al /dev/spi*
ls: cannot access '/dev/spi*': No such file or directory
```
My config:
```
debian@beaglebone:~$ uname -r; cat /etc/dogtag; sudo /opt/scripts/tools/version.sh
4.19.9-ti-r5
BeagleBoard.org Debian Image 2018-10-07
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTBWA51646BBWG1785]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.11-rc3-00002-g73df7f]:[location: dd MBR]
kernel:[4.19.9-ti-r5]
nodejs:[v6.16.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo ]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190104.0-0rcnee0~stretch+20190104]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20181227.0-0rcnee0~stretch+20181228]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.051022] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.052302] gpio-of-helper ocp:cape-universal: ready
END
```
```
debian@beaglebone:~$ cat /boot/uEnv.txt 
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.19.9-ti-r5
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

##Enabling SPI https://groups.google.com/forum/#!topic/beagleboard/RShjePl0noY
uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo 
```

RobertCNelson, 2019-01-08:
@himanshujha199640 

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.11-rc3-00002-g73df7f]:[location: dd MBR]
```

U-Boot in the eMMC is too old, blocking the newer version from doing it's job Just run:

```code
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

Regards,

RobertCNelson, 2019-01-08:
Ps disable:

```
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
```
Via chaning:
```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
```
to:
```
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
```
4.19 is going to use a different pru subsystem again..

Regards,

himanshujha199640, 2019-01-08:
@RobertCNelson Thanks! It worked :)
```
debian@beaglebone:~$ ls /dev/spi*
/dev/spidev0.0  /dev/spidev0.1  /dev/spidev1.0  /dev/spidev1.1
```
Do I need to set pins to `spi` using `config-pin` utility before connecting the sensor
to use it ?
```
debian@beaglebone:~$ config-pin -i P9.21
Pin name: P9_21
Function if no cape loaded: gpio
Function if cape loaded: default gpio gpio_pu gpio_pd gpio_input spi uart i2c pwm pru_uart
Function information: gpio0_3 default gpio0_3 gpio0_3 gpio0_3 gpio0_3 spi0_d0 uart2_txd i2c2_scl ehrpwm0b pru_uart
Kernel GPIO id: 3
PRU GPIO id: 35
```
I suspect that I'm doing something very stupid since I want to use driver:
https://github.com/beagleboard/linux/blob/4.19/drivers/iio/chemical/bme680_spi.c
and I think simply connecting the sensor while moduled loaded could have sufficed.

OTOH, I tested the `bme680_i2c` driver with the help of `echo bme680 0x76 > /sys/..../new_device`
since it doesn't have DT support and I was just lazy for other options such as using overlays.

himanshujha199640, 2019-01-13:
I'm still waiting for an answer..
Or should ask this on beagleboard-google-group mailing list ?

RobertCNelson, 2019-01-13:
@himanshujha199640 , yes you need to setup the pins thru config-pin or just use an overlay with the pins setup and the bme680 i2c node setup.

Regards,

himanshujha199640, 2019-02-17:
@RobertCNelson 
I tried using `config-pin` but it doesn't work:
```
root@beaglebone:~# config-pin -l P9.28
default gpio gpio_pu gpio_pd gpio_input spi_cs pwm pwm2 pruout pruin
root@beaglebone:~# config-pin -q P9.28
P9_28 pinmux file not found!
Cannot read pinmux file: /sys/devices/platform/ocp/ocp*P9_28_pinmux/state
root@beaglebone:~# config-pin P9.28 spi
P9_28 pinmux file not found!
Invalid mode: spi
root@beaglebone:~# config-pin P9.28 spi_cs
P9_28 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P9_28_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_28_pinmux/state
```
Do I need to add the following ?
`uboot_overlay_addr4=/lib/firmware/cape-universal-00A0.dtbo` in `uEnv.txt`
```
debian@beaglebone:~$ ls /lib/firmware/cape-universal*
/lib/firmware/cape-universal-00A0.dtbo   /lib/firmware/cape-universalh-00A0.dtbo
/lib/firmware/cape-universala-00A0.dtbo  /lib/firmware/cape-universaln-00A0.dtbo
```

All config/env are the same as stated above.

Also, I'm not sure how to write a dt-overlay containing a single compatible property for device matching.

`"4.19 is going to use a different pru subsystem again.."`
Is this posing any problem ?
```
debian@beaglebone:~$ ls /lib/firmware/AM335X-PRU-RPROC-4-*
/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo          /lib/firmware/AM335X-PRU-RPROC-4-4-TI-PRUCAPE-00A0.dtbo
/lib/firmware/AM335X-PRU-RPROC-4-14-TI-PRUCAPE-00A0.dtbo  /lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo
/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo           /lib/firmware/AM335X-PRU-RPROC-4-9-TI-PRUCAPE-00A0.dtbo
```

RobertCNelson, 2019-02-18:
@himanshujha199640, what's your current out put of:

```
sudo /opt/scripts/tools/version.sh
```

Regards,

himanshujha199640, 2019-02-18:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian: 
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTBWA51646BBWG1785]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.19.9-ti-r5]
nodejs:[v6.16.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo ]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190215.0-0rcnee0~stretch+20190215]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.123217] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.136633] gpio-of-helper ocp:cape-universal: ready
END
```

RobertCNelson, 2019-02-18:
@himanshujha199640 , there is you answer..

```
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo ]
```

You can't use config-pin to put the pin in spi mode as it's already in spi mode..

Regards,

closed 2019-03-16 by himanshujha199640