beagleboard/bb.org-overlays #110: Pins P9.14 and P8.13 pi pinmux file not found!

FelipeGo, 2019-02-22:
Hi some how when I activate the SPI these pins dont work ( P9.14 and P8.13 pi pinmux file not found!)
Any ideas?

**Information**:
ls /sys/devices/platform/ocp/ | grep pinmux*
ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_11_pinmux
ocp:P8_12_pinmux
ocp:P8_14_pinmux
ocp:P8_15_pinmux
ocp:P8_16_pinmux
ocp:P8_17_pinmux
ocp:P8_18_pinmux
ocp:P8_26_pinmux
ocp:P9_11_pinmux
ocp:P9_12_pinmux
ocp:P9_13_pinmux
ocp:P9_15_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_23_pinmux
ocp:P9_24_pinmux
ocp:P9_25_pinmux
ocp:P9_26_pinmux
ocp:P9_27_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux
----------------------------------------------------------------------------------------------------------------------------------------
/opt/scripts/tools/version.sh
git:/opt/scripts/:[38ce81fcdafad0892b6c485acd4e9a2245dfb07a]
eeprom:[A335BNLT00C03516BBBK036F]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[5.0.0-rc7-bone2]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-PWM1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr3=/lib/firmware/BB-PWM2-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190215.0-0rcnee0~stretch+20190215]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.554579] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.564216] gpio-of-helper ocp:cape-universal: ready
END
=========================================================================
uEnv.tx
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=5.0.0-rc7-bone2
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BB-PWM1-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/BB-PWM2-00A0.dtbo
disable_uboot_overlay_audio=1
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

RobertCNelson, 2019-02-22:
@FelipeGo , well you setup P9_14 and P8_13 to pwm via:

```
uboot_overlay_addr2=/lib/firmware/BB-PWM1-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/BB-PWM2-00A0.dtbo
```
So do you want them as spi or pwm?

Regards,

FelipeGo, 2019-02-22:
Dear Robert 
Thank you for your reply. I want them as pwm. I have a setup where P9_14 and P8_13 control a motor and a servo motor. And the other spi pins read from a encoder. Initially these pins P9_14 and P8_13 were working perfectly. However, after I added 
```bash
uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo
```
They stopped working. I checked the pinmux and they are not there anymore. It is not clear to me why they are disabled as this pins are not mentioned in  BB-SPIDEV0-00A0.dtbo and BB-SPIDEV1-00A0.dtbo

RobertCNelson, 2019-02-22:
@FelipeGo

```
ls /sys/devices/platform/ocp/ | grep pinmux*
```

List's all the current "available" pin's. BUT YOU forced P9_14 and P8_13 into pwm mode via;

```
uboot_overlay_addr2=/lib/firmware/BB-PWM1-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/BB-PWM2-00A0.dtbo
```

Thus, they are "un-available" and no longer listed under "/sys/devices/platform/ocp/" for change.

Regards,

FelipeGo, 2019-02-22:
Ok I understand. My main problem is when I run my code to control the servo and the motor
```python
import Adafruit_BBIO.PWM as PWM
motor_pin  = 'P8_13'
PWM.start(motor_pin,0.0,100000) # This line fails (cannot access this pin)
```
or 
```python
import Adafruit_BBIO.PWM as PWM
servo_pin  = 'P9_14'
PWM.start(servo_pin, ...,...  # This line fails (cannot access this pin)
```
As stated in my previous post  this code only fails when I add `uboot_overlay_addr0=/lib/firmware/BB-SPIDEV[0;1]-00A0.dtbo` in _uEnv.txt_. However, when I remove it and reboot the BBB the code runs correctly. Consequently, I decided to add the lines `uboot_overlay_addr2=/lib/firmware/BB-PWM[1;2]-00A0.dtbo` expecting that this will force the pwm pins to work but it did not.
 
Im sorry to ask you so much, do you have any ideas why my code fails when I force the other pins to be SPI via  _uEnv.txt_? I really need to work with the spi to read the encoder data while controlling the motor.

Thank you in advance!!

RobertCNelson, 2019-02-22:
@FelipeGo well your bug is elsewhere, adding BB-SPIDEV0-00A0.dtbo/BB-SPIDEV0-00A1.dtbo doesn't break P8_13 or P9_14, as seen below:

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[38ce81fcdafad0892b6c485acd4e9a2245dfb07a]
eeprom:[A335BNLT000C1830BBBG0785]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[5.0.0-rc7-bone2]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190215.0-0rcnee0~stretch+20190215]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.553297] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.563033] gpio-of-helper ocp:cape-universal: ready
END
```

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh | grep uboot
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
```

```
debian@beaglebone:~$ config-pin -q P8_13
P8_13 Mode: default Direction: in Value: 0
debian@beaglebone:~$ config-pin -q P9_14
P9_14 Mode: default Direction: in Value: 0
```
```
+ uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo
```

```
debian@beaglebone:~$ config-pin -q P8_13
P8_13 Mode: default Direction: in Value: 0
debian@beaglebone:~$ config-pin -q P9_14
P9_14 Mode: default Direction: in Value: 0
```

```
+ uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo
```

```
debian@beaglebone:~$ config-pin -q P8_13
P8_13 Mode: default Direction: in Value: 0
debian@beaglebone:~$ config-pin -q P9_14
P9_14 Mode: default Direction: in Value: 0
```

```
debian@beaglebone:~$ config-pin P8_13 pwm
debian@beaglebone:~$ config-pin P9_14 pwm
```

Regards,

FelipeGo, 2019-02-22:
ok I will check it out,

Thank you for your help

closed 2019-02-22 by FelipeGo