beagleboard/bb.org-overlays #112: Status on the BeagleBone BLUE regarding Cape Universal & PRU-RPROC

patrickpoirier51, 2019-03-03:
@RobertCNelson @jadonk 
Try to get the Blue working with ArduPilot with latest 4.19 and I experience problems.
Still not traces of the blue in the latest RPROC
https://github.com/beagleboard/bb.org-overlays/blob/3eae6276f9e0ea5f9aec3be30ac1424f8abc6a7c/src/arm/AM335X-PRU-RPROC-4-19-TI-00A0.dts#L12

Still not traces of the blue in cape universal 
https://github.com/beagleboard/bb.org-overlays/blob/3eae6276f9e0ea5f9aec3be30ac1424f8abc6a7c/src/arm/cape-universal-00A0.dts#L27

As for the dtb=am335x-boneblue.dtb, is it still compatible with the new kernel ?

patrickpoirier51, 2019-03-03:
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21712EL004215]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gbc653b9d91]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.11-00002-g7b415acfc1]:[location: dd MBR]
kernel:[4.19.26-bone-rt-r26]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.712809] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper

patrickpoirier51, 2019-03-03:
`debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21712EL004215]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gbc653b9d91]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.11-00002-g7b415acfc1]:[location: dd MBR]
kernel:[4.19.26-bone-rt-r26]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.712809] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
`

patrickpoirier51, 2019-03-03:
> debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21712EL004215]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gbc653b9d91]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.11-00002-g7b415acfc1]:[location: dd MBR]
kernel:[4.19.26-bone-rt-r26]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.712809] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper

patrickpoirier51, 2019-03-03:
Here is the version

git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21712EL004215]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gbc653b9d91]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.11-00002-g7b415acfc1]:[location: dd MBR]
kernel:[4.19.26-bone-rt-r26]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.712809] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper

patrickpoirier51, 2019-03-03:
Juste "nuked" the eMMc bootloader to make sure it was not interfering
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10

patrickpoirier51, 2019-03-03:
git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21712EL004215]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gbc653b9                                         d91]:[location: dd MBR]
kernel:[4.19.26-bone-rt-r26]
device-tree-override:[dtb=am335x-boneblue.dtb]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-                                         00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                                         ers systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenom                                         ai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/m                                         mcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.714635] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21712EL004215]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gbc653b9d91]:[location: dd MBR]
kernel:[4.19.26-bone-rt-r26]
device-tree-override:[dtb=am335x-boneblue.dtb]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.714635] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper

patrickpoirier51, 2019-03-03:
git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21712EL004215]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gbc653b9d91]:[location: dd MBR]
kernel:[4.19.26-bone-rt-r26]
device-tree-override:[dtb=am335x-boneblue.dtb]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.714635] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper

RobertCNelson, 2019-03-03:
@patrickpoirier51 , wrong kernel for remoteproc_pruss

```
sudo /opt/scripts/tools/update_kernel.sh --ti-rt-channel --lts-4_19
```

Wrong overlay:
```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
```
Use:
```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
```

patrickpoirier51, 2019-03-03:
Updated to the correct version and  dtbo 

git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21712EL004215]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gbc653b9d91]:[location: dd MBR]
kernel:[4.19.25-ti-rt-r12]
device-tree-override:[dtb=am335x-boneblue.dtb]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.234661] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper

patrickpoirier51, 2019-03-03:
@RobertCNelson  but still NO pru loaded in kernel as lsmod | grep pru
 shows nothing. 
This is expected as the BeagleBone Blue is not in the list
bb.org-overlays/src/arm/AM335X-PRU-RPROC-4-19-TI-00A0.dts

Line 12 in 3eae627

 compatible = "ti,beaglebone", "ti,beaglebone-black", "ti,beaglebone-green";

RobertCNelson, 2019-03-03:
@patrickpoirier51 it's actually built in, mostly to test to see if we can go direction, it helps speed up boot times with librobotcontrol installed...

https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/bebf3e6379f29d564cbb71cdc37e847dbbef723e#diff-ba79210cd3316674807a5388d265a3f2

Regards,

RobertCNelson, 2019-03-03:
I don't remember how much i tested it last friday..

```
dmesg | remote
```
Regards,

RobertCNelson, 2019-03-04:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBLA21722EL002477]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-02-24]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g08ae12f051]:[location: dd MBR]
kernel:[4.19.25-ti-r12]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.943611] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
debian@beaglebone:~$ dmesg | grep remote
[    1.226700] remoteproc remoteproc0: wkup_m3 is available
[    1.228451] remoteproc remoteproc1: 4a334000.pru is available
[    1.229048] remoteproc remoteproc2: 4a338000.pru is available
[    1.365740] remoteproc remoteproc0: powering up wkup_m3
[    1.365872] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.369359] remoteproc remoteproc0: remote processor wkup_m3 is now up
```

patrickpoirier51, 2019-03-04:
Ok I will reflash and test again
Thanks for the follow-up

patrickpoirier51, 2019-03-09:
@RobertCNelson   try to test this morning but cannot get file
info: [linux-image-4.19.25-ti-rt-r14] (latest) is currently unavailable on [rcn-ee.com/repos]

RobertCNelson, 2019-03-09:
@patrickpoirier51 try again, it might have been in the middle of the upload, i only have a 20Mb/s upload..

Regards,

patrickpoirier51, 2019-03-09:
There is  an issue with the server .. checking for archives takes forever and cannot get the file.
Anyway I could update to 4.19 on my green and cannot get the PRU loaded in kernel

eeprom:[A335BNLTBBG1BBG216084326]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Image 2019-03-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g08ae12f                                      051]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.10-00001-g143c9ee]:[loca                                      tion: dd MBR]
kernel:[4.19.25-ti-r14]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-                                      00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                                      ers systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenom                                      ai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait                                       coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.933204] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.934955] gpio-of-helper ocp:cape-universal: ready
END

dmesg | grep remote
[    5.176301] remoteproc remoteproc0: wkup_m3 is available
[    5.315171] remoteproc remoteproc0: powering up wkup_m3
[    5.315301] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    5.318721] remoteproc remoteproc0: remote processor wkup_m3 is now up

debian@beaglebone:~$ dmesg | grep pru
debian@beaglebone:~$ lsmod | grep pru
debian@beaglebone:~$

patrickpoirier51, 2019-03-09:
@RobertCNelson   I checked in apt-cache and r14 was not there.
I loaded  4.19.25-ti-rt-r13  but still cannot get the PRU started

RobertCNelson, 2019-03-10:
```
dogtag:[BeagleBoard.org Debian Image 2019-03-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g08ae12f 051]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.10-00001-g143c9ee]:[loca tion: dd MBR]
kernel:[4.19.25-ti-r14]
```
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```
ah i see what happened, i had an ssd lockup on my x2 builder:

[image]

Regards,

patrickpoirier51, 2019-03-10:
Yes I had deleted the eMMc partition
Does it takes a while to get the new released published , as it is not yet in cache

RobertCNelson, 2019-03-10:
```
dpkg-buildpackage: info: binary-only upload (no source included)

real	106m20.590s
user	181m7.137s
sys	26m39.482s
```

http://gfnd.rcn-ee.org:81/farm/deb/X2-BUILDING-4.19.25-ti-rt-r14-stretch-armhf.txt

(log updated every 5 minutes)

Regards,

patrickpoirier51, 2019-03-10:
OK
Could make BBBlue work with ArduPilot on 4.19  finally !! ;-)

Thanks Robert

closed 2019-03-10 by patrickpoirier51