beagleboard/bb.org-overlays #113: BeagleBone Black Boot hangs: "Trying to boot from MMC1"

lukasz139, 2019-03-21:
Hi,

I was trying to get my device tree loaded such that I can use PRU capabilities. One of  the solutions that worked for me was given in this issue:

https://github.com/beagleboard/bb.org-overlays/issues/76#issue-309484275

It worked for me, however, now during the boot I see following message on serial terminal:

U-Boot SPL 2018.01-00002-g9aa111a004 (Jan 20 2018 - 12:45:29)
Trying to boot from MMC1

It tries to boot from eMMC and nothing happens. I have tried with and without BOOT button. 

Linux: kernel:[4.9.78-ti-r94]

Thank you in advance

RobertCNelson, 2019-03-21:
@lukasz139 

```
U-Boot SPL 2018.01-00002-g9aa111a004 (Jan 20 2018 - 12:45:29)
Trying to boot from MMC1
```

I kinda remember this bug.. Does your BBB say "Embest or Element 14" on it?  This got fixed last fall, please verify with this image: 

https://rcn-ee.net/rootfs/bb.org/testing/2019-03-17/stretch-iot/bone-debian-9.8-iot-armhf-2019-03-17-4gb.img.xz

Regards,

edgex004, 2023-03-10:
@RobertCNelson Sorry to grave-dig, but I am encountering this problem on a BB AI-64. I updated U-Boot using these steps: https://forum.beagleboard.org/t/enable-uart-1-on-bb-ai-64/32997/3

Now I am stuck at boot:
```
init_env from device 17 not supported!
Starting ATF on ARM64 core...

NOTICE:  BL31: v2.5(release):e0d9fdb
NOTICE:  BL31: Built : 20:08:51, Jan  7 2022

U-Boot SPL 2021.01-g3419da0b (Jan 13 2022 - 15:29:57 +0000)
Model: Texas Instruments K3 J721E SoC
Board: BBONEAI-64-B0- rev B0
SYSFW ABI: 3.1 (firmware rev 0x0015 '21.5.0--v2021.05 (Terrific Llam')
ti_i2c_eeprom_am6_parse_record: Ignoring record id 17
Trying to boot from MMC1
```

edgex004, 2023-03-10:
Ah, the solution for me is perhaps here:
https://forum.beagleboard.org/t/ai-64-update-bricked-device/32329

It looks like console is disabled on my U-Boot, so I can't break into it via UART. I'll need to try to boot using the reset button.

RobertCNelson, 2023-03-11:
@edgex004 use a flasher, early on (early 2022) we had some early u-boot builds that weren't ready..

https://forum.beagleboard.org/t/debian-11-x-bullseye-monthly-snapshots-arm64/32318