beagleboard/bb.org-overlays #114: BBBMINI /dev/spidev  new -vs- old mapping

patrickpoirier51, 2019-03-26:
Hello Robert,

On the BBBMINI on pre 4.9 releases we had this mapping
ls /dev/spi*
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1

Now I have upgraded to 4.19 and I have this mapping
 ls /dev/spi*
/dev/spidev1.0  /dev/spidev1.1  /dev/spidev2.0  /dev/spidev2.1

Problem : We builded ArduPilot to use SPI 2
Note: I have not implemented config-pin and I tried with or without loading 
dtb=am335x-boneblack-bbbmini.dtb
and with SPI DTBO == no change 

----version----
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBBG1BBG216084326]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Image 2019-03-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g08ae12f051]:[location: dd MBR]
kernel:[4.19.29-bone-rt-r29]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190320.1-0rcnee0~stretch+20190320]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.814231] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.826045] gpio-of-helper ocp:cape-universal: ready
END

RobertCNelson, 2019-03-26:
@patrickpoirier51 , i'll take a look at this issue this week..  @jadonk  and I was hoping you guys would have stopped hardcoding the /dev/spidev* node and used the sys lookup..

Regards,

patrickpoirier51, 2019-03-26:
Ok, So I will check on our side with @mirkix

Thanks

patrickpoirier51, 2019-03-26:
Would you have any pointers for the programming  methodology to acces these devices ?

patrickpoirier51, 2019-03-27:
Guess what ? 
The SPI mapping I can manage it with 
dtb_overlay=/lib/firmware/BB-SPIDEV1-00A0.dtbo

I can set the 
sudo echo spi_cs >/sys/devices/platform/ocp/ocp:P9_42_pinmux/state
config-pin -q P9.42
P9_42 Mode: spi_cs

but cannot t set P9.28 
I see it is mapped to audio  even disabled:

git:/opt/scripts/:[36a476c67f707ba2698bfb30452ebd2951162ce2]
eeprom:[A335BNLTBBG1BBG216084326]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Image 2019-03-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g08ae12f051]:[location: dd MBR]
kernel:[4.19.29-bone-rt-r29]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-SPIDEV1-00A0.dtbo ]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190320.1-0rcnee0~stretch+20190320]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.814046] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.825380] gpio-of-helper ocp:cape-universal: ready
END

RobertCNelson, 2019-03-27:
@patrickpoirier51 

When using: dtb=am335x-boneblack-bbbmini.dtb, i see:

debian@beaglebone:~$ uname -r ; ls /dev/spidev*
```
4.4.155-ti-r153
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1

4.9.147-ti-r119
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1

4.14.103-ti-r100
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1

4.19.25-ti-r16
/dev/spidev0.0  /dev/spidev1.0  /dev/spidev1.1

4.4.145-bone23
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1

4.9.146-bone13
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1

4.14.106-bone19
/dev/spidev0.0  /dev/spidev1.0  /dev/spidev1.1

4.19.29-bone29
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1

5.0.4-bone6
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1

5.1.0-rc2-bone0
/dev/spidev1.0  /dev/spidev2.0  /dev/spidev2.1
```
Thus it looks like i need to fix the 4.19.x-ti and 4.14.x-bone branches..

So are you not using "dtb=am335x-boneblack-bbbmini.dtb" anymore or what?

patrickpoirier51, 2019-03-27:
@RobertCNelson  
We are willing to use  a config-pin configuration file as we did with the PocketPilot, 
providing cape universal offers  the needed GPIO (and particulary the mapping of P9-28 as spi-cs0)

@mirkix

RobertCNelson, 2019-03-27:
We should just create a custom overlay, so that all the base boards (bbb, bbgw, etc) and there we can patch the spi alias name..

patrickpoirier51, 2019-03-27:
That makes a lot of sense, what do you think @mirkix ?

mirkix, 2019-03-28:
@RobertCNelson yes, may a custom overlay would be a good solution. and it is possible to map spi names there?

RobertCNelson, 2019-04-04:
@mirkix and @patrickpoirier51 we have another option rolling thru the kernel, a little udev rule from: https://github.com/mvduin/py-uio/blob/master/etc/udev/rules.d/10-of-symlink.rules

```
debian@beaglebone:~$ uname -r ; ls -lha /dev/spi*
4.4.155-ti-r154
crw-rw---- 1 root spi  153, 1 Apr  4 02:57 /dev/spidev1.0
crw-rw---- 1 root spi  153, 0 Apr  4 02:57 /dev/spidev1.1
crw-rw---- 1 root spi  153, 3 Apr  4 02:57 /dev/spidev2.0
crw-rw---- 1 root spi  153, 2 Apr  4 02:57 /dev/spidev2.1

/dev/spi:
total 0
drwxr-xr-x  2 root root  120 May 21  2016 .
drwxr-xr-x 16 root root 3.2K Apr  4 15:41 ..
lrwxrwxrwx  1 root root   12 Apr  4 02:57 0.0 -> ../spidev1.0
lrwxrwxrwx  1 root root   12 Apr  4 02:57 0.1 -> ../spidev1.1
lrwxrwxrwx  1 root root   12 Apr  4 02:57 1.0 -> ../spidev2.0
lrwxrwxrwx  1 root root   12 Apr  4 02:57 1.1 -> ../spidev2.1
```

Thus instead of hard-coding "/dev/spidevX.Y" just use "/dev/spi/X.Y"

After yesterday these kernel currently support that:

```
4.4.155-ti-r154
4.4.155-ti-rt-r154
4.9.147-ti-r120
4.9.147-ti-rt-r120
4.14.108-ti-r103
4.14.108-ti-rt-r103
4.14.108-ti-xenomai-r103
```

Will be tackling 4.19.x and all bone kernels today...  Should have all "latest" kernel supporting this by the weekend..

Regards,