beagleboard/bb.org-overlays #120: BB-W1-P9.12-00A0.dtbo, uboot overlay did not work correct

ASbetebe, 2019-05-20:
Hi,
i have a problem with my DS18b20 temperature Sensor.

Wired:
Black = GND
Red = 3,3V with resistor 4,7k to Data
Yellow = Data = Pin9.12

My System
Beaglebone Black
Static hostname: beaglebone
Icon name: computer
Machine ID: 206cba11664750b7d62ec81d5cd92719
Boot ID: 26e3097bb8794bdc99f4fefb772370aa
Operating System: Debian GNU/Linux 9 (stretch)
Kernel: Linux 4.14.108-ti-r104
Architecture: arm
BeagleBoard.org Debian Image 2018-10-07

I have made the following changes in the uEnv.txt
uboot_overlay_addr4=/lib/firmware/BB-W1-P9.12-00A0.dtbo

disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1_

If I list all the devices now.
/sys/bus/w1/devices$ ls
Then always different bus addresses are listed there but none with a correct address.
Example
00-140000000000
00-540000000000
00-940000000000
w1_bus_master1 --> This is always listed identically

What am I doing wrong? 
Thanks in advance

ASbetebe

RobertCNelson, 2019-05-20:
@ASbetebe i've had reports that 4.19.x fixes the issue:

```
cd /opt/scripts/
git pull
sudo /opt/scripts/tools/update_kernel.sh --ti-channel --lts-4_19
sudo reboot
```

Regards,

ASbetebe, 2019-05-21:
Good morning,

thank you Robert.
I did the update but unfortunately I still have the same problem.
Static hostname: beaglebone
Icon name: computer
Machine ID: 206cba11664750b7d62ec81d5cd92719
Boot ID: 3b10448f4fc347a6bb2b5c11fd247bd7
Operating System: Debian GNU/Linux 9 (stretch)
Kernel: Linux 4.19.31-ti-r18
Architecture: arm

Are there any othe solutions ?

WalterCEden, 2019-05-28:
I'm having problems too and also using 4.14.71.  My BBB stopped connecting to the Internet via USB and I have not been able to get it to connect again no matter what.  I've used different cables, different computers, etc.  No luck.  So no luck getting 4.19 yet.

WalterCEden, 2019-05-29:
I performed the update indicated above.  It gave the following two warnings.  I'm not sure if they are serious. 

cryptsetup: WARNING: failed to detect canonical device of /dev/mmcblk0p1
cryptsetup: WARNING: could not determine root device from /etc/fstab

WalterCEden, 2019-05-29:
I was able to get my BBB with a hardwired Ethernet cable (still can't get there via USB) and updated to 4.19 using the information above.   I have the same sensor configuration as mentioned in the initial post.   I checked in /sys/bus/w1/devices and don't see anything other than the w1_bus_master1 directory.   Is there something else I can try?   

I ran dmsg and found this at the very end of the listing.

gpio-60 (P9_12): enforced open drain please flag it properly in DT/ACPI DSDT/board file

I do not have a microSD card inserted

uname --all gives this output

Linux beaglebone 4.19.31-ti-r19 #1stretch SMP PREEMPT Sun May 26 04:18:13 UTC 2019 armv7l GNU/Linux

Here is my /boot/uEnv.txt 

#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.19.31-ti-r19
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.9.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.9.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this nextline:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

What should I try next?

WalterCEden, 2019-05-30:
Latest update!  I see the sensor in the directory tree!  Apparently, I had a bad connection that I fixed when I rewired the board today.  Woohoo!  It's always a hardware problem right?   Now I can get on to writing my C code to read the temp!