beagleboard/bb.org-overlays #125: Can boot with custom overlay

bluePain, 2019-06-15:
Hi there,

I'm kinda new around here and trying to build my own beagle from scratch to learn some basics. Currently, I'm dealing with device trees and trying to enable all uarts with a custom cape. I've successfully enabled all but except UART-5, beagle just stucks in u-boot whenever I try to enable this port.

Here is the moment that it hangs in u-boot:

> U-Boot 2019.04 (May 31 2019 - 17:17:12 +0300)
> 
> CPU  : AM335X-GP rev 2.1
> I2C:   ready
> DRAM:  512 MiB
> No match for driver 'omap_hsmmc'
> No match for driver 'omap_hsmmc'
> Some drivers were not found
> Reset Source: Global external warm reset has occurred.
> Reset Source: Global warm SW reset has occurred.
> Reset Source: Power-on reset has occurred.
> RTC 32KCLK Source: External.
> MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
> Loading Environment from EXT4... Card did not respond to voltage select!
> Board: BeagleBone Black
> <ethaddr> not set. Validating first E-fuse MAC
> BeagleBone Black:
> Model: SeeedStudio BeagleBone Green:
> BeagleBone: cape eeprom: i2c_probe: 0x54:
> BeagleBone: cape eeprom: i2c_probe: 0x55:
> BeagleBone: cape eeprom: i2c_probe: 0x56:
> BeagleBone: cape eeprom: i2c_probe: 0x57:
> Net:   eth0: MII MODE
> cpsw, usb_ether
> Press SPACE to abort autoboot in 2 seconds
> board_name=[BBG1] ...
> Card did not respond to voltage select!
> Card did not respond to voltage select!
> Card did not respond to voltage select!
> gpio: pin 56 (gpio 56) value is 0
> gpio: pin 55 (gpio 55) value is 0
> gpio: pin 54 (gpio 54) value is 0
> gpio: pin 53 (gpio 53) value is 1
> Card did not respond to voltage select!
> Card did not respond to voltage select!
> switch to partitions #0, OK
> mmc1(part 0) is current device
> Scanning mmc 1:1...
> gpio: pin 56 (gpio 56) value is 0
> gpio: pin 55 (gpio 55) value is 0
> gpio: pin 54 (gpio 54) value is 0
> gpio: pin 53 (gpio 53) value is 1
> switch to partitions #0, OK
> mmc1(part 0) is current device
> gpio: pin 54 (gpio 54) value is 1
> Checking for: /uEnv.txt ...
> Checking for: /boot.scr ...
> Checking for: /boot/boot.scr ...
> Checking for: /boot/uEnv.txt ...
> gpio: pin 55 (gpio 55) value is 1
> 587 bytes read in 12 ms (46.9 KiB/s)
> Loaded environment from /boot/uEnv.txt
> Checking if uname_r is set in /boot/uEnv.txt...
> gpio: pin 56 (gpio 56) value is 1
> Running uname_boot ...
> loading /boot/vmlinuz-4.19.31 ...
> 10375680 bytes read in 663 ms (14.9 MiB/s)
> debug: [enable_uboot_overlays=1] ...
> debug: [enable_uboot_cape_universal=1] ...
> debug: [uboot_base_dtb_univ=am335x-boneblack-uboot-univ.dtb] ...
> uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot-univ.dtb] ...
> debug: unable to find [am335x-boneblack-uboot-univ.dtb] using [am335x-boneblack-uboot.dtb] instead ...
> debug: [uboot_base_dtb_univ=am335x-boneblack-uboot.dtb] ...
> uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot.dtb] ...
> loading /boot/dtbs/4.19.31/am335x-bonegreen.dtb ...
> 59403 bytes read in 20 ms (2.8 MiB/s)
> uboot_overlays: [fdt_buffer=0x60000] ...
> uboot_overlays: loading /lib/firmware/M-BB-BBG-00A0.dtbo ...
> 277 bytes read in 445 ms (0 Bytes/s)
> uboot_overlays: uboot loading of [/lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo] disabled by /boot/uEnv.txt [dis
> able_uboot_overlay_emmc=1]...
> uboot_overlays: uboot loading of [/lib/firmware/BB-ADC-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot
> _overlay_adc=1]...
> uboot_overlays: [dtb_overlay=/lib/firmware/issd-test-00A0.dtbo] ...
> uboot_overlays: loading /lib/firmware/issd-test-00A0.dtbo ...
> 2291 bytes read in 238 ms (8.8 KiB/s)
> failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
> uboot_overlays: cape universal disabled, external cape enabled or detected...
> loading /boot/initrd.img-4.19.31 ...
> 6147279 bytes read in 399 ms (14.7 MiB/s)
> debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 
> rootwait coherent_pool=1M] ...
> debug: [bootz 0x82000000 0x88080000:5dcccf 88000000] ...
> ERROR: Did not find a cmdline Flattened Device Tree
> Could not find a valid device tree
> ** Invalid partition 2 **
> ** Invalid partition 3 **
> ** Invalid partition 4 **
> ** Invalid partition 5 **
> ** Invalid partition 6 **
> ** Invalid partition 7 **
> starting USB...
> USB0:   Port not available.
> link up on port 0, speed 100, full duplex
> BOOTP broadcast 1
> DHCP client bound to address 192.168.0.232 (371 ms)
> Using cpsw device
> TFTP from server 192.168.1.1; our IP address is 192.168.0.232; sending through gateway 192.168.0.1
> Filename 'vmlinuz-4.19.31'.
> Load address: 0x82000000
> Loading: T T T T T T 

Here is the uEnv.txt:

> enable_uboot_overlays=1
> cmdline=coherent_pool=1M
> #dtb_overlay=/lib/firmware/BB-UART5-00A0.dtbo
> dtb_overlay=/lib/firmware/uart-test-00A0.dtbo
> enable_uboot_cape_universal=1
> disable_uboot_overlay_emmc=1
> disable_uboot_overlay_video=1
> disable_uboot_overlay_audio=1
> disable_uboot_overlay_adc=1
> disable_uboot_overlay_addr0=1
> disable_uboot_overlay_addr1=1
> disable_uboot_overlay_addr2=1
> disable_uboot_overlay_addr3=1

And lastly here's the overlay file:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black", "ti,beaglebone-green";

	/* identification */
	part-number = "uart-test";
	version = "00A0"; 

	/* state the resources this cape uses */
	exclusive-use =
		/* the pin header uses */
		"P8.37",	/* uart5_txd */
        "P8.38",    /* uart5_rxd */
        
        /* the hardware ip uses */
        "uart5";

	fragment@0 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			bb_uart5_pins: pinmux_bb_uart5_pins {
				pinctrl-single,pins = <
					/* the uart pins */
					0x0C4 0x24	/* lcd_data9.uart5_rxd | MODE4 */
					0x0C0 0x04	/* lcd_data8.uart5_txd | MODE4 */
				>;
			};
		};
    };
    
	fragment@1 {
		target = <&uart6>;	/* really uart5 */
		__overlay__ {
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_uart5_pins>;
		};
    };
};
```

Btw, Beagle works perfect with the BB-UART5 overlay from the filesystem and registers the /dev/ttyO6 successfully.

Any suggestion or correction?

Thanks!

RobertCNelson, 2019-06-15:
```
uboot_overlays: [dtb_overlay=/lib/firmware/issd-test-00A0.dtbo] ...
uboot_overlays: loading /lib/firmware/issd-test-00A0.dtbo ...
2291 bytes read in 238 ms (8.8 KiB/s)
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
```

```
	fragment@1 {
		target = <&uart6>;	/* really uart5 */
		__overlay__ {
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_uart5_pins>;
		};
    };
```

Wrong target 'label'.

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-UART5-00A0.dts

Regards,

closed 2019-06-15 by RobertCNelson

bluePain, 2019-06-17:
I started to hate myself.

I'm deeply sorry for stealing a couple of minutes from your life.

And thanks a lot.

RobertCNelson, 2019-06-17:
@bluePain no worries, so back in the 3.8.x days, your overlay would have worked, as the usarts where mis-numbered.. but this got fixed somewhere around 3.10.x, or so..