beagleboard/bb.org-overlays #126: Overlays in U-boot for beaglebone

GauravChoube, 2019-06-27:
Hi,
I have build uboot and kernel from https://www.digikey.com/eewiki/display/linuxonarm/BeagleBone+Black
 and create SD card as per instrucution.This one having minimal image.
i want to add overlays in order to do change status of some pin(High to low).
I dont want to build kernel again.
When searched the inertnet all are old one slote base overlay.
but in my image there in bb_cap_manager in /etc/device/ and not slot file.
After read some thread,got to know and overlays is shifted uboot and there is not kernel overlays.
Can anyone explain me how to create overlays first all and how to add in uboot.
Any help will be appreciated.

tehkillerbee, 2019-06-28:
A really good starting point is https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
Take a look at "U-Boot Override external capes"

Creating your own overlays requires writing your own .dts files and compiling them to .dtbo. Use existing sources as a starting point and modify them according to your needs:
https://github.com/beagleboard/bb.org-overlays/tree/master/src/arm

1. Clone this git repository (https://github.com/beagleboard/bb.org-overlays) and install it (look at the readme.md  for instructions). This includes the toolchain for compiling the dts to dtbo.

2. Create your own dts source file/modify existing sources and save it in your local clone of the repository (eg. save it in ~/bb.org-overlays/src/arm/BB-XYZ-00A0.dts) 

3. Compile it:
cd ~/bb.org-overlays
make
sudo cp src/arm/BB-XYZ-00A0.dtbo /lib/firmware/BB-XYZ-00A0.dtbo

jadonk, 2019-07-08:
@GauravChoube can this issue be closed?