beagleboard/bb.org-overlays #136: porblems with seeed-4mic-voicecard

tulptuiz, 2019-09-06:
Hi, I want to run 4-mic-array (http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/) on beaglebone green and I've some problems. These are my commands: 

```
sudo apt install bb-cape-overlays
sudo /opt/scripts/tools/update_kernel.sh --lts-4_14 --ti-rt-channel
git clone https://github.com/beagleboard/bb.org-overlays
cd ./bb.org-overlays
./install.sh
```

and finally change in uEnv.txt address to the file:

```uboot_overlay_addr0=/lib/firmware/seeed-4mic-voicecard-00A0.dtbo```

Next I rebooted beaglebone and it freezed permamently (all diodes were constantly lightning) and I had no connection via usb or ssh. I must to reinstall debian (user and reset button didn't working)
Maybe I should change something more in uEnv.txt, but I don't know exactly what.  
Could anyone find what could causes that problems?

RobertCNelson, 2019-09-06:
@tulptuiz can you grab serial debug (j1 connector)..

Regards,

tulptuiz, 2019-09-09:
**I've got something like this:**

U-Boot SPL 2018.09-00002-g0b54a51eee (Sep 10 2018 - 19:41:39 -0500) 
Trying to boot from MMC2 
Loading Environment from EXT4... Card did not respond to voltage select! 
 
 
U-Boot 2018.09-00002-g0b54a51eee (Sep 10 2018 - 19:41:39 -0500), Build: jenkins-github_Bootloader-Builder-65 
 
CPU  : AM335X-GP rev 2.1 
I2C:   ready 
DRAM:  512 MiB 
No match for driver 'omap_hsmmc' 
No match for driver 'omap_hsmmc' 
Some drivers were not found 
Reset Source: Power-on reset has occurred. 
RTC 32KCLK Source: External. 
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1 
Loading Environment from EXT4... Card did not respond to voltage select! 
Board: BeagleBone Black 
<ethaddr> not set. Validating first E-fuse MAC 
BeagleBone Black: 
Model: SeeedStudio BeagleBone Green Wireless: 
BeagleBone: cape eeprom: i2c_probe: 0x54: 
BeagleBone: cape eeprom: i2c_probe: 0x55: 
BeagleBone: cape eeprom: i2c_probe: 0x56: 
BeagleBone: cape eeprom: i2c_probe: 0x57: 
Net:   eth0: MII MODE 
Could not get PHY for cpsw: addr 0 
cpsw, usb_ether 
Press SPACE to abort autoboot in 2 seconds 
board_name=[A335BNLT] ... 
board_rev=[GW1A] ... 
Card did not respond to voltage select! 
Card did not respond to voltage select! 
Card did not respond to voltage select! 
gpio: pin 56 (gpio 56) value is 0 
gpio: pin 55 (gpio 55) value is 0 
gpio: pin 54 (gpio 54) value is 0 
gpio: pin 53 (gpio 53) value is 1 
Card did not respond to voltage select! 
Card did not respond to voltage select! 
switch to partitions #0, OK 
mmc1(part 0) is current device 
Scanning mmc 1:1... 
gpio: pin 56 (gpio 56) value is 0 
gpio: pin 55 (gpio 55) value is 0 
gpio: pin 54 (gpio 54) value is 0 
gpio: pin 53 (gpio 53) value is 1 
switch to partitions #0, OK 
mmc1(part 0) is current device 
gpio: pin 54 (gpio 54) value is 1 
Checking for: /uEnv.txt ... 
Checking for: /boot.scr ... 
Checking for: /boot/boot.scr ... 
Checking for: /boot/uEnv.txt ... 
gpio: pin 55 (gpio 55) value is 1 
2059 bytes read in 7 ms (287.1 KiB/s) 
Loaded environment from /boot/uEnv.txt 
Checking if uname_r is set in /boot/uEnv.txt... 
gpio: pin 56 (gpio 56) value is 1 
Running uname_boot ... 
loading /boot/vmlinuz-4.14.71-ti-r80 ... 
10416640 bytes read in 664 ms (15 MiB/s) 
debug: [enable_uboot_overlays=1] ... 
debug: [enable_uboot_cape_universal=1] ... 
debug: [uboot_base_dtb_univ=am335x-bonegreen-wireless-uboot-univ.dtb] ... 
uboot_overlays: [uboot_base_dtb=am335x-bonegreen-wireless-uboot-univ.dtb] ... 
uboot_overlays: Switching too: dtb=am335x-bonegreen-wireless-uboot-univ.dtb ... 
loading /boot/dtbs/4.14.71-ti-r80/am335x-bonegreen-wireless-uboot-univ.dtb ... 
[... 110 lines cut ...]
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS 
The remote end did not respond in time.Retrieving file: pxelinux.cfg/default-arm 
using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in 
MAC c4:f3:12:81:b4:29 
HOST MAC de:ad:be:af:00:00 
RNDIS ready 
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS 
The remote end did not respond in time.Retrieving file: pxelinux.cfg/default 
using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in 
MAC c4:f3:12:81:b4:29 
HOST MAC de:ad:be:af:00:00 
RNDIS ready 
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS 
The remote end did not respond in time.Config file not found 
using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in 
MAC c4:f3:12:81:b4:29 
HOST MAC de:ad:be:af:00:00 
RNDIS ready
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS 
The remote end did not respond in time.using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in 
MAC c4:f3:12:81:b4:29 
HOST MAC de:ad:be:af:00:00 
RNDIS ready 
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS 
The remote end did not respond in time.=>