beagleboard/bb.org-overlays #137: eMMC not available on BBB

eblot, 2019-09-10:
Hi,

I'm updating our old BBB / buildroot environment (three years old) to the latest Buildroot (2019.08) & Kernel (4.19.59). Everything is ok, except the access to the eMMC: when the BBB boots up with the new kernel with the new uboot - both from the SD, only the SD is available from Linux *i.e.* only `/dev/mmcblk0` is listed as a device, `/dev/mmcblk1` never shows up.

uboot finds both devices, but for some reason Linux kernel does not. dmesg only reports a single device.

I guess that is somewhat related to the DTB & overlays, but I failed to find the info to sort this out. I specified `am335x-boneblack-uboot-univ.dtb` as the device tree for uboot, nothing else.

I should say I'm lost with what enable_uboot_overlays and other uboot-related overlays do and most importantly how they do it.

How to enable both SD & eMMC at boot up?
Thanks.

RobertCNelson, 2019-09-10:
You are Correct, am335x-boneblack-uboot-univ.dtb does not enable the eMMC 

https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x-ti/src/arm/am335x-boneblack-uboot-univ.dts

Our overlay manager is not in mainline yet..  It's also tied to u-boot v2019.04

https://github.com/RobertCNelson/Bootloader-Builder/blob/master/patches/v2019.04/0001-am335x_evm-uEnv.txt-bootz-n-fixes.patch

https://github.com/RobertCNelson/Bootloader-Builder/blob/master/patches/v2019.04/0002-U-Boot-BeagleBone-Cape-Manager.patch

You can also use dtc's tool to apply the eMMC overlay on top of the am335x-boneblack-uboot-univ.dtb in normal userspace.. (don't remember it's sytnax, use google..)

Regards,

eblot, 2019-09-11:
Ok @RobertCNelson thanks for the details.

I'll try to find the proper syntax with `dtc` tool - and post the solution here if I manage to successfully use it :-)

RobertCNelson, 2019-09-11:
Found my note..

```fdtoverlay -i am335x-boneblack-uboot-univ.dtb -o am335x-boneblack.dtbo BB-BONE-eMMC1-01-00A0.dtbo AM335X-PRU-UIO-00A0.dtbo BB-UART2-00A0.dtbo BB-UART4-00A0.dtbo BB-UART5-00A0.dtbo```

eblot, 2019-09-13:
Closing this ticket, as this method works fine. Thanks again.

closed 2019-09-13 by eblot