beagleboard/bb.org-overlays #139: udev pwm vs gpio on   4.9.78-ti-r94 #1 SMP PREEMPT Fri Jan 26 21:26:24 UTC 2018

comarius, 2019-10-11:
Hello,
This is on BBB or BBG. Trying to have the /sys/class/pwm/pwmchip#/pwm# as non root:root.
IN the ditro I have there are udev rules for gpio and pwm to chown them as root:gpio and root:pwm on udev events, though...

    /sys/class/gpio$ echo 24 > ./export it does
    udevadm monitor
    monitor will print the received events for:
    UDEV - the event which udev sends out after rule processing
    KERNEL - the kernel uevent

    KERNEL[46318.324637] add      /devices/platform/ocp/44e07000.gpio/gpiochip0/gpio/gpio24 (gpio)
    UDEV  [46318.648582] add      /devices/platform/ocp/44e07000.gpio/gpiochip0/gpio/gpio24 (gpio)

and the new gpio24 is indeed owned by goio user,
but

    /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6$ echo 0 >export
    does not trigger any udev event,

and the new /sys/class/pwm/pwmchip6/pwm0 is owned by root

So, how do I make the new created node to be not owned by root:root

RobertCNelson, 2019-10-11:
@comarius this was fixed on Feb 5, 2018 with the release of 4.9.80-ti-r97, feel free to ugprade..

https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/09b7c9f919c3e7b7273091ff29196f8cf059c458#diff-e048185012bdb8bcbefadaf7c1c0dcd0

Regards,

comarius, 2019-10-11:
Thank you. upgraded to  4.14.108-ti-r108
Now I see all the pwm's being at boot there, but any attempt to write to enable/duty/polarity

```
marius@beaglebone:/sys/class/pwm/pwmchip0/pwm-0:0$ echo 1 > polarity 
-bash: echo: write error: Invalid argument
marius@beaglebone:/sys/class/pwm/pwmchip0/pwm-0:0$ echo 1 > enable 
-bash: echo: write error: Invalid argument
marius@beaglebone:/sys/class/pwm/pwmchip0/pwm-0:0$ echo on > enable 
-bash: echo: write error: Invalid argument
marius@beaglebone:/sys/class/pwm/pwmchip0/pwm-0:0$ echo okay > enable 
-bash: echo: write error: Invalid argument
marius@beaglebone:/sys/class/pwm/pwmchip0/pwm-0:0$ echo 20000 > ./duty_cycle 
-bash: echo: write error: Invalid argument
marius@beaglebone:/sys/class/pwm/pwmchip0/pwm-0:0$ sudo su
[sudo] password for marius: 
root@beaglebone:/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0#  echo 20000 > ./duty_cycle
bash: echo: write error: Invalid argument
root@beaglebone:/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0#  echo 1 > ./enable 
bash: echo: write error: Invalid argument
root@beaglebone:/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0#  echo true > ./enable 
bash: echo: write error: Invalid argument
root@beaglebone:/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0#  echo on > ./enable 
bash: echo: write error: Invalid argument
root@beaglebone:/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0# 

```

RobertCNelson, 2019-10-12:
@comarius at some point order of events changed..  do period before duty_cycle..

Regards,