beagleboard/bb.org-overlays #152: Unable to use config-pin on beagleboneAI

Zeki411, 2020-02-05:
Hello !
I cant use config-pin on beaglebone AI
When I use "config-pin P8.07 gpio" it shows the failure:
"P8_07 pinmux file not found!
Pin has no cape: P8_07"

Here is my version
`git:/opt/scripts/:[e457c010efc76c4e6fecccb7a5f7ff9b0597c4c0]
model:[BeagleBoard.org_BeagleBone_AI]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
UBOOT: Booted Device-Tree:[am5729-beagleboneai.dts]
kernel:[4.14.108-ti-r127]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~stretch+20200131]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyS0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet cape_universal=enable]
dmesg | grep remote
[   21.184257] remoteproc remoteproc0: 4b234000.pru is available
[   21.200250] remoteproc remoteproc1: 4b238000.pru is available
[   21.238528] remoteproc remoteproc2: 4b2b4000.pru is available
[   21.245386] remoteproc remoteproc3: 4b2b8000.pru is available
[   57.477947] remoteproc remoteproc4: 58820000.ipu is available
[   57.577602] remoteproc remoteproc5: 55020000.ipu is available
[   57.588544] remoteproc remoteproc6: 40800000.dsp is available
[   57.602690] remoteproc remoteproc7: 41000000.dsp is available
[   57.916450] remoteproc remoteproc4: powering up 58820000.ipu
[   57.916496] remoteproc remoteproc4: Booting fw image dra7-ipu1-fw.xem4, size 6867360
[   57.969234] remoteproc remoteproc5: powering up 55020000.ipu
[   57.969281] remoteproc remoteproc5: Booting fw image dra7-ipu2-fw.xem4, size 3751356
[   57.996729] remoteproc remoteproc4: registered virtio0 (type 7)
[   57.996753] remoteproc remoteproc4: remote processor 58820000.ipu is now up
[   58.152011] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core usb_f_ecm usb_f_mass_storage usb_f_rndis u_ether libcomposite iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack iptable_mangle iptable_filter cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc ip_tables x_tables
[   58.355121] remoteproc remoteproc5: registered virtio1 (type 7)
[   58.355147] remoteproc remoteproc5: remote processor 55020000.ipu is now up
[   59.824315] remoteproc remoteproc7: powering up 41000000.dsp
[   59.824362] remoteproc remoteproc7: Booting fw image dra7-dsp2-fw.xe66, size 20998684
[   59.840507] remoteproc remoteproc6: powering up 40800000.dsp
[   59.840554] remoteproc remoteproc6: Booting fw image dra7-dsp1-fw.xe66, size 20998684
[   59.985249] remoteproc remoteproc7: registered virtio2 (type 7)
[   59.985276] remoteproc remoteproc7: remote processor 41000000.dsp is now up
[   60.026705] remoteproc remoteproc6: registered virtio3 (type 7)
[   60.026729] remoteproc remoteproc6: remote processor 40800000.dsp is now up
dmesg | grep pru
[   21.156479] pruss 4b200000.pruss: creating PRU cores and other child platform devices
[   21.184257] remoteproc remoteproc0: 4b234000.pru is available
[   21.184427] pru-rproc 4b234000.pru: PRU rproc node /ocp/pruss_soc_bus@4b226004/pruss@0/pru@34000 probed successfully
[   21.200250] remoteproc remoteproc1: 4b238000.pru is available
[   21.200413] pru-rproc 4b238000.pru: PRU rproc node /ocp/pruss_soc_bus@4b226004/pruss@0/pru@38000 probed successfully
[   21.205026] pruss 4b280000.pruss: creating PRU cores and other child platform devices
[   21.238528] remoteproc remoteproc2: 4b2b4000.pru is available
[   21.238709] pru-rproc 4b2b4000.pru: PRU rproc node /ocp/pruss_soc_bus@4b2a6004/pruss@0/pru@34000 probed successfully
[   21.245386] remoteproc remoteproc3: 4b2b8000.pru is available
[   21.245543] pru-rproc 4b2b8000.pru: PRU rproc node /ocp/pruss_soc_bus@4b2a6004/pruss@0/pru@38000 probed successfully
[   58.152011] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core usb_f_ecm usb_f_mass_storage usb_f_rndis u_ether libcomposite iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack iptable_mangle iptable_filter cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc ip_tables x_tables
[   71.791728] pruss_uio_shmem 4b200000.pruss_shmem: Allocating gdev
[   71.791757] pruss_uio_shmem 4b200000.pruss_shmem: Allocating info
[   71.791780] pruss_uio_shmem 4b200000.pruss_shmem: Requesting resource
[   71.791846] pruss_uio_shmem 4b200000.pruss_shmem: Mapping resource
[   71.792081] pruss_uio_shmem 4b200000.pruss_shmem: Registering with uio driver
[   71.820452] pruss_uio_shmem 4b200000.pruss_shmem: Saving platform data
[   71.820802] pruss_uio_shmem 4b280000.pruss_shmem: Allocating gdev
[   71.820825] pruss_uio_shmem 4b280000.pruss_shmem: Allocating info
[   71.820845] pruss_uio_shmem 4b280000.pruss_shmem: Requesting resource
[   71.820903] pruss_uio_shmem 4b280000.pruss_shmem: Mapping resource
[   71.820944] pruss_uio_shmem 4b280000.pruss_shmem: Registering with uio driver
[   71.822480] pruss_uio_shmem 4b280000.pruss_shmem: Saving platform data
dmesg | grep pinctrl-single
[    0.974751] pinctrl-single 4a003400.pinmux: 282 pins at pa fc003400 size 1128
dmesg | grep gpio-of-helper
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END`

can anybody help me pls ?

RobertCNelson, 2020-02-05:
@Zeki411 config-pin will "NEVER" be supported on the AM5x based BeagleBone AI.

You will have to use u-boot overlays..

Regards,

Zeki411, 2020-02-05:
> @Zeki411 config-pin will "NEVER" be supported on the AM5x based BeagleBone AI.
> 
> You will have to use u-boot overlays..
> 
> Regards,

now i want to enable spi at P9.17,P9.18,P9.21,P9.22 but i dont know how to do it 
can you help me pls ?

Zeki411, 2020-02-05:
i try to use u-boot overlays by adding these following line in uEnv.txt but it doesnt work:

> ###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable
uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo

RobertCNelson, 2020-02-06:
@Zeki411 correct, those won't work.. All the "BB-*" and "PB-*" overlays are am335x based and will not work on the AM5x. There are a few "BBAI_*" example overlays.

But I haven't had any free personal time over this last few months to add more..

So, patches welcome..

Regards,

willisjr40, 2020-03-13:
I know when you write a python script using adafruit_bbio.spi as the import, it will auto load SPIDEV0 or SPIDEV1 into an available cape slot based on your code i.e. spi(0,0). 
Wouldn't it do this for this board? Auto load the correct cape?.... assuming there was one.

RobertCNelson, 2020-03-13:
@willisjr40 that doesn't support the BBAI yet..