beagleboard/bb.org-overlays #154: PIN49 conflict while enabling the UART2

allanlee0905, 2020-02-19:
Hi,

I am seeing error as follows: 

root@home:/sys/devices/platform/ocp# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLTBWA51650BBWG4272]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.11-rc3-dirty]
kernel:[4.9.78-ti-r94]
nodejs:[v8.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.14.20200122.1-0rcnee0~stretch+20200122]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups: ‘debian’: no such user
groups:[]
dmesg | grep pinctrl-single
[    1.424508] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    2.131931] pinctrl-single 44e10800.pinmux: pin PIN107 already requested by 48038000.mcasp; cannot claim for ocp:epp_pinmux
[    2.131942] pinctrl-single 44e10800.pinmux: pin-107 (ocp:amby-epm_pinmux) status -22
[    2.131951] pinctrl-single 44e10800.pinmux: could not request pin 107 (PIN107) from group pinmux_amby-epm  on device pinctrl-single
[    2.133095] pinctrl-single 44e10800.pinmux: pin PIN49 already requested by 0-0070; cannot claim for 48024000.serial
[    2.133105] pinctrl-single 44e10800.pinmux: pin-49 (48024000.serial) status -22
[    2.133113] pinctrl-single 44e10800.pinmux: could not request pin 49 (PIN49) from group pinmux_epp_uart  on device pinctrl-single
END

PIN49 can't be assigned as UART2's RTS/CTS pin, since this pin was already claimed by group - nxp_hdmi_bonelt_pins. 
The HDMI was disabled in uEnv.txt, but it seems HDMI related pins are still allocated before our customized overlay requests this pin.

debug: [console=ttyO0,115200n8 bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.enable_partno= root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet] ...

Is there any easy way to prevent the creation of pin group - 'nxp_hdmi_bonelt_pins'?

Thanks

RobertCNelson, 2020-02-19:
@allanlee0905 

u-boot is too old:

```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.11-rc3-dirty]
```

Upgrade it..
```
sudo /opt/scripts/tools/developers/update_bootloader.sh
```

Should work after a reboot..

Regards,

allanlee0905, 2020-02-21:
Thank you so much, Robert, Uboot update works for me.

closed 2020-02-21 by allanlee0905